project-chip/connectedhomeip #7543: M5Stack device core paniced after start  dns-sd server

Hugh-Bao, 2021-06-11:
#### Problem
On branch of test_event_3.1 , Using the python chip-tool to test with M5Stack device, there is an error occurred on device side when it starts dns-sd server after network commissioning completed.

#### Logs:
I (797455) chip[DIS]: Start dns-sd server
I (797455) chip[DIS]: Found admin pairing for admin 0, node 0x0000000000BC5C01
V (797455) chip[DIS]: Using wifi MAC for hostname
I (797465) chip[DIS]: Advertise operational node 0000000000000000-0000000000BC5C01
I (797475) app-devicecallbacks: Current free heap: 62508
Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC   : 0x400014dc PS   : 0x00060230 A0   : 0x801429d4 A1   : 0x3fff1800  
A2   : 0x00000001 A3   : 0xfffffffd A4   : 0x000000ff A5   : 0x0000ff00  
A6   : 0x00ff0000 A7   : 0xff000000 A8   : 0x43333538 A9   : 0x00000087  
A10   : 0x000005b3 A11   : 0x00000002 A12   : 0x3ffd3c69 A13   : 0x0000007f  
A14   : 0x3ffd3caa A15   : 0x00000005 SAR   : 0x00000010 EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001 LBEG  : 0x4000c349 LEND  : 0x4000c36b LCOUNT : 0xffffffff  
Backtrace:0x400014d9:0x3fff1800 0x401429d1:0x3fff1810 0x40142a8e:0x3fff1950 0x40143032:0x3fff1a90 0x401434dd:0x3fff1b00 0x401451cd:0x3fff1b40 0x4009059d:0x3fff1ba0
0x401429d1: _mdns_append_fqdn at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:422
0x40142a8e: _mdns_append_fqdn at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:467
0x40143032: _mdns_append_srv_record at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:674
 (inlined by) _mdns_append_answer at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:912
0x401434dd: _mdns_dispatch_tx_packet at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:1006
0x401451cd: _mdns_tx_handle_packet at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:3706
 (inlined by) _mdns_execute_action at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:3975
 (inlined by) _mdns_service_task at /Users/80266089/work/chip/esp/esp-idf/components/mdns/mdns.c:4098
0x4009059d: vPortTaskWrapper at /Users/80266089/work/chip/esp/esp-idf/components/freertos/xtensa/port.c:143

Hugh-Bao, 2021-06-11:
Python chip-tool issue "zcl NetworkCommissioning EnableNetwork" to device, device enter the network commissioning and success, but when device enter the process of start dns-sd the core panic occured on device.

dhrishi, 2021-06-11:
@Hugh-Bao  As per our discussion, can you please clean and re-build the all-clusters-app. Also, there are can be some issues with test_event_3.1 branch with Mac (fixed in master #7388  and #7389). I would suggest you to use Raspberry Pi (or Ubuntu Linux) as the controller which would be used during the actual test event.

closed 2022-01-26 by woody-apple