project-chip/connectedhomeip #7571: Android app failed to commission EFR32 Light App(CHIP over Thread device)

virupaksha, 2021-06-12:
I have Matter Over Thread device and I would like to commission it & control it by Android app But Unfortunately commission is not working.

Please find the logs below.

2021-06-13 01:09:23.813 28025-28025/com.google.chip.chiptool D/Camera: app passed NULL surface
2021-06-13 01:09:24.089 28025-28025/com.google.chip.chiptool D/SPL: JNI_OnLoad() called
2021-06-13 01:09:24.204 28025-28025/com.google.chip.chiptool D/SurfaceView: onWindowVisibilityChanged(8) false android.view.SurfaceView{2b5879a V.E...... ........ 0,0-1080,1884} of ViewRootImpl@11d6f65[CHIPToolActivity]
2021-06-13 01:09:24.205 28025-28025/com.google.chip.chiptool D/SurfaceView: surfaceDestroyed callback.size 1 #2 android.view.SurfaceView{2b5879a V.E...... ........ 0,0-1080,1884}
2021-06-13 01:09:24.205 28025-28025/com.google.chip.chiptool D/SurfaceView: remove() android.view.SurfaceView{2b5879a V.E...... ........ 0,0-1080,1884} Surface(name=SurfaceView - com.google.chip.chiptool/com.google.chip.chiptool.CHIPToolActivity@2b5879a@0)/@0xedb0180
2021-06-13 01:09:24.229 28025-28025/com.google.chip.chiptool D/CTL: JNI_OnLoad() called
2021-06-13 01:09:24.229 28025-28025/com.google.chip.chiptool D/CTL: Loading Java class references.
2021-06-13 01:09:24.230 28025-28025/com.google.chip.chiptool D/CTL: Java class references loaded.
2021-06-13 01:09:24.230 28025-28025/com.google.chip.chiptool D/CTL: Inet layer initialized.
2021-06-13 01:09:24.230 28025-28025/com.google.chip.chiptool D/CTL: BLE Layer being configured.
2021-06-13 01:09:24.231 28025-28025/com.google.chip.chiptool D/CTL: Asking for BLE Layer initialization.
2021-06-13 01:09:24.231 28025-28025/com.google.chip.chiptool D/CTL: BLE was initialized.
2021-06-13 01:09:24.231 28025-28025/com.google.chip.chiptool D/CTL: newDeviceController() called
2021-06-13 01:09:24.231 28025-28025/com.google.chip.chiptool D/TOO: KVS: Getting key ExampleOpCredsCAKey
2021-06-13 01:09:24.231 28025-28305/com.google.chip.chiptool D/CTL: IO thread starting
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/IN: TransportMgr initialized
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/DIS: Init admin pairing table with server storage
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/TOO: KVS: Getting key CHIPAdmin0
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/IN: Loading certs from KVS
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/IN: local node id is 0x000000000001B669
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/ZCL: Using ZAP configuration...
2021-06-13 01:09:24.232 28025-28025/com.google.chip.chiptool D/ZCL: deactivate report event
2021-06-13 01:09:24.233 28025-28025/com.google.chip.chiptool D/CTL: Getting operational keys
2021-06-13 01:09:24.233 28025-28025/com.google.chip.chiptool D/CTL: Generating credentials
2021-06-13 01:09:24.233 28025-28025/com.google.chip.chiptool D/CTL: Loaded credentials successfully
2021-06-13 01:09:24.233 28025-28025/com.google.chip.chiptool D/TOO: KVS: Getting key StartKeyID
2021-06-13 01:09:24.253 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-06-13 01:09:24.286 28025-28025/com.google.chip.chiptool D/ViewRootImpl@c99b357[Toast]: setView = android.widget.LinearLayout@e600e44 TM=true MM=false
2021-06-13 01:09:24.286 28025-28025/com.google.chip.chiptool V/Toast: Text: Tcan in android.widget.Toast$TN@fd6fb2d
2021-06-13 01:09:24.286 28025-28025/com.google.chip.chiptool I/chip.BluetoothManager: Starting Bluetooth scan
2021-06-13 01:09:24.287 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-06-13 01:09:24.287 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-06-13 01:09:24.288 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: BLE support array set: 010011
2021-06-13 01:09:24.288 28025-28025/com.google.chip.chiptool D/BluetoothLeScanner: Start Scan with callback
2021-06-13 01:09:24.290 28025-28058/com.google.chip.chiptool D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2021-06-13 01:09:24.307 28025-28025/com.google.chip.chiptool D/ViewRootImpl@c99b357[Toast]: Relayout returned: old=(0,63,1080,2094) new=(209,1810,871,1926) req=(662,116)0 dur=8 res=0x7 s={true 481337339904} ch=true
2021-06-13 01:09:24.307 28025-28142/com.google.chip.chiptool D/OpenGLRenderer: createReliableSurface : 0x6f9fe0a400, 0x7011f0f000
2021-06-13 01:09:24.308 28025-28142/com.google.chip.chiptool D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-06-13 01:09:24.346 28025-28025/com.google.chip.chiptool D/ViewRootImpl@c99b357[Toast]: MSG_RESIZED: frame=(209,1810,871,1926) ci=(0,0,0,0) vi=(0,0,662,116) or=1
2021-06-13 01:09:24.433 28025-28025/com.google.chip.chiptool I/chip.BluetoothManager: Bluetooth Device Scanned Addr: 6C:67:1D:49:A1:9D, Name EFR32_LIGHT
2021-06-13 01:09:24.495 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-06-13 01:09:24.495 28025-28025/com.google.chip.chiptool D/BluetoothLeScanner: Stop Scan with callback
2021-06-13 01:09:24.524 28025-28025/com.google.chip.chiptool I/chip.BluetoothManager: Connecting
2021-06-13 01:09:24.525 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-06-13 01:09:24.527 28025-28025/com.google.chip.chiptool D/BluetoothGatt: connect() - device: 6C:67:1D:49:A1:9D, auto: false
2021-06-13 01:09:24.527 28025-28025/com.google.chip.chiptool D/BluetoothAdapter: isSecureModeEnabled
2021-06-13 01:09:24.527 28025-28025/com.google.chip.chiptool D/BluetoothGatt: registerApp()
2021-06-13 01:09:24.528 28025-28025/com.google.chip.chiptool D/BluetoothGatt: registerApp() - UUID=38366e39-0d7b-4ea4-9eaf-97a9484d951c
2021-06-13 01:09:24.533 28025-28058/com.google.chip.chiptool D/BluetoothGatt: onClientRegistered() - status=0 clientIf=12
2021-06-13 01:09:24.828 28025-28058/com.google.chip.chiptool D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=12 device=6C:67:1D:49:A1:9D
2021-06-13 01:09:24.842 28025-28058/com.google.chip.chiptool I/chip.BluetoothManager|onConnectionStateChange: Discovering Services...
2021-06-13 01:09:24.843 28025-28058/com.google.chip.chiptool D/BluetoothGatt: discoverServices() - device: 6C:67:1D:49:A1:9D
2021-06-13 01:09:25.568 28025-28058/com.google.chip.chiptool D/BluetoothGatt: onConnectionUpdated() - Device=6C:67:1D:49:A1:9D interval=6 latency=0 timeout=500 status=0
2021-06-13 01:09:25.793 28025-28058/com.google.chip.chiptool D/BluetoothGatt: onSearchComplete() = Device=6C:67:1D:49:A1:9D Status=0
2021-06-13 01:09:25.794 28025-28058/com.google.chip.chiptool I/chip.BluetoothManager|onServicesDiscovered: Services Discovered
2021-06-13 01:09:25.794 28025-28058/com.google.chip.chiptool D/BluetoothGatt: configureMTU() - device: 6C:67:1D:49:A1:9D mtu: 131
2021-06-13 01:09:25.815 28025-28058/com.google.chip.chiptool D/BluetoothGatt: onConfigureMTU() - Device=6C:67:1D:49:A1:9D mtu=131 status=0
2021-06-13 01:09:25.816 28025-28058/com.google.chip.chiptool D/chip.BluetoothManager: MTU changed: connecting to CHIP device
2021-06-13 01:09:25.890 28025-28058/com.google.chip.chiptool D/BluetoothGatt: onConnectionUpdated() - Device=6C:67:1D:49:A1:9D interval=39 latency=0 timeout=500 status=0
2021-06-13 01:09:25.893 28025-28025/com.google.chip.chiptool D/ChipDeviceController: Bluetooth connection added with ID: 1
2021-06-13 01:09:25.893 28025-28025/com.google.chip.chiptool D/ChipDeviceController: Pairing device with ID: 2
2021-06-13 01:09:25.894 28025-28025/com.google.chip.chiptool D/CTL: pairDevice() called with device ID, connection object, and pincode
2021-06-13 01:09:25.894 28025-28025/com.google.chip.chiptool D/TOO: KVS: Getting key ListPairedDevices0
2021-06-13 01:09:25.894 28025-28025/com.google.chip.chiptool D/KeyValueStoreManager: Key 'ListPairedDevices0' not found in shared preferences
2021-06-13 01:09:25.895 28025-28025/com.google.chip.chiptool D/IN: BleConnectionComplete: endPoint 0x700f718ef8
2021-06-13 01:09:25.895 28025-28025/com.google.chip.chiptool D/CTL: Received GetMTU
[... 56 lines cut ...]
2021-06-13 01:09:55.913 28025-28305/com.google.chip.chiptool D/CTL: FindMethod:: javaClass exists? 1
2021-06-13 01:09:55.913 28025-28305/com.google.chip.chiptool D/CTL: FindMethod Returning 0
2021-06-13 01:09:55.914 28025-28305/com.google.chip.chiptool D/DeviceProvisioningFragment: Pairing status update: 1
[... cut ...]
2021-06-13 01:09:58.151 28025-28025/com.google.chip.chiptool D/ViewRootImpl@c22d13f[Toast]: MSG_RESIZED: frame=(356,1810,723,1926) ci=(0,0,0,0) vi=(0,0,367,116) or=1
2021-06-13 01:10:00.052 28025-28142/com.google.chip.chiptool W/libEGL: EGLNativeWindowType 0x6f9fe0a410 disconnect failed
2021-06-13 01:10:00.053 28025-28025/com.google.chip.chiptool D/ViewRootImpl@c22d13f[Toast]: dispatchDetachedFromWindow
2021-06-13 01:10:00.077 28025-28025/com.google.chip.chiptool D/InputTransport: Input channel destroyed: '3be0e69', fd=99

leasual, 2021-07-19:
i have the same issue. it says "connect handshake timed out, closing ep." have you solved it?

jmartinez-silabs, 2021-08-04:
Fixed with #8666

closed 2021-08-04 by jmartinez-silabs