project-chip/connectedhomeip #7602: tv-app command processing is broken and untested

bzbarsky-apple, 2021-06-14:
#### Problem
If I look at `MediaPlaybackManager`, it has things like this:
```
bool emberAfMediaPlaybackClusterMediaFastForwardCallback(chip::app::Command *)
{
    EmberAfMediaPlaybackStatus status = MediaPlaybackManager().proxyMediaPlaybackRequest(MediaPlaybackRequest::FastForward);
    sendResponse("MediaFastForward", ZCL_MEDIA_FAST_FORWARD_RESPONSE_COMMAND_ID, status);
    return true;
}
```
where `sendResponse` encodes a response in the ZCL encoding, which does not match the Matter command/response encoding.  This is obviously broken, in that the sender will not be able to decode the response, but nothing in CI tests this, so no one notices.

 #### Proposed Solution
Encode the response commands properly and add CI for tv-app.

@chrisdecenzo

lazarkov, 2021-06-14:
@bzbarsky-apple I'm working on this.

bzbarsky-apple, 2021-06-14:
Looks like this got broken by https://github.com/project-chip/connectedhomeip/pull/5945 because that only updated some command impls in the tree, not others?  As @lazarkov points out, door-lock is broken too.

@erjiaqing is that expected?  Are there issues tracking fixing the various things that #5945 should have fixed and did not?

lazarkov, 2021-07-01:
This issue is being resolved under [PR](https://github.com/project-chip/connectedhomeip/pull/7906#pullrequestreview-697359464)

woody-apple, 2022-01-26:
Looks resolved.

closed 2022-01-26 by woody-apple