project-chip/connectedhomeip #7641: CHIP All-Clusters-App cannot be build for Android OS

lazarkov, 2021-06-15:
#### Problem 1
Currently you cannot build all-clusters-app for Android OS. For example by using command:
`cd examples/all-clusters-app/linux/`
`gn gen out/android_arm --args='target_os="android" target_cpu="arm" android_ndk_root="/Users/username/Library/Android/sdk/ndk-bundle" android_sdk_root="/Users/username/Library/Android/sdk"'`
`ninja -C out/android_arm all`

 #### Problem 2
Even if you build somehow the app, the pairing onnetwork doesn't seem to work with the app:
`[1623753923790] [0x3f4fcb] CHIP: [SC] Assigned local session key ID 46
[1623753923790] [0x3f4fcb] CHIP: [IN] RETAINED IN SESS: 0x7f9ed180ba60 0
[1623753923790] [0x3f4fcb] CHIP: [SC] Sent PBKDF param request
[1623753928787] [0x3f4fce] CHIP: [EM] Retransmit MsgId:00000000 Send Cnt 1
[1623753929042] [0x3f4fce] CHIP: [EM] Retransmit MsgId:00000000 Send Cnt 2
[1623753929298] [0x3f4fe8] CHIP: [EM] Retransmit MsgId:00000000 Send Cnt 3
[1623753929555] [0x3f4fce] CHIP: [EM] Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3
[1623753953791] [0x3f507e] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1623753953791] [0x3f507e] CHIP: [SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1623753953791] [0x3f507e] CHIP: [TOO] Secure Pairing Failed`

 #### Proposed Solution
Go through the steps of building app and fix every error you come accross

@erjiaqing @andreilitvin

andy31415, 2021-06-15:
@austinh0

austinh0, 2021-06-23:
We haven't tried this use-case yet so it's not surprising that there are errors. For now I have to prioritize updating the Android controller code.

closed 2022-01-06 by lazarkov