project-chip/connectedhomeip #7648: Bare-minimum BLE functionality for Matter provisioning to be defined

dougleaf, 2021-06-15:
If we are able to scope the bare minimum BLE features needed, the BLE stacks provided by the silicon vendors may be optimized in Flash and RAM utilization

tcarmelveilleux, 2021-06-15:
What's the exact ask? Right now, the spec just calls for basic GATT support for BTP, so there is no mandated pairing/bonding or RPA support. Encryption is done at app layer over BTP, using PASE then secure messaging. The first thing I would disable in any BLE stack where it's possible, is everything related to BLE security manager. Otherwise, pretty much everything else of basic BLE (GAP, GATT) is needed. Also, the BLE stack may need features not explicitly stated in Matter, but commonly needed, to be able to BLE-certify a product with BT SIG, such as having HCI pipe support for RF tests, etc.

woody-apple, 2022-01-26:
Agree with @tcarmelveilleux, please clarify what's necessary here.

closed 2022-01-26 by woody-apple