project-chip/connectedhomeip #7667: The fabric id on the controller and paired device is not same

yufengwangca, 2021-06-15:
#### Problem

After connect device from chip-device-ctrl
chip-device-ctrl > connect -ip 192.168.86.157 20202021 12344321

On the controller side:
chip-device-ctrl > connect -ip 192.168.86.157 20202021 12344321
Device is assigned with nodeid = 12344321
[1623794346.220228][899522] CHIP:IN: RETAINED IN SESS: 0x2887500 0
[1623794346.430007][899530] CHIP:EM: Received message of type 33 and protocolId 0 on exchange 39017
[1623794346.432353][899530] CHIP:IN: RETAINED IN SESS: 0x2887500 0
[1623794346.443027][899530] CHIP:EM: Received message of type 35 and protocolId 0 on exchange 39017
[1623794346.444790][899530] CHIP:IN: RETAINED IN SESS: 0x2887500 0
[1623794346.445109][899530] CHIP:CTL: Arming failsafe
[1623794346.445336][899530] CHIP:IN: Send message of type 8 and protocolId 5 on exchange 39018
[1623794346.445367][899530] CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000BC5C01 at utc time: 376500372 msec
[1623794346.445390][899530] CHIP:IN: Sending secure msg on generic transport
[1623794346.445468][899530] CHIP:IN: Secure msg send status No Error
[1623794346.455158][899530] CHIP:EM: Received message of type 16 and protocolId 0 on exchange 39017
[1623794346.469771][899530] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 0
[1623794346.469807][899530] CHIP:IN: Setting fabricID BC5C01 on admin.
[1623794346.469812][899530] CHIP:IN: Since admin was modified, persisting changes to KVS
[1623794346.469842][899530] CHIP:EM: Received message of type 9 and protocolId 5 on exchange 39018
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1623794346.470075][899530] CHIP:ZCL: DefaultResponse:
[1623794346.470081][899530] CHIP:ZCL:   Transaction: 0x7fa5af694398
[1623794346.470086][899530] CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
[1623794346.470092][899530] CHIP:CTL: Received success response 0xe1c0
.......
Secure Session to Device Established

On the paired node side:
> [1623794346.429247][89556] CHIP:EM: Received message of type 32 and protocolId 0 on exchange 39017
[1623794346.429276][89556] CHIP:EM: ec id: 39017, Delegate: 0x0x55f6071c2b60
[1623794346.429317][89556] CHIP:SC: Received PBKDF param request
[1623794346.429905][89556] CHIP:EM: Piggybacking Ack for MsgId:00000000 with msg
[1623794346.429923][89556] CHIP:IN: RETAINED IN SESS: 0x55f6071c4800 0
[1623794346.429975][89556] CHIP:SC: Sent PBKDF param response
[1623794346.472232][89556] CHIP:EM: Received message of type 34 and protocolId 0 on exchange 39017
[1623794346.472266][89556] CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1623794346.472275][89556] CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
[1623794346.472323][89556] CHIP:SC: Received spake2p msg1
[1623794346.472620][89556] CHIP:SC: Peer assigned session key ID 0
[1623794346.472968][89556] CHIP:EM: Piggybacking Ack for MsgId:00000000 with msg
[1623794346.472983][89556] CHIP:IN: RETAINED IN SESS: 0x55f6071c4800 0
[1623794346.473045][89556] CHIP:SC: Sent spake2p msg2
[1623794346.485590][89556] CHIP:EM: Received message of type 36 and protocolId 0 on exchange 39017
[1623794346.485623][89556] CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1623794346.485648][89556] CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
[1623794346.485679][89556] CHIP:SC: Received spake2p msg3
[1623794346.485735][89556] CHIP:EM: Sending Standalone Ack for MsgId:00000000
[1623794346.485798][89556] CHIP:EM: Flushed pending ack for MsgId:00000000
[1623794346.485811][89556] CHIP:IN: New pairing for device 0x0000000000000000, key 0!!
[1623794346.485858][89556] CHIP:SVR: Device completed SPAKE2+ handshake
[1623794346.485867][89556] CHIP:SVR: Device completed Rendezvous process
[1623794346.485888][89556] CHIP:SVR: Stored value in server storage
[1623794346.486032][89556] CHIP:DL: writing settings to file (/tmp/chip_server_kvs.tmp)
[1623794346.486274][89556] CHIP:DL: renamed tmp file to file (/tmp/chip_server_kvs)
[1623794346.486287][89556] CHIP:SVR: Stored value in server storage
[1623794346.486335][89556] CHIP:DL: writing settings to file (/tmp/chip_server_kvs.tmp)
[1623794346.492742][89556] CHIP:DL: renamed tmp file to file (/tmp/chip_server_kvs)
[1623794346.492826][89556] CHIP:IN: Secure transport received message destined to fabric 0, node 0x0000000000BC5C01. Key ID 0
[1623794346.492886][89556] CHIP:IN: Setting nodeID BC5C01 on admin.
[1623794346.492908][89556] CHIP:IN: Setting fabricID 1B669 on admin.
[1623794346.492927][89556] CHIP:IN: Since admin was modified, persisting changes to KVS
[1623794346.494079][89556] CHIP:SVR: Stored value in server storage

 #### Proposed Solution
<suggested fix, suggested enhancement>

yufengwangca, 2021-06-15:
@pan-apple @bzbarsky-apple

yufengwangca, 2021-06-15:
I expect the controller and paired device share the same fabric id. we need to check if the following code cause the mismatch

    // TODO: Remove temporary code once AddOptCert is implemented
    if (packetHeader.GetSourceNodeId().HasValue())
    {
        fabricId = packetHeader.GetSourceNodeId().Value();
        if (fabricId != kUndefinedFabricId && admin->GetFabricId() != fabricId)
        {
            admin->SetFabricId(packetHeader.GetSourceNodeId().Value());
            ChipLogProgress(Inet, "Setting fabricID %" PRIX64 " on admin.", admin->GetFabricId());
            modifiedAdmin = true;
        }
    }

bzbarsky-apple, 2021-06-16:
We don't really implement fabric ids in a meaningful way yet.  Last I checked, the device used the node id of the commissioner as the "fabric" id in some cases, and 0 in other cases....  And the commissioner does whatever it does.

yufengwangca, 2021-06-16:
The key issue here is the paired device and controller have different fabric ID.

bzbarsky-apple, 2021-06-17:
Right, because we don't have anything resembling a spec-compliant implementation of fabric (e.g. derived from the cert) on the device side.

xylophone21, 2021-07-01:
So how could we debug it now?

yufengwangca, 2021-07-01:
Here is the expected flow:

The Fabric ID should not be the same as a Node ID, It should be a random number created by the Commissioner (Controller) that creates the network while pairing the first device. Before fabric network is created, the default fabricId is 0.

Then it's up to the controller to appropriately mint an op cert for the paired device that contains the same FabricID as that of the controller itself, and when the paired device 'joins' a fabric at the time it receives an operational cert, with the fabric ID embedded as a subject DN within that cert.

bzbarsky-apple, 2021-07-01:
That is the theory, yes.  At the moment the device does not get its fabric id from the opcert, so none of that works.  Obviously we need to fix things to work the way you describe; it's just not done yet.

bzbarsky-apple, 2021-07-01:
@xylophone21 I guess my question is: what are you trying to debug?  What are the exact things you are seeing fail, and under what circumstances?

xylophone21, 2021-07-02:
@bzbarsky-apple  Since the Fabric ID is  0 on node, when I finished Commissioning, the chip-tool will discover node by ChipMdnsResolve with  Fabric ID and nodeId. This step will failed and I can not going on.

I think I have a work-around fix by foruce set Fabric ID to 0 at chip-tool side, thanks

bzbarsky-apple, 2021-07-02:
Yes, you need to tell chip-tool what fabric id the node will use; that's why it has that command-line argument.  If the node is going to use 0, that's what you should tell chip-tool to use as well.

woody-apple, 2022-01-26:
Marking as closed for now, let us know if you need anything else.

closed 2022-01-26 by woody-apple