project-chip/connectedhomeip #7682: chip-device-ctrl timeout

doru91, 2021-06-16:
#### Problem
 
The connection between chip-device-ctrl and NXP K32W is not successful. A recent commit from master branch is used (b0eeffc345840f84a25a69b77c996d9e7fe1b915). 

The main issue seems to be a timer which seems to expire in the middle of operational credentials processing on the chip-device-ctrl-side. I run several experiments and it looks that there are 2 times when the timeout occurs:

1. while waiting for OpCsrRsp:

- see chip-device-ctrl-timeout_chip_error_incorrect_state for logs on the chip-device-ctrl side;
- see dut_timeout_chip_error_incorrect_state for logs on the DUT side
- see wireshark_timeout_chip_error_incorrect_state for air traffic

2. while sending/receiving the operational certificate to/from dut:
- see chip-device-ctrl-timeout for logs on the chip-device-ctrl side;
- see dut_timeout for logs on the DUT side
- see wireshark_timeout for air traffic

cc @pan-apple @tcarmelveilleux @jmartinez-silabs 

[chip-device-ctrl-archive.zip](https://github.com/project-chip/connectedhomeip/files/6661649/chip-device-ctrl-archive.zip)

erjiaqing, 2021-06-16:
From the log, I see

```
[1623830986.241045][484454] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623830986.241123][484454] CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
[1623830986.241176][484454] CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
[1623830986.241221][484454] CHIP:BLE: local and remote recv window size = 3
[1623830986.241410][484454] CHIP:IN: BLE EndPoint 0xffff938e2e80 Connection Complete
[1623830986.386475][484454] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1623831000.037347][484449] CHIP:DL: Indication received, conn = 0xffff88036fe0
[1623831000.037640][484453] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623831000.183527][484449] CHIP:DL: Indication received, conn = 0xffff88036fe0
[1623831000.183805][484453] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623831000.475582][484453] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
```

Noticed that it waits for 14s between BLE connection complete and on the device side before it received PBKDF response, I sww

```
[Info]Received message of type 32 and protocolId 0 on exchange 39017
[Info]ec id: 39017, Delegate: 0x0x4010008
[Info]Received PBKDF param request
[Info]Sent PBKDF param response
[Info]Long dispatch time: 20680 ms
[Info]Long dispatch time: 140 ms
[Info]Long dispatch time: 290 ms
[Info]Long dispatch time: 150 ms
[Info]Long dispatch time: 220 ms
```

```
[Info]Long dispatch time: 20680 ms
```

I guess the timeout happened on controller, which the following timeout reached

https://github.com/project-chip/connectedhomeip/blob/1397b4b854e7386138d6d162df4d3e5a00454863/src/controller/CHIPDeviceController.cpp#L101

Since the device suffered from pretty long dispatch time (20seconds!), I think this is not an issue on chip-device-ctrl.

Also, maybe you can lift the timeout above to 1minute to see if it can pair.

erjiaqing, 2021-06-16:
After checking the other log, I think this two logs is caused by the same root cause, but the state when timeout happened is different.

doru91, 2021-06-16:
Thanks, @erjiaqing 

Indeed, the problem seems to be cause by the long delay between receiving the PBKDF param request and sending PBKDF param response. I am investigating this.

jmartinez-silabs, 2021-06-16:
@erjiaqing @doru91 
I don't see a long delay to receive PBKDF para response. But i get  **Device failed to receive the operational certificate Response: 0x01**

Here is the log For the controller followed by the efr32 device
```
chip-device-ctrl > connect -ble 3840 73141520 1
Device is assigned with nodeid = 1
[1623849992.733497][54851] CHIP:SC: Assigned local session key ID 0
[1623849992.733759][54851] CHIP:IN: Message appended to BLE send queue
[1623849992.733854][54851] CHIP:SC: Sent PBKDF param request
[1623849992.733880][54868] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1623849992.734130][54854] CHIP:DL: TRACE: Bus acquired for name C-d643
[1623849992.733947][54851] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1623849992.745727][54858] CHIP:DL: PlatformBlueZInit init success
[1623849992.751371][54854] CHIP:BLE: BLE removing known devices.
[1623849992.751435][54854] CHIP:BLE: BLE initiating scan.
[1623849993.420803][54854] CHIP:BLE: Device 16:6B:DB:48:73:00 does not look like a CHIP device.
[1623849993.453386][54854] CHIP:BLE: New device scanned: 75:89:2B:48:49:D1
[1623849993.453464][54854] CHIP:BLE: Device discriminator match. Attempting to connect.
[1623849993.458475][54854] CHIP:BLE: Scan complete notification without an active scan.
[1623849993.817371][54854] CHIP:DL: ConnectDevice complete
[1623849995.255380][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service000e
[1623849995.255433][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255466][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service000e
[1623849995.255488][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255512][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service000e
[1623849995.255531][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255558][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0001
[1623849995.255578][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255601][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0001
[1623849995.255620][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255642][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0001
[1623849995.255661][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255691][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service000e
[1623849995.255711][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255734][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service000e
[1623849995.255753][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255775][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service000e
[1623849995.255795][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255825][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255845][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255877][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255897][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255921][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255940][54854] CHIP:DL: Char1 /org/bluez/hci0/dev_75_89_2B_48_49_D1/service0018
[1623849995.255967][54854] CHIP:DL: New BLE connection 0xffff900403f0, device 75:89:2B:48:49:D1, path /org/bluez/hci0/dev_75_89_2B_48_49_D1
[1623849995.256115][54858] CHIP:DL: HandlePlatformSpecificBLEEvent 16385
[1623849995.256154][54858] CHIP:IN: BleConnectionComplete: endPoint 0xffff9acfde48
[1623849995.716993][54858] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1623849995.815644][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16388
[1623849995.815741][54859] CHIP:BLE: subscribe complete, ep = 0xffff9acfde48
[1623849995.815998][54854] CHIP:DL: Indication received, conn = 0xffff900403f0
[1623849995.816230][54858] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623849995.816360][54858] CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
[1623849995.816420][54858] CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
[1623849995.816469][54858] CHIP:BLE: local and remote recv window size = 3
[1623849995.816673][54858] CHIP:IN: BLE EndPoint 0xffff9acfde48 Connection Complete
[1623849995.911778][54858] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1623849996.010308][54854] CHIP:DL: Indication received, conn = 0xffff900403f0
[1623849996.010602][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623849996.107751][54854] CHIP:DL: Indication received, conn = 0xffff900403f0
[1623849996.107978][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623849996.204814][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1623849996.205577][54854] CHIP:DL: Indication received, conn = 0xffff900403f0
[1623849996.205801][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623849996.303341][54854] CHIP:DL: Indication received, conn = 0xffff900403f0
[1623849996.303625][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623849996.399302][54859] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1623849996.400075][54854] CHIP:DL: Indication received, conn = 0xffff900403f0
[... cut ...]
00> <info  > [IN] Sending msg from 0x0000000000000001 to 0x000000000001B669 at utc time: ld msec
00> <info  > [IN] Sending secure msg on generic transport
00> <info  > [IN] Secure msg send status No Error
00> <detail> [DMG] ICR moving to [   Sending]
00> <detail> [DMG] ICR moving to [Uninitiali]
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] Char Write Req, char : 26

```

tcarmelveilleux, 2021-06-16:
@jmartinez-silabs 

```
00> <info  > [ZCL] OpCreds: commissioner has added an Op Cert
00> <info  > [ZCL] OpCreds: Finding admin with fabricId 0x000000000001B669.
00> <info  > [DIS] Checking ind:0 [fabricId 0x000000000001B669 nodeId 0x0000000000000001 vendorId 0] vs [fabricId 0x000000000001B669 nodeId 0x0000000000000000 vendorId 0]
00> <info  > [DIS] Found a match!
00> <detail> [DMG] ICR moving to [AddCommand]
00> <info  > [ZCL] OpCreds: Failed AddOpCert request.
```

Based on that log, the code is:

```
bool emberAfOperationalCredentialsClusterAddOpCertCallback(chip::app::Command * commandObj, chip::ByteSpan OperationalCert,
                                                           chip::ByteSpan IPKValue, chip::NodeId CaseAdminNode,
                                                           uint16_t AdminVendorId)
{
    EmberAfStatus status = EMBER_ZCL_STATUS_SUCCESS;

    emberAfPrintln(EMBER_AF_PRINT_DEBUG, "OpCreds: commissioner has added an Op Cert");

    AdminPairingInfo * admin = retrieveCurrentAdmin();
    VerifyOrExit(admin != nullptr, status = EMBER_ZCL_STATUS_FAILURE);

    VerifyOrExit(admin->SetOperationalCert(OperationalCert) == CHIP_NO_ERROR, status = EMBER_ZCL_STATUS_FAILURE);
    VerifyOrExit(GetGlobalAdminPairingTable().Store(admin->GetAdminId()) == CHIP_NO_ERROR, status = EMBER_ZCL_STATUS_FAILURE);

exit:
    emberAfSendImmediateDefaultResponse(status);
    if (status == EMBER_ZCL_STATUS_FAILURE)
    {
        emberAfPrintln(EMBER_AF_PRINT_DEBUG, "OpCreds: Failed AddOpCert request.");
    }

    return true;
}
```

There are 3 paths that lead to `EMBER_ZCL_STATUS_FAILURE`.

I suggest **replacing all `VerifyOrExit` with `if()` blocks that do proper logging**, while we await proper support of `OpCertResponse` which would allow sending back detailed failure indication.

tcarmelveilleux, 2021-06-16:
@doru91 Strongly suggest the two following things:

1. in `src/include/platform/internal/GenericPlatformManagerImpl.cpp`
```
    // TODO: make this configurable
#if CHIP_PROGRESS_LOGGING
    uint32_t delta = (static_cast<uint32_t>(System::Layer::GetClock_MonotonicHiRes() - startUS)) / 1000;
    if (delta > 100)
    {
        ChipLogError(DeviceLayer, "Long dispatch time: %" PRId32 " ms", delta);
    }
#endif // CHIP_PROGRESS_LOGGING
```
  - Augment the log to print the event type that caused long dispatch, if you also can determine if it is a particular type, to print a bit more info. These long dispatches can cause really bad things and you need to find the culprit to go on debugging.

2. Add at least "milliseconds since boot" over 16 or 32 bits to each log on your platform impl. Not having timestamps on logs makes debugging these issues very difficult since you cannot "time-align" between a packet capture and the protocol stack.

In `PASESession::SendPBKDFParamResponse()` (, there is the execution of the PBDKF2, whioch is easily the **costliest time operation overall** for any crypto.

```
    // Update commissioning hash with the pbkdf2 param response that's being sent.

    // ==== HERE, before doing more, do:
    // --> ChipLogDetail(SecureChannel, "Start of pbkdf: %ums, iteration count: %u", somehow_get_ms_timestamps_here(), mIterationCount);

    ReturnErrorOnFailure(mCommissioningHash.AddData(resp->Start(), resp->DataLength()));
    ReturnErrorOnFailure(SetupSpake2p(mIterationCount, mSalt, mSaltLength));
    ReturnErrorOnFailure(mSpake2p.ComputeL(mPoint, &sizeof_point, &mPASEVerifier[1][0], kSpake2p_WS_Length));

    // --> HERE:
    // ChipLogDetail(SecureChannel, "End of pbkdf: %ums", somehow_get_ms_timestamps_here());
```

I added inline suggestions above for extra logging to root cause. If you add timestamps to all log lines, you don't need to add the ms logging in the actual log call arguments.

Overall, my hypothesis is that PBKDF computation is taking too long.

Would you be able to answer the following questions about K32W (sorry, not familiar with all that platform's details):

- Which implementation of PBKDF2 you are using
- CPU clock speed during your run
- Peripheral clock speed during your run if using HW assisted PBKDF2

In `src/crypto/hsm/nxp/CHIPCryptoPALHsm_SE05X_PBKDF.cpp`, I see 
```
 smStatus = Se05x_API_PBKDF2(&((sss_se05x_session_t *) &gex_sss_chip_ctx.session)->s_ctx, keyid, kSE05x_Pbkdf2_HMAC_SHA256, salt,
                                slen, (uint16_t) iteration_count, (uint16_t) key_length, output, (size_t *) &key_length);
```

Looking at the code, I think there will be issues since iteration count possible is 10000..100000 per spec, but iteration_count passed to HW for that impl is clipped to 0..65535 (`uint16_t`), so if you are using this impl and have iteration count ever configured > 65535, you will fail to generate the correct output.

doru91, 2021-06-17:
> @doru91 Strongly suggest the two following things:
> 
> 1. in `src/include/platform/internal/GenericPlatformManagerImpl.cpp`
> 
> ```
>     // TODO: make this configurable
> #if CHIP_PROGRESS_LOGGING
>     uint32_t delta = (static_cast<uint32_t>(System::Layer::GetClock_MonotonicHiRes() - startUS)) / 1000;
>     if (delta > 100)
>     {
>         ChipLogError(DeviceLayer, "Long dispatch time: %" PRId32 " ms", delta);
>     }
> #endif // CHIP_PROGRESS_LOGGING
> ```
> 
> * Augment the log to print the event type that caused long dispatch, if you also can determine if it is a particular type, to print a bit more info. These long dispatches can cause really bad things and you need to find the culprit to go on debugging.
> 
> 1. Add at least "milliseconds since boot" over 16 or 32 bits to each log on your platform impl. Not having timestamps on logs makes debugging these issues very difficult since you cannot "time-align" between a packet capture and the protocol stack.
> 
> In `PASESession::SendPBKDFParamResponse()` (, there is the execution of the PBDKF2, whioch is easily the **costliest time operation overall** for any crypto.
> 
> ```
>     // Update commissioning hash with the pbkdf2 param response that's being sent.
> 
>     // ==== HERE, before doing more, do:
>     // --> ChipLogDetail(SecureChannel, "Start of pbkdf: %ums, iteration count: %u", somehow_get_ms_timestamps_here(), mIterationCount);
> 
>     ReturnErrorOnFailure(mCommissioningHash.AddData(resp->Start(), resp->DataLength()));
>     ReturnErrorOnFailure(SetupSpake2p(mIterationCount, mSalt, mSaltLength));
>     ReturnErrorOnFailure(mSpake2p.ComputeL(mPoint, &sizeof_point, &mPASEVerifier[1][0], kSpake2p_WS_Length));
> 
>     // --> HERE:
>     // ChipLogDetail(SecureChannel, "End of pbkdf: %ums", somehow_get_ms_timestamps_here());
> ```
> 
> I added inline suggestions above for extra logging to root cause. If you add timestamps to all log lines, you don't need to add the ms logging in the actual log call arguments.
> 
> Overall, my hypothesis is that PBKDF computation is taking too long.
> 

@tcarmelveilleux thanks a lot for your help.

I submitted a PR (https://github.com/project-chip/connectedhomeip/pull/7717) that improves the logging capabilities on our platform for also including a timestamp. Indeed, this is very helpful.

Time needed for sending the PBKDF param response is around 15.6s (100 iterations) when using the SE and we are investigating this internally. Indeed, this is also root issue for the timeout. 

> Would you be able to answer the following questions about K32W (sorry, not familiar with all that platform's details):
> 
> * Which implementation of PBKDF2 you are using
> * CPU clock speed during your run
> * Peripheral clock speed during your run if using HW assisted PBKDF2
> 

At this moment PBKDF is offloaded to SE but we plan to move it back to K32W (48Mhz). Anyway, from my understanding, there are 2 cases when the PBKDF needs to be generated on the commissionee side:

1.	Either they use a fixed passcode and the passcode is transformed into a verifier (w0,L) stored in the device at manufacturing time
2.	Or they use a dynamic passcode and the end-node device will take some time to establish the corresponding verifier but this is happening rarely

So I believe that the current implementation is compliant with point 2 but the iteration number seems to be too low (only 100).

> In `src/crypto/hsm/nxp/CHIPCryptoPALHsm_SE05X_PBKDF.cpp`, I see
>
> ```
>  smStatus = Se05x_API_PBKDF2(&((sss_se05x_session_t *) &gex_sss_chip_ctx.session)->s_ctx, keyid, kSE05x_Pbkdf2_HMAC_SHA256, [...]

Good catch, will fix this.

tcarmelveilleux, 2021-06-17:
@doru91 

> At this moment PBKDF is offloaded to SE but we plan to move it back to K32W (48Mhz). Anyway, from my understanding, there are 2 cases when the PBKDF needs to be generated on the commissionee side: [...]

For the onboarding payload passcode, there never needs to be a computation. Right now, PBKDFParameterResponse processing causes the full computation of verifier, which is incorrect and costly. The verifier should have been available to obtain from the platform in that "out of box" case.

Quote from SPAKE2+:

```

3.3.  SPAKE2+

   Let w0 and w1 be two integers derived by hashing the password pw with
   the identities of the two participants, A and B.  Specifically,
   compute:

   w0s || w1s = PBKDF(len(pw) || pw || len(A) || A || len(B) || B)
   w0 = w0s mod p
   w1 = w1s mod p

   If both identities A and B are absent, i.e. len(A) = len(B) = 0, then
   the length prefixes are omitted as in Section 3.1.

   w0s || w1s = PBKDF(pw)
```

w0 and L are never needed to be computed by B (the receiver).

> 2. Or they use a dynamic passcode and the end-node device will take some time to establish the corresponding verifier but this is happening rarely

For this w0/L verifier, it actually needs to be computed by the first admin that opens the commissioning window, and **given**  to the responder. 

> So I believe that the current implementation is compliant with point 2 but the iteration number seems to be too low (only 100).

Overall, I would say **only the commissioner** needs to compute PBKDF. The code for it should not be required on device. Right now I think it is required because nobody has done the work to allow examples to precompute and store everything related to the PASE for commissioning:

- The onboarding payload with passcode and discriminator (different for each device)
- The w0/L verifier pair

doru91, 2021-06-17:
Thanks, @tcarmelveilleux . I agree that the code for PBKDF should not be required on the device and the CHIP code needs to be refactored such that the w0/L is provisioned at manufacturing time.

I think the issue can be closed.

tcarmelveilleux, 2021-06-18:
@doru91 Could you please open a separate issue about not doing PBKDF on target?

doru91, 2021-06-18:
Sure, opened https://github.com/project-chip/connectedhomeip/issues/7738

tcarmelveilleux, 2021-06-18:
Issues root-caused in workarounds in progress.

closed 2021-06-18 by tcarmelveilleux