project-chip/connectedhomeip #7692: KeypadInputManager's SendKey callback is never called.

bzbarsky-apple, 2021-06-16:
#### Problem
We have two implementations of `emberAfKeypadInputClusterSendKeyCallback` in the tree:
```
bool emberAfKeypadInputClusterSendKeyCallback(chip::app::Command * commandObj, unsigned char)
```
in `src/app/clusters/keypad-input-server/keypad-input-server.cpp` and 
```
emberAfKeypadInputClusterSendKeyCallback(EmberAfKeypadInputCecKeyCode keyCode)
```
in `examples/tv-app/linux/include/keypad-input/KeypadInputManager.cpp`.

The codegen, if it had that command enabled (which it does not) would call the two-arg version, which just sends a DefaultResponse and does nothing.  It definitely does not call into `KeypadInputManager` in any way, which is what I would expect it to do.

 #### Proposed Solution
1. Enable the SendKey command in the tv-app zap configuration.
2. Make the SDK-provided cluster callback call the app-provided callback.
3. Add a CI test to make sure this all works so we don't break it again.

@lazarkov

lazarkov, 2021-07-01:
This issue is being resolved under [PR](https://github.com/project-chip/connectedhomeip/pull/7906#pullrequestreview-697359464)

woody-apple, 2022-01-26:
Looks resolved?

closed 2022-01-26 by woody-apple

bzbarsky-apple, 2022-01-26:
Yes, this looks fixed now.