project-chip/connectedhomeip #7705: chip-device-ctrl - resolve command hangs with lock-app

PSONALl, 2021-06-17:
#### Problem
chip-device-ctrl - resolve command hangs with lock-app on esp32
 * expected behavior : resolve dns and update address of the node in the device controller after command
      `resolve 0 135246`
 * actual behavior - resolve command hangs
 * system configuration
    - platform : macOS
    - kit : esp32
 *  logs
    - device side : [device-logs.txt](https://github.com/project-chip/connectedhomeip/files/6667681/device-logs.txt)

    - python controller side : [chip-device-ctrl.txt](https://github.com/project-chip/connectedhomeip/files/6667678/chip-device-ctrl.txt)

DamienKastner, 2021-06-17:
I have exactly the same issue.
It turns out that mDNS is not running on the ESP32 (lock-app example). I compared with the all-cluster-example log, and we should clearly see mDNS logs.

We can see that the mDNS server is not started when the target is ESP32.  
```cpp
// src/app/server/Server.cpp (line 530)

// ESP32 examples have a custom logic for enabling DNS-SD
#if CHIP_DEVICE_CONFIG_ENABLE_MDNS && !CHIP_DEVICE_LAYER_TARGET_ESP32
    app::Mdns::StartServer();
#endif
```
I don't know what is the custom logic for enabling DNS-SD for ESP32...
  
**Solution to fix the example:**  

Add this line after server initialization
```cpp
// examples/lock-app/esp32/main/main.cpp (line 70)

    InitServer();

    app::Mdns::AdvertiseCommissionableNode();
```

(Re)start mDNS server at connectivity changes
```cpp
// examples/lock-app/esp32/main/DeviceCallbacks.cpp (line 90)

void DeviceCallbacks::OnInternetConnectivityChange(const ChipDeviceEvent * event)
{
    if (event->InternetConnectivityChange.IPv4 == kConnectivity_Established)
    {
        ESP_LOGI(TAG, "Server ready at: %s:%d", event->InternetConnectivityChange.address, CHIP_PORT);
        chip::app::Mdns::StartServer();
    }
    else if (event->InternetConnectivityChange.IPv4 == kConnectivity_Lost)
    {
        ESP_LOGE(TAG, "Lost IPv4 connectivity...");
    }
    if (event->InternetConnectivityChange.IPv6 == kConnectivity_Established)
    {
        ESP_LOGI(TAG, "IPv6 Server ready...");
        chip::app::Mdns::StartServer();
    }
    else if (event->InternetConnectivityChange.IPv6 == kConnectivity_Lost)
    {
        ESP_LOGE(TAG, "Lost IPv6 connectivity...");
    }
}
```  

Don't forget to include headers in both files.

```cpp
// examples/lock-app/esp32/main/DeviceCallbacks.cpp
#include <app/server/Mdns.h>
#include <lib/mdns/Advertiser.h>
```
```cpp
// examples/lock-app/esp32/main/main.cpp
#include <app/server/Mdns.h>
```

klem-everywhere, 2021-06-17:
I work with @DamienKastner .
We can confirm that for now we have to manually call `app::Mdns::StartServer` every time on IP change to make it work, but we wonder what the proper solution should be.
Is it better remove the `custom logic` for ESP32 and we fix it "properly"? Is there a limitation in `app/server/Server.cpp` or `lib/mdns` that we have to restart mdns on IP change every time or can we trigger the restart every time on IP change in `platform/ESP32`?

klem-everywhere, 2021-06-17:
@DamienKastner can you send a PR for the fix?

klem-everywhere, 2021-06-17:
Reading further into https://github.com/project-chip/connectedhomeip/pull/5154 gives me no clue of why `PlatformMgr().AddEventHandler(ChipEventHandler, {});` wouldn't work for ESP32 in `https://github.com/project-chip/connectedhomeip/pull/6227/files` (code is removed now but ESP32 remains disabled).
If there actually is a problem we could maybe put `PlatformMgr().AddEventHandler(ChipEventHandler, {});` back into `app/server/Server.cpp` with `#if CHIP_DEVICE_LAYER_TARGET_ESP32`?

PR #7723 (2021-06-21): esp32 lock-app mdns server start at connectivity change
#### Problem
What is being fixed?  Examples:
* Fixes #7705 chip-device-ctrl - resolve command hangs with lock-app.

#### Change overview
esp32 lock-app mdns server start at connectivity change

#### Testing
How was this tested? (at least one bullet point required)
* using lock-app on esp32
closed 2021-06-21 by woody-apple