project-chip/connectedhomeip #7725: Re-factor of how threading and event dispatch is handled in the SDK

mrjerryjohns, 2021-06-17:
#### Problem
* Approaches to threading and event dispatch vary quite widely between different platforms and approaches (POSIX vs GCD on Darwin vs FreeRTOS)
* PlatformManager attempted to provide a singular API to unify these differences, but it ended up being too opinionated about policy, and makes it restrictive on certain platforms.
    * E.g. `DeviceController::Init` calls `StartEventLoopTask`, which always spins up a thread on POSIX platforms.
* Results in an illusion of API conformity, when in reality, different platforms are implementing the different PlatformManager APIs quite differently, resulting in race conditions.

#### Solution
* Make threading and event dispatch 'external' to the SDK core, with that being decided by the application that is embedding the SDK within a larger whole.
* SDK itself should rely on a fairly narrow 'RuntimeDispatch' interface that is responsible for managing posting and addition of event handlers (this still needs analysis), as well as for scheduling timers.
* Still have the various PlatformManager implementations as 'cookie cutter' solutions that provide implementations of the above 'RuntimeDispatch' interface (or its equivalent).
* Applications can choose to instantiate one of these impls if they choose to, or bring their own.
* Use `chip-tool` as a way to showcase the different possible threading strategies an application can use, serving as a reference of sorts for developers to see how they can construct their own. The strategies include (but are not limited to):
   * Separable threads of execution for the application and the chip stack, with locks as the basis for achieving serialization
   * Singular thread of execution for both the application and the chip stack, all running in the same dispatch loop
   * Separable threads of execution for the application and the chip stack, with the application always making calls into the stack through the threading context of the stack, either asynchronously or synchronously. 

#### Tactical
1. Design out the 'RuntimeDispatch' interface and put it up for review.
2. Explore pivoting the existing PlatformManager and association implementations away from CRTP
3. Figure out how to make it possible for the application to select a particular 'stock' PlatformManager impl, or bring their own.
4. Modify chip-tool to showcase the above three strategies, perhaps with POSIX and separably, with Darwin GCD.

gerickson, 2021-06-17:
Haven't gotten far enough into it; however, as noted in the #6561 review, I am working towards porting these CoreFoundation / CFNetwork libraries and iOS apps to CHIP:

- [Open HLX](http://github.com/gerickson/openhlx)
- [Open HLX for iOS](http://github.com/gerickson/openhlx-ios)

In theory, I should need only the singular main CFRunLoop with no further threading to make these/this work.

As noted in #6561, I should be able to import the InetLayer sockets into CFSocketStreams, schedule them onto the run loop, and go.

mrjerryjohns, 2021-06-17:
That should definitely be possible with the model being suggested here.

kpschoedel, 2021-06-22:
Tagging #7818 for inspection

kpschoedel, 2021-06-22:
Tagging #7803

closed 2022-09-18 by stale[bot]