project-chip/connectedhomeip #7742: iOS chiptool right now can't onboard the "on network" accessory

Kevin552270, 2021-06-18:
#### Problem
We have an accessory(without BLE) already connected to home Wii router(via WiFi ). then we put the iOS chiptool under same WiFi router, try to commission the accessory by using "onnetwork" method. 
 * expectation: iOS chiptool should be able to onboard the accory, then control it 
 * actual behavior: iOS chiptool can't onboard this accessory, and nothing was send out from both sides.
 For more details of setup you can find in attached pic.
 >

 #### Proposed Solution
iOS chiptool need to add related implementation for "onnetwork" accessory discovery and commissioning.

[image]

Kevin552270, 2021-06-18:
@pan-apple hello Pankaj, i found i have no acees to submit a PR, so i have to create a issue here, hope you can convert it as a PR, and it's an issue from Schneider Electric

woody-apple, 2022-01-26:
@Kevin552270 Thanks for the issue. Anyone can post a PR :) I'll tag this issue however in the interim.

turon, 2022-02-01:
This is an older issue.  Commissioning over IP on Darwin is required for v1.
I am certain support for commissioning over IP on darwin generally has been added to chip-tool since this issue was filed.
I have validated locally on *Mac* using `./out/debug/standalone/chip-tool pairing onnetwork 1 34567890`.
What I have not validated is that this feature works on iOS.
iOS onnetwork commissioning is required for v1.
Removing tag `v1_triage_split_4`.

woody-apple, 2022-02-01:
This feature is complete and working on iOS.

closed 2022-02-01 by woody-apple