project-chip/connectedhomeip #7745: Add ESP32C3 support for the other apps

wqx6, 2021-06-18:
#### Problem
#6345 add ESP32C3 support for all-clusters app, the other apps' support need to be added.

 #### Proposed Solution
A PR will be raised soon to support ESP32C3 for the other apps.

klem-everywhere, 2021-06-18:
@wqx6 have you tested the regular ESP32 with your new version?
Can you test if https://github.com/project-chip/connectedhomeip/pull/7749 fixed the other issues for you?

568760310, 2021-06-22:
@wqx6 
Hi,Qixiang
Does ESP have a plan to support the bridge-app demo?

limic-amazon, 2021-11-12:
Hello, is there any plan to pick up this work soon? My team is interested in experimenting with the temperature-measurement-app as well based on the ESP blog post: https://blog.espressif.com/announcing-matter-previously-chip-on-esp32-84164316c0e3#6184

chshu, 2021-11-12:
We will prioritize this task, will add other apps support for ESP32C3 in following 2 weeks.

closed 2021-11-23 by wqx6

wqx6, 2021-11-23:
@limic-amazon The ESP32C3 support has already added to other examples now, you can try the temperature-measurement-app on C3 now.
Close the issue.