project-chip/connectedhomeip #7827: chip-tool PASESession timed out, fails to commission device

cjandhyala, 2021-06-22:
With ToT (92fc1fb86625250bc59b47b2dae1bd45e65b8163) chip-tool "PASESession timed out"

-------- Chip tool log ---------------------
[1624394756638] [0x672671] CHIP: [DL] _Init
[1624394756638] [0x672671] CHIP: [IN] TransportMgr initialized
[1624394756638] [0x672671] CHIP: [DIS] Init admin pairing table with server storage
[1624394756638] [0x672671] CHIP: [IN] local node id is 0x000000000001B669
[1624394756638] [0x672671] CHIP: [ZCL] Using ZAP configuration...
[1624394756638] [0x672671] CHIP: [ZCL] deactivate report event
[1624394756638] [0x672671] CHIP: [CTL] Getting operational keys
[1624394756639] [0x672671] CHIP: [CTL] Generating operational certificate for the controller
[1624394756639] [0x672671] CHIP: [CTL] Getting root certificate for the controller from the issuer
[1624394756640] [0x672671] CHIP: [CTL] Generating credentials
[1624394756640] [0x672671] CHIP: [CTL] Loaded credentials successfully
[1624394756640] [0x672671] CHIP: [BLE] NewConnection
[1624394756642] [0x672671] CHIP: [SC] Assigned local session key ID 0
[1624394756642] [0x672671] CHIP: [IN] Message appended to BLE send queue
[1624394756642] [0x672671] CHIP: [SC] Sent PBKDF param request
[1624394756698] [0x672676] CHIP: [BLE] CBManagerState: ON
[1624394759904] [0x672675] CHIP: [BLE] Connecting to device with discriminator: 3840
[1624394762975] [0x672675] CHIP: [IN] BleConnectionComplete: endPoint 0x10c0f7428
[1624394762975] [0x672675] CHIP: [BLE] Ble endpoint pool FULL
[1624394762975] [0x672675] CHIP: [IN] Clearing BLE pending packets.
[1624394762975] [0x672675] CHIP: [IN] BleConnection Error: Ble Error 6002 (0x00001772): No more BLE endpoints
[1624394763829] [0x672675] CHIP: [BLE] subscribe complete, ep = 0x10c0f7428
[1624394763877] [0x672675] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1624394763877] [0x672675] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1624394763877] [0x672675] CHIP: [BLE] local and remote recv window size = 3
[1624394763877] [0x672675] CHIP: [IN] BLE EndPoint 0x10c0f7428 Connection Complete
[1624394786645] [0x672675] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1624394786645] [0x672675] CHIP: [SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1624394786645] [0x672675] CHIP: [TOO] Secure Pairing Failed
[1624394786645] [0x672671] CHIP: [TOO] Run command failure: CHIP Error 4172 (0x0000104C): Internal error
[1624394786646] [0x672675] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1624394786646] [0x672671] CHIP: [CTL] Shutting down the commissioner
[1624394786646] [0x672671] CHIP: [CTL] Shutting down the controller
[1624394786646] [0x672671] CHIP: [DL] Inet Layer shutdown
[1624394786646] [0x672671] CHIP: [DL] BLE layer shutdown
[1624394786646] [0x672671] CHIP: [IN] Clearing BLE pending packets.
[1624394786646] [0x672671] CHIP: [BLE] Auto-closing end point's BLE connection.
[1624394786646] [0x672671] CHIP: [DL] System Layer shutdown
chaitanyajandhyala@MacBook-Pro chip-tool %

bzbarsky-apple, 2021-06-22:
What does the device side log look like?  

That said, this bit:

```
[1624394762975] [0x672675] CHIP: [BLE] Ble endpoint pool FULL
[1624394762975] [0x672675] CHIP: [IN] Clearing BLE pending packets.
[1624394762975] [0x672675] CHIP: [IN] BleConnection Error: Ble Error 6002 (0x00001772): No more BLE endpoints
```

might be relevant.  If this is reproducible reliably, please check whether https://github.com/project-chip/connectedhomeip/pull/7806 fixes things?

cjandhyala, 2021-06-23:
#7806 fixes the BLE endpoint Full issue and moves fwd with other parts of provisioning flow. but EnableNetwork command seem to fail

[1624408165247] [0x6b003e] CHIP: [DL] _Init
[1624408165247] [0x6b003e] CHIP: [IN] TransportMgr initialized
[1624408165247] [0x6b003e] CHIP: [DIS] Init admin pairing table with server storage
[1624408165248] [0x6b003e] CHIP: [IN] Loading certs from KVS
[1624408165248] [0x6b003e] CHIP: [IN] local node id is 0x000000000001B669
[1624408165248] [0x6b003e] CHIP: [ZCL] Using ZAP configuration...
[1624408165248] [0x6b003e] CHIP: [ZCL] deactivate report event
[1624408165248] [0x6b003e] CHIP: [CTL] Getting operational keys
[1624408165248] [0x6b003e] CHIP: [CTL] Generating credentials
[1624408165248] [0x6b003e] CHIP: [CTL] Loaded credentials successfully
[1624408165248] [0x6b003e] CHIP: [BLE] NewConnection
[1624408165249] [0x6b003e] CHIP: [SC] Assigned local session key ID 1
[1624408165249] [0x6b003e] CHIP: [IN] Message appended to BLE send queue
[1624408165249] [0x6b003e] CHIP: [SC] Sent PBKDF param request
[1624408165303] [0x6b0041] CHIP: [BLE] CBManagerState: ON
[1624408179177] [0x6b00dd] CHIP: [BLE] Connecting to device with discriminator: 3840
[1624408182229] [0x6b00dd] CHIP: [IN] BleConnectionComplete: endPoint 0x10cd180c8
[1624408183083] [0x6b00dd] CHIP: [BLE] subscribe complete, ep = 0x10cd180c8
[1624408183131] [0x6b0044] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1624408183131] [0x6b0044] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1624408183131] [0x6b0044] CHIP: [BLE] local and remote recv window size = 3
[1624408183131] [0x6b0044] CHIP: [IN] BLE EndPoint 0x10cd180c8 Connection Complete
[1624408183759] [0x6b00dd] CHIP: [EM] Received message of type 33 and protocolId 0 on exchange 44249
[1624408183759] [0x6b00dd] CHIP: [SC] Received PBKDF param response
[1624408183762] [0x6b00dd] CHIP: [SC] Sent spake2p msg1
[1624408184660] [0x6b00dd] CHIP: [EM] Received message of type 35 and protocolId 0 on exchange 44249
[1624408184660] [0x6b00dd] CHIP: [SC] Received spake2p msg2
[1624408184660] [0x6b00dd] CHIP: [SC] Peer assigned session key ID 0
[1624408184660] [0x6b00dd] CHIP: [SC] Sent spake2p msg3
[1624408184660] [0x6b00dd] CHIP: [IN] New pairing for device 0x0000000000bc5c01, key 0!!
[1624408184660] [0x6b00dd] CHIP: [CTL] Remote device completed SPAKE2+ handshake

[1624408184660] [0x6b00dd] CHIP: [CTL] Sending root certificate to the device
[1624408184660] [0x6b00dd] CHIP: [DMG] ICR moving to [Initialize]
[1624408184660] [0x6b00dd] CHIP: [DMG] ICR moving to [AddCommand]
[1624408184661] [0x6b00dd] CHIP: [IN] Secure message was encrypted: Msg ID 0
[1624408184661] [0x6b00dd] CHIP: [IN] Encrypted message 0x70000e6dbc60 from 0x000000000001B669 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 44250.
[1624408184661] [0x6b00dd] CHIP: [IN] Sending msg 0x70000e6dbc60 to 0x0000000000BC5C01 at utc time: 552368815 msec
[1624408184661] [0x6b00dd] CHIP: [IN] Sending secure msg on generic transport
[1624408184661] [0x6b00dd] CHIP: [IN] Secure msg send status No Error
[1624408184661] [0x6b00dd] CHIP: [DMG] ICR moving to [   Sending]
[1624408184661] [0x6b00dd] CHIP: [CTL] Sent root certificate to the device
[1624408185785] [0x6b00dd] CHIP: [IN] Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1624408185785] [0x6b00dd] CHIP: [EM] Received message of type 9 and protocolId 5 on exchange 44250
[1624408185785] [0x6b00dd] CHIP: [DMG] InvokeCommand =
[1624408185785] [0x6b00dd] CHIP: [DMG] {
[1624408185785] [0x6b00dd] CHIP: [DMG] 	CommandList =
[1624408185785] [0x6b00dd] CHIP: [DMG] 	[
[1624408185785] [0x6b00dd] CHIP: [DMG] 		CommandDataElement =
[1624408185785] [0x6b00dd] CHIP: [DMG] 		{
[1624408185785] [0x6b00dd] CHIP: [DMG] 			CommandPath =
[1624408185785] [0x6b00dd] CHIP: [DMG] 			{
[1624408185785] [0x6b00dd] CHIP: [DMG] 				EndpointId = 0x0,
[1624408185785] [0x6b00dd] CHIP: [DMG] 				ClusterId = 0x3e,
[1624408185785] [0x6b00dd] CHIP: [DMG] 				CommandId = 0xa1,
[1624408185785] [0x6b00dd] CHIP: [DMG] 			},
[1624408185785] [0x6b00dd] CHIP: [DMG] 			
[1624408185785] [0x6b00dd] CHIP: [DMG] 			StatusElement =
[1624408185785] [0x6b00dd] CHIP: [DMG] 			{
[1624408185785] [0x6b00dd] CHIP: [DMG] 				GeneralCode = 0x0,
[1624408185785] [0x6b00dd] CHIP: [DMG] 				ProtocolId = 0x5,
[1624408185785] [0x6b00dd] CHIP: [DMG] 				protocolCode = 0x0,
[1624408185785] [0x6b00dd] CHIP: [DMG] 			},
[1624408185785] [0x6b00dd] CHIP: [DMG] 			
[1624408185785] [0x6b00dd] CHIP: [DMG] 		},
[1624408185785] [0x6b00dd] CHIP: [DMG] 		
[1624408185785] [0x6b00dd] CHIP: [DMG] 	],
[1624408185785] [0x6b00dd] CHIP: [DMG] 	
[... 210 lines cut ...]
[1624408190.291369][1520283] CHIP:DL: wpa_supplicant: failed to add network: Partyguys: GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: invalid message format
[1624408190.291456][1520283] CHIP:NP: Failed to connect to WiFi network: CHIP Error 4172 (0x0000104C): Internal error
[1624408190.291517][1520283] CHIP:DMG: ICR moving to [AddCommand]
[1624408190.291658][1520283] CHIP:IN: Secure message was encrypted: Msg ID 5
[1624408190.291709][1520283] CHIP:IN: Encrypted message 0xffffd1889690 from 0x0000000000BC5C01 to 0x000000000001B669 of type 9 and protocolId 5 on exchange 44254.
[1624408190.291760][1520283] CHIP:IN: Sending msg 0xffffd1889690 to 0x000000000001B669 at utc time: 720929152 msec
[1624408190.291800][1520283] CHIP:IN: Sending secure msg on generic transport
[... cut ...]
[1624408190.822746][1520286] CHIP:BLE: Indication confirmation, 0xffffac044e10
[1624408190.957603][1520286] CHIP:BLE: Indication confirmation, 0xffffac044e10
[1624408191.274139][1520286] CHIP:DL: Bluez disconnected
[1624408191.274215][1520286] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1624408191.274413][1520283] CHIP:IN: Clearing BLE pending packets.
[1624408191.274571][1520286] CHIP:BLE: Indication confirmation, (nil)
[1624408191.275036][1520283] CHIP:BLE: no endpoint for BLE sent data ack
^C

bzbarsky-apple, 2021-06-23:
Hard to say what's going on there without more details about what exact commands are being run and whatnot.

cjandhyala, 2021-06-24:
@bzbarsky-apple do you need any additional info ?

bzbarsky-apple, 2021-06-24:
Things like knowing what device is being commissioned might have been useful.  That said, device log says:
```
 [1624408190.291369][1520283] CHIP:DL: wpa_supplicant: failed to add network: Partyguys: GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: invalid message format
```
Is the device being commissioned a Linux device?

If so, it's worth checking whether the right data is landing inside `src/platform/Linux/ConnectivityManagerImpl.cpp` at `ConnectivityManagerImpl::ProvisionWiFiNetwork`.  Past that, @yufengwangca any ideas what might be going on here?

cjandhyala, 2021-06-24:
@bzbarsky-apple the device is a raspi (running ubuntu server OS).

yufengwangca, 2021-06-29:
From my test, I am able to provision the rp4 to my home network using chip-shell. When I try to provision the rp4 using chip-tool from mac w/o #7806, I observed different errors:

1. On the mac side
yufengs-mbp:host yufengw$ ./chip-tool pairing ble-wifi test-2g 12345678 12344321 20202021 3840
CHIP: [ZCL] Using ZAP configuration...
CHIP: [ZCL] deactivate report event
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 0x000000000001b669
CHIP: [ZCL] Using ZAP configuration...
CHIP: [ZCL] deactivate report event
CHIP: [DL] CHIP task running
CHIP: [BLE] NewConnection
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] Warning: OnBleConnectionComplete is called multiple times for the same peripheral.
CHIP: [BLE] subscribe complete, ep = 0x1049caf98
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [BLE] Assigned local session key ID 0
CHIP: [BLE] SessionEstablishmentExchangeDispatch::SendMessageImpl mBLETransport 0x7fcb17708b90, mTransportMgr 0x7fcb17708a60
CHIP: [BLE] Sending message using BLE transport
CHIP: [BLE] Sent PBKDF param request
CHIP: [BLE] Received PBKDF param response
CHIP: [BLE] SessionEstablishmentExchangeDispatch::SendMessageImpl mBLETransport 0x7fcb17708b90, mTransportMgr 0x7fcb17708a60
CHIP: [BLE] Sending message using BLE transport
CHIP: [BLE] Sent spake2p msg1
CHIP: [BLE] Received spake2p msg2
CHIP: [BLE] Peer assigned session key ID 0
CHIP: [BLE] SessionEstablishmentExchangeDispatch::SendMessageImpl mBLETransport 0x7fcb17708b90, mTransportMgr 0x7fcb17708a60
CHIP: [BLE] Sending message using BLE transport
CHIP: [BLE] Sent spake2p msg3
CHIP: [IN] New pairing for device 0x0000000000bc5c01, key 0!!
CHIP: [CTL] Remote device completed SPAKE2+ handshake

CHIP: [TOO] OnNetworkCredentialsRequested
CHIP: [NP] Sending Network Creds

CHIP: [IN] Secure message was encrypted: Msg ID 0
CHIP: [IN] Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 2127318792 msec
CHIP: [IN] Sending secure msg on connection specific transport
CHIP: [IN] Secure msg send status No Error
CHIP: [TOO] Secure Pairing Success
CHIP: [BLE] ack recv timeout, closing ep 0x1049caf98
CHIP: [CTL] Remote device failed in network provisioning

CHIP: [TOO] Network Provisioning Failed
CHIP: [BLE] CancelConnection
CHIP: [TOO] Pairing Failure: Ble Error 6024 (0x00001788): Message fragment acknowledgement timed out
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller
CHIP: [DL] System Layer shutdown
CHIP: [DL] Inet Layer shutdown
CHIP: [IN] Async DNS worker thread woke up.
CHIP: [IN] Async DNS worker thread exiting.
CHIP: [IN] Async DNS worker thread woke up.

2. On RP4 side
CHIP:DL: Device 8C:85:90:58:49:03 (Path: /org/bluez/hci0/dev_8C_85_90_58_49_03) Connected
CHIP:DL: BluezCharacteristicAcquireWrite is called, conn: 0xffffa002fcc0
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa002fcc0
CHIP:BLE: local and remote recv window sizes = 3
CHIP:BLE: selected BTP version 3
CHIP:BLE: using BTP fragment sizes rx 128 / tx 20.
CHIP:DL: CHIPoBLE subscribe received
CHIP:DL: HandlePlatformSpecificBLEEvent 32777
CHIP:DL: Receive kCHIPoBLEConnectionEstablished
CHIP:BLE: Indication confirmation, 0xffffa002fcc0
[... 50 lines cut ...]

CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: test-2g

(process:20763): GLib-GIO-CRITICAL **: 18:30:34.896: g_dbus_proxy_call_sync_internal: assertion 'G_IS_DBUS_PROXY (proxy)' failed
CHIP:DL: wpa_supplicant: failed to add network: test-2g: unknown error
CHIP:NP: Failed to connect to WiFi network: CHIP Error 4172 (0x0000104C): Internal error
CHIP:NP: Failed in HandleNetworkProvisioningMessage. error CHIP Error 4172 (0x0000104C): Internal error

CHIP:SVR: OnRendezvousStatusUpdate: CHIP Error 4172 (0x0000104C): Internal error
CHIP:SVR: Persisting admin ID 0, next available 1
CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
CHIP:BLE: Releasing end point's BLE connection back to application.
CHIP:BLE: Got notification regarding chip connection closure
CHIP:SVR: OnRendezvousError: CHIP Error 4172 (0x0000104C): Internal error
CHIP:DL: UnregisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16395
CHIP:DL: CHIPoBLE advertising stopped
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa002fcc0
CHIP:BLE: Releasing end point's BLE connection back to application.
CHIP:BLE: Got notification regarding chip connection closure
CHIP:BLE: HandleChipConnectionReceived failed, err = 6018
CHIP:BLE: failed handle new chip BLE connection, status = 6018
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected

cjandhyala, 2021-06-29:
Test Setp Details:
Accessory : Lighting App on Raspi running Ubutun Server OS 20.10 , App built with 'scripts/examples/gn_build_example.sh examples/lighting-app/linux out/lighting-app' . App launched with cmd ' sudo ./out/lighting-app/chip-lighting-app --wifi
'

Controller : Chip-tool on Mac , built with instructions https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool , launch chip tool with cmd ' ./out/debug/chip-tool pairing ble-wifi <MYSSID> <MYPASSWD> 12344321 20202021 3840

With ToT (129c67e8e7b60a959aa3b781d45c523635347196) on both chip-tool and lighting app, I see the  Ble endpoint pool FULL error followed by PASE session timeout with BLE

Mac log 
--------
% ./out/debug/chip-tool pairing ble-wifi <MYSSID> <MYPASSWD> 12344321 20202021 3840
[1624993699573] [0x23b659] CHIP: [DL] _Init
[1624993699573] [0x23b659] CHIP: [IN] TransportMgr initialized
[1624993699573] [0x23b659] CHIP: [DIS] Init admin pairing table with server storage
[1624993699573] [0x23b659] CHIP: [IN] Loading certs from KVS
[1624993699573] [0x23b659] CHIP: [IN] local node id is 0x000000000001B669
[1624993699573] [0x23b659] CHIP: [ZCL] Using ZAP configuration...
[1624993699574] [0x23b659] CHIP: [ZCL] deactivate report event
[1624993699574] [0x23b659] CHIP: [CTL] Getting operational keys
[1624993699574] [0x23b659] CHIP: [CTL] Generating credentials
[1624993699574] [0x23b659] CHIP: [CTL] Loaded credentials successfully
[1624993699574] [0x23b65c] CHIP: [BLE] NewConnection
[1624993699576] [0x23b65c] CHIP: [SC] Assigned local session key ID 1
[1624993699577] [0x23b65c] CHIP: [IN] Message appended to BLE send queue
[1624993699577] [0x23b65c] CHIP: [SC] Sent PBKDF param request
[1624993703450] [0x23b65c] CHIP: [BLE] CBManagerState: ON
[1624993713092] [0x23b65c] CHIP: [BLE] Connecting to device with discriminator: 3840
[1624993715964] [0x23b65c] CHIP: [IN] BleConnectionComplete: endPoint 0x10d9827d8
[1624993715964] [0x23b65c] CHIP: [BLE] Ble endpoint pool FULL
[1624993715964] [0x23b65c] CHIP: [IN] Clearing BLE pending packets.
[1624993715964] [0x23b65c] CHIP: [IN] BleConnection Error: Ble Error 6002 (0x00001772): No more BLE endpoints
[1624993716233] [0x23b65d] CHIP: [BLE] subscribe complete, ep = 0x10d9827d8
[1624993716236] [0x23b65c] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1624993716236] [0x23b65c] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1624993716236] [0x23b65c] CHIP: [BLE] local and remote recv window size = 3
[1624993716236] [0x23b65c] CHIP: [IN] BLE EndPoint 0x10d9827d8 Connection Complete
[1624993729578] [0x23b65d] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1624993729578] [0x23b65d] CHIP: [SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1624993729578] [0x23b65d] CHIP: [TOO] Secure Pairing Failed
[1624993729578] [0x23b659] CHIP: [TOO] Run command failure: CHIP Error 4050 (0x00000FD2): Timeout
[1624993729578] [0x23b65d] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1624993729578] [0x23b659] CHIP: [CTL] Shutting down the commissioner
[1624993729578] [0x23b659] CHIP: [CTL] Shutting down the controller
[1624993729578] [0x23b659] CHIP: [DL] Inet Layer shutdown
[1624993729578] [0x23b659] CHIP: [DL] BLE layer shutdown
[1624993729578] [0x23b659] CHIP: [IN] Clearing BLE pending packets.
[1624993729578] [0x23b659] CHIP: [BLE] Auto-closing end point's BLE connection.
[1624993729578] [0x23b659] CHIP: [DL] System Layer shutdown

Device side log
----------------

$ sudo ./out/lighting-app/chip-lighting-app --wifi
[1624993695.183161][28913] CHIP:DL: Device Configuration:
[1624993695.183215][28913] CHIP:DL:   Serial Number: TEST_SN
[1624993695.183240][28913] CHIP:DL:   Vendor Id: 9050 (0x235A)
[1624993695.183257][28913] CHIP:DL:   Product Id: 65279 (0xFEFF)
[1624993695.183277][28913] CHIP:DL:   Product Revision: 1
[1624993695.183296][28913] CHIP:DL:   Setup Pin Code: 20202021
[1624993695.183314][28913] CHIP:DL:   Setup Discriminator: 3840 (0xF00)
[1624993695.183333][28913] CHIP:DL:   Manufacturing Date: (not set)
[1624993695.183350][28913] CHIP:DL:   Device Type: 257 (0x101)
[1624993695.183414][28913] CHIP:SVR: SetupQRCode: [MT:YNJV7VSC00KA0648G00]
[1624993695.183438][28913] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1624993695.183452][28913] CHIP:SVR: https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3AYNJV7VSC00KA0648G00
[1624993695.183480][28913] CHIP:SVR: Manual pairing code: [34970112332]
[1624993695.183924][28913] CHIP:ZCL: Using ZAP configuration...
[1624993695.184037][28913] CHIP:ZCL: deactivate report event
[1624993695.184057][28913] CHIP:ZCL: OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pairing table.
[1624993695.184072][28913] CHIP:DIS: Set the admin pairing table delegate
[... 922 lines cut ...]
[1624995154.411514][28953] CHIP:DIS: Broadcasting mDns reply
[1624995234.732741][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995234.732810][28953] CHIP:DIS: Broadcasting mDns reply
[1624995252.969374][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995252.969440][28953] CHIP:DIS: Broadcasting mDns reply
[1624995279.920184][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995279.920246][28953] CHIP:DIS: Broadcasting mDns reply
[1624995350.157802][28953] CHIP:DIS: Replying to DNS-SD service listing request
[... cut ...]
[1624995486.669350][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995486.669416][28953] CHIP:DIS: Broadcasting mDns reply
[1624995505.236254][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995505.236322][28953] CHIP:DIS: Broadcasting mDns reply
[1624995515.372603][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995515.372673][28953] CHIP:DIS: Broadcasting mDns reply
[1624995521.336352][28953] CHIP:DIS: Replying to DNS-SD service listing request
[1624995521.336416][28953] CHIP:DIS: Broadcasting mDns reply

yufengwangca, 2021-06-29:
@cjandhyala  Thanks for update, I re-run my test with --wifi, and I can confirm #7806 did fix the BLE endpoint Full issue. With #7806, I can provision my RP4 successfully. 

1. On MacPro side.
```
yufengs-mbp:host yufengw$ ./chip-tool pairing ble-wifi test-2g 12345678 12344321 20202021 3840
[1624996212353] [0x48e58e] CHIP: [DL] _Init
[1624996212353] [0x48e58e] CHIP: [IN] TransportMgr initialized
[1624996212353] [0x48e58e] CHIP: [DIS] Init admin pairing table with server storage
[1624996212353] [0x48e58e] CHIP: [IN] Loading certs from KVS
[1624996212353] [0x48e58e] CHIP: [IN] local node id is 0x000000000001B669
[1624996212354] [0x48e58e] CHIP: [ZCL] Using ZAP configuration...
[1624996212354] [0x48e58e] CHIP: [ZCL] deactivate report event
[1624996212354] [0x48e58e] CHIP: [CTL] Getting operational keys
[1624996212354] [0x48e58e] CHIP: [CTL] Generating credentials
[1624996212354] [0x48e58e] CHIP: [CTL] Loaded credentials successfully
[1624996212354] [0x48e5a4] CHIP: [BLE] NewConnection
[1624996212356] [0x48e5a4] CHIP: [SC] Assigned local session key ID 9
[1624996212356] [0x48e5a4] CHIP: [IN] Message appended to BLE send queue
[1624996212356] [0x48e5a4] CHIP: [SC] Sent PBKDF param request
[1624996212404] [0x48e5a6] CHIP: [BLE] CBManagerState: ON
[1624996217452] [0x48e5a4] CHIP: [BLE] Connecting to device with discriminator: 3840
[1624996230798] [0x48e69a] CHIP: [IN] BleConnectionComplete: endPoint 0x106c998e8
[1624996231652] [0x48e5a7] CHIP: [BLE] subscribe complete, ep = 0x106c998e8
[1624996231698] [0x48e69a] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1624996231698] [0x48e69a] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1624996231698] [0x48e69a] CHIP: [BLE] local and remote recv window size = 3
[1624996231698] [0x48e69a] CHIP: [IN] BLE EndPoint 0x106c998e8 Connection Complete
[1624996232418] [0x48e5a7] CHIP: [EM] Received message of type 33 and protocolId 0 on exchange 44249
[1624996232418] [0x48e5a7] CHIP: [SC] Received PBKDF param response
[1624996232419] [0x48e5a7] CHIP: [SC] Sent spake2p msg1
[1624996233588] [0x48e69a] CHIP: [EM] Received message of type 35 and protocolId 0 on exchange 44249
[1624996233588] [0x48e69a] CHIP: [SC] Received spake2p msg2
[1624996233588] [0x48e69a] CHIP: [SC] Peer assigned session key ID 0
[1624996233588] [0x48e69a] CHIP: [SC] Sent spake2p msg3
[1624996233588] [0x48e69a] CHIP: [IN] New pairing for device 0x0000000000bc5c01, key 0!!
[1624996233588] [0x48e69a] CHIP: [CTL] Remote device completed SPAKE2+ handshake
[1624996233588] [0x48e69a] CHIP: [CTL] Sending root certificate to the device
[1624996233588] [0x48e69a] CHIP: [DMG] ICR moving to [Initialize]
[1624996233588] [0x48e69a] CHIP: [DMG] ICR moving to [AddCommand]
[1624996233588] [0x48e69a] CHIP: [IN] Secure message was encrypted: Msg ID 0
[1624996233588] [0x48e69a] CHIP: [IN] Encrypted message 0x70000ea6bc30 from 0x000000000001B669 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 44250.
[1624996233588] [0x48e69a] CHIP: [IN] Sending msg 0x70000ea6bc30 to 0x0000000000BC5C01 at utc time: 2132121536 msec
[1624996233588] [0x48e69a] CHIP: [IN] Sending secure msg on generic transport
[1624996233588] [0x48e69a] CHIP: [IN] Secure msg send status No Error
[1624996233588] [0x48e69a] CHIP: [DMG] ICR moving to [   Sending]
[1624996233588] [0x48e69a] CHIP: [CTL] Sent root certificate to the device
[1624996234893] [0x48e5a7] CHIP: [IN] Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 9
[1624996234893] [0x48e5a7] CHIP: [IN] Setting fabricID BC5C01 on admin.
[1624996234893] [0x48e5a7] CHIP: [IN] Since admin was modified, persisting changes to KVS
[1624996234894] [0x48e5a7] CHIP: [EM] Received message of type 9 and protocolId 5 on exchange 44250
[1624996234894] [0x48e5a7] CHIP: [DMG] InvokeCommand =
[1624996234894] [0x48e5a7] CHIP: [DMG] {
[1624996234894] [0x48e5a7] CHIP: [DMG] 	CommandList =
[1624996234894] [0x48e5a7] CHIP: [DMG] 	[
[1624996234894] [0x48e5a7] CHIP: [DMG] 		CommandDataElement =
[1624996234894] [0x48e5a7] CHIP: [DMG] 		{
[1624996234894] [0x48e5a7] CHIP: [DMG] 			CommandPath =
[1624996234894] [0x48e5a7] CHIP: [DMG] 			{
[1624996234894] [0x48e5a7] CHIP: [DMG] 				EndpointId = 0x0,
[1624996234894] [0x48e5a7] CHIP: [DMG] 				ClusterId = 0x3e,
[1624996234894] [0x48e5a7] CHIP: [DMG] 				CommandId = 0xa1,
[1624996234894] [0x48e5a7] CHIP: [DMG] 			},
[1624996234894] [0x48e5a7] CHIP: [DMG] 			
[1624996234894] [0x48e5a7] CHIP: [DMG] 			StatusElement =
[1624996234894] [0x48e5a7] CHIP: [DMG] 			{
[1624996234894] [0x48e5a7] CHIP: [DMG] 				GeneralCode = 0x0,
[1624996234894] [0x48e5a7] CHIP: [DMG] 				ProtocolId = 0x5,
[1624996234894] [0x48e5a7] CHIP: [DMG] 				protocolCode = 0x0,
[1624996234894] [0x48e5a7] CHIP: [DMG] 			},
[1624996234894] [0x48e5a7] CHIP: [DMG] 			
[... 593 lines cut ...]
[1624996244.855355][22711] CHIP:DMG: 			{
[1624996244.855417][22711] CHIP:DMG: 				EndpointId = 0x0,
[1624996244.855485][22711] CHIP:DMG: 				ClusterId = 0x31,
[1624996244.855550][22711] CHIP:DMG: 				CommandId = 0xc,
[1624996244.855619][22711] CHIP:DMG: 			},
[1624996244.855674][22711] CHIP:DMG: 			
[1624996244.855719][22711] CHIP:DMG: 			CommandData = 
[1624996244.855768][22711] CHIP:DMG: 			{
[1624996244.855819][22711] CHIP:DMG: 				0x0 = [
[1624996244.855877][22711] CHIP:DMG: 					0x67, 0x61, 0x6f, 0x72, 0x75, 0x69, 0x2d, 0x32, 0x67, 
[1624996244.855934][22711] CHIP:DMG: 			]
[1624996244.855985][22711] CHIP:DMG: 				0x1 = 0, 
[... cut ...]
[1624996244.856343][22711] CHIP:DMG: }
[1624996244.856414][22711] CHIP:ZCL: Received Cluster Command: Cluster=31 Command=c Endpoint=0
[1624996244.856476][22711] CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: test-2g
[1624996244.874600][22711] CHIP:DL: wpa_supplicant: added network: SSID: test-2g: /fi/w1/wpa_supplicant1/Interfaces/0/Networks/2
[1624996244.877436][22711] CHIP:DL: wpa_supplicant: connected to network: SSID: test-2g
[1624996244.904903][22711] CHIP:DL: wpa_supplicant: save config succeeded!
```

yufengwangca, 2021-06-29:
@cjandhyala 
It looks your wap_supplicant on your raspi failed to recognize the DBUS message from lighting-app. What is your version? Here is mine.

<pre>wpa_supplicant -v
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen &lt;j@w1.fi&gt; and contributors
</pre>

cjandhyala, 2021-06-29:
@yufengwangca , the same raspi works fine when provisioning with python controller. 

$ wpa_supplicant -v
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors

yufengwangca, 2021-06-29:
Conducted online joint debugging session with @cjandhyala,  and we found the the passphrase after the special character '$' is lost from chip-tool, after we change the wifi password to another one w/o any special character, we can connect to the network as expected.

bzbarsky-apple, 2021-06-30:
> and we found the the passphrase after the special character '$' is lost from chip-tool

Ah, because the shell interpreted the `$` as a variable reference?  So using either single-quotes or the hex representation would work too?

yufengwangca, 2021-06-30:
I can provision the device with chip-shell on the device or python controller from laptop, but when I use chip-tool to provision the device from mac, the original password "1234$1234 " is received as "1234234"

[1625033720.835327][6853] CHIP:ZCL: emberAfNetworkCommissioningClusterAddWiFiNetworkCallback
[1625033720.835378][6853] CHIP:ZCL: WiFi provisioning data: SSID: gaorui-guest
[1625033720.835422][6853] CHIP:ZCL: WiFi provisioning data: Key: 1234234

yufengwangca, 2021-06-30:
@bzbarsky-apple I think you are right, using single quotes solve the issue

./chip-tool pairing ble-wifi gaorui-guest '1234$1234' 12344321 20202021 3840

[1625035275.587147][7024] CHIP:ZCL: emberAfNetworkCommissioningClusterEnableNetworkCallback
[1625035275.587238][7024] CHIP:ZCL: DoEnableNetwork: ssid:gaorui-guest
[1625035275.587278][7024] CHIP:ZCL: DoEnableNetwork: key:1234$1234
[1625035275.587318][7024] CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: gaorui-guest
[1625035275.587359][7024] CHIP:NP: LinuxNetworkProvisioningDelegate: PASSPHRASE: 1234$1234

cjandhyala, 2021-06-30:
@yufengwangca having a single '' to the password gets past that issue of chopping password after the special chars.. 

Per our conversation, we need to debug why chip tool uses the fabricID 'bc5c01' instead of '0000' , pls ref to chip tool and device logs. chip tool running on Mac and device being raspi (with ubuntu server OS)

Mac Log
---------
% sudo ./out/debug/chip-tool pairing ble-wifi MYSSID 'MYPASSWD' 12344321 20202021 3840
Password:
[1625070608799] [0x2c85ed] CHIP: [DL] _Init
[1625070608802] [0x2c85ed] CHIP: [IN] TransportMgr initialized
[1625070608802] [0x2c85ed] CHIP: [DIS] Init admin pairing table with server storage
[1625070608802] [0x2c85ed] CHIP: [IN] Loading certs from KVS
[1625070608803] [0x2c85ed] CHIP: [IN] local node id is 0x000000000001B669
[1625070608803] [0x2c85ed] CHIP: [ZCL] Using ZAP configuration...
[1625070608803] [0x2c85ed] CHIP: [ZCL] deactivate report event
[1625070608803] [0x2c85ed] CHIP: [CTL] Getting operational keys
[1625070608803] [0x2c85ed] CHIP: [CTL] Generating credentials
[1625070608804] [0x2c85ed] CHIP: [CTL] Loaded credentials successfully
[1625070608804] [0x2c85f2] CHIP: [BLE] NewConnection
[1625070608805] [0x2c85f2] CHIP: [SC] Assigned local session key ID 19
[1625070608805] [0x2c85f2] CHIP: [IN] Message appended to BLE send queue
[1625070608805] [0x2c85f2] CHIP: [SC] Sent PBKDF param request
[1625070608857] [0x2c85f2] CHIP: [BLE] CBManagerState: ON
[1625070610663] [0x2c85f2] CHIP: [BLE] Connecting to device with discriminator: 3840
[1625070613641] [0x2c85f4] CHIP: [IN] BleConnectionComplete: endPoint 0x10caea858
[1625070614450] [0x2c85f4] CHIP: [BLE] subscribe complete, ep = 0x10caea858
[1625070614498] [0x2c85f4] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1625070614498] [0x2c85f4] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1625070614498] [0x2c85f4] CHIP: [BLE] local and remote recv window size = 3
[1625070614498] [0x2c85f4] CHIP: [IN] BLE EndPoint 0x10caea858 Connection Complete
[1625070615083] [0x2c85f3] CHIP: [EM] Received message of type 33 and protocolId 0 on exchange 44249
[1625070615083] [0x2c85f3] CHIP: [SC] Received PBKDF param response
[1625070615083] [0x2c85f3] CHIP: [SC] Sent spake2p msg1
[1625070615983] [0x2c85f4] CHIP: [EM] Received message of type 35 and protocolId 0 on exchange 44249
[1625070615983] [0x2c85f4] CHIP: [SC] Received spake2p msg2
[1625070615983] [0x2c85f4] CHIP: [SC] Peer assigned session key ID 0
[1625070615983] [0x2c85f4] CHIP: [SC] Sent spake2p msg3
[1625070615983] [0x2c85f4] CHIP: [IN] New pairing for device 0x0000000000bc5c01, key 0!!
[1625070615983] [0x2c85f4] CHIP: [CTL] Remote device completed SPAKE2+ handshake
[1625070615983] [0x2c85f4] CHIP: [CTL] Sending root certificate to the device
[1625070615984] [0x2c85f4] CHIP: [DMG] ICR moving to [Initialize]
[1625070615984] [0x2c85f4] CHIP: [DMG] ICR moving to [AddCommand]
[1625070615984] [0x2c85f4] CHIP: [IN] Secure message was encrypted: Msg ID 0
[1625070615984] [0x2c85f4] CHIP: [IN] Encrypted message 0x700005240c30 from 0x000000000001B669 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 44250.
[1625070615984] [0x2c85f4] CHIP: [IN] Sending msg 0x700005240c30 to 0x0000000000BC5C01 at utc time: 408354605 msec
[1625070615984] [0x2c85f4] CHIP: [IN] Sending secure msg on generic transport
[1625070615984] [0x2c85f4] CHIP: [IN] Secure msg send status No Error
[1625070615984] [0x2c85f4] CHIP: [DMG] ICR moving to [   Sending]
[1625070615984] [0x2c85f4] CHIP: [CTL] Sent root certificate to the device
[1625070617108] [0x2c85f3] CHIP: [IN] Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 19
[1625070617108] [0x2c85f3] CHIP: [EM] Received message of type 9 and protocolId 5 on exchange 44250
[1625070617108] [0x2c85f3] CHIP: [DMG] InvokeCommand =
[1625070617108] [0x2c85f3] CHIP: [DMG] {
[1625070617108] [0x2c85f3] CHIP: [DMG] 	CommandList =
[1625070617108] [0x2c85f3] CHIP: [DMG] 	[
[1625070617108] [0x2c85f3] CHIP: [DMG] 		CommandDataElement =
[1625070617108] [0x2c85f3] CHIP: [DMG] 		{
[1625070617108] [0x2c85f3] CHIP: [DMG] 			CommandPath =
[1625070617108] [0x2c85f3] CHIP: [DMG] 			{
[1625070617108] [0x2c85f3] CHIP: [DMG] 				EndpointId = 0x0,
[1625070617108] [0x2c85f3] CHIP: [DMG] 				ClusterId = 0x3e,
[1625070617108] [0x2c85f3] CHIP: [DMG] 				CommandId = 0xa1,
[1625070617108] [0x2c85f3] CHIP: [DMG] 			},
[1625070617109] [0x2c85f3] CHIP: [DMG] 			
[1625070617109] [0x2c85f3] CHIP: [DMG] 			StatusElement =
[1625070617109] [0x2c85f3] CHIP: [DMG] 			{
[1625070617109] [0x2c85f3] CHIP: [DMG] 				GeneralCode = 0x0,
[1625070617109] [0x2c85f3] CHIP: [DMG] 				ProtocolId = 0x5,
[1625070617109] [0x2c85f3] CHIP: [DMG] 				protocolCode = 0x0,
[1625070617109] [0x2c85f3] CHIP: [DMG] 			},
[... 803 lines cut ...]
[1625070708.265381][35100] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
[1625070708.265584][35100] CHIP:BLE: Write request received debug 0xffff94046730
[1625070708.686029][35097] CHIP:DIS: Replying to DNS-SD service listing request
[1625070708.686090][35097] CHIP:DIS: Broadcasting mDns reply
[1625070710.830768][35100] CHIP:BLE: Indication confirmation, 0xffff94046730
[1625070713.305477][35100] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
[1625070713.305634][35100] CHIP:BLE: Write request received debug 0xffff94046730
[1625070715.203101][35097] CHIP:DIS: Replying to DNS-SD service listing request
[1625070715.203190][35097] CHIP:DIS: Broadcasting mDns reply
[... cut ...]
[1625070728.425913][35100] CHIP:BLE: Write request received debug 0xffff94046730
[1625070729.237662][35100] CHIP:DL: Bluez disconnected
[1625070729.237704][35100] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1625070729.237761][35097] CHIP:IN: Clearing BLE pending packets.
[1625070809.487766][35097] CHIP:DIS: Replying to DNS-SD service listing request
[1625070809.487828][35097] CHIP:DIS: Broadcasting mDns reply
[1625070910.066435][35097] CHIP:DIS: Replying to DNS-SD service listing request
[1625070910.066499][35097] CHIP:DIS: Broadcasting mDns reply

bzbarsky-apple, 2021-07-01:
> we need to debug why chip tool uses the fabricID 'bc5c01' instead of '0000'

Because the command-line told it to?

If you run `/out/debug/chip-tool pairing ble-wifi` it'll tell you what the arguments mean:
```
Usage:
  ./out/debug/standalone/chip-tool pairing ble-wifi ssid password fabric-id setup-pin-code discriminator
```
So the command above passes `12344321` as the fabric id, which is `0xbc5c01`.  If you want fabric id 0 (and you probably do for now, since the device hardcodes that), you want to do:
```
./out/debug/chip-tool pairing ble-wifi MYSSID 'MYPASSWD' 0 20202021 3840
```

cjandhyala, 2021-07-01:
you are right, but the MDNS advertisement shows fabric ID as 0 (below screenshot), But the 'PairingCommand::UpdateNetworkAddress()' API is using the fabric ID provided in our command line. 

[image]

bzbarsky-apple, 2021-07-01:
That's right.  `chip-tool` uses the fabric id you tell it to use.  This is NOT currently communicated to the device in any sort of useful way.  The device (which is what's doing the mdns advertisement) uses a fabric based on whatever is stored under `kConfigKey_FabricId` (probably nothing), with fallback to `const uint64_t TestFabricId = 0;` if `CHIP_DEVICE_CONFIG_ENABLE_TEST_DEVICE_IDENTITY` is defined.  See `GenericConfigurationManagerImpl<ImplClass>::_GetFabricId` (which is what the mdns code ends up calling in the end to get the fabric id to advertise).

The only point of the `chip-tool` argument right now is that _if_ you configure your device with a different (nonzero) fabric, you can tell `chip-tool` to use that fabric so it can find it.

We need to stand up parsing the fabric id out of the opcert and whatnot.  And advertising all our fabric/node identities.  It just hasn't happened yet.

yufengwangca, 2021-07-01:
The Fabric ID should not be the same as a Node ID, It should be a random number created by the Commissioner (Controller) that creates the network while pairing the first device. Before fabric network is created, the default fabricId is 0.

Then it's up to the controller to appropriately mint an op cert for the paired device that contains the same FabricID as that of the controller itself, and when the paired device 'joins' a fabric at the time it receives an operational cert, with the fabric ID embedded as a subject DN within that cert.

cjandhyala, 2021-07-01:
@bzbarsky-apple  changing the fabricID to 0 in the Chip-tool CMD line arg gets past the MDNS error and able to send zcl on off commands. You are right the confusion here is that chip-device-ctrl allows specifying a node id when commissioning; chip-tool does not. chip-tool allows specifying the fabric id to use, whereas chip-device-ctrl does not.

To Summarize , we got the provisioning flow with chip-tool to working with https://github.com/project-chip/connectedhomeip/pull/7806 on top of the other day's ToT and passing the fabricID argument parameter to the chip-tool same as what the device's default fabricID.