project-chip/connectedhomeip #7840: How to track the status of lock demo of NRF52840?

568760310, 2021-06-23:
#### Problem
After pairing the LOCK demo(NRF52840) using chip-device-ctrl(RP4)

1. How can I track the lock status on RP4,  if I toggle the state of the lock by press botton2 on nrf52840DK?
2. Does lock report when the state changes? 
3. I see the lock keeps sending UDP  multicast messages to ff03::1?  What is the effect of those packets, and How to parse these messages?

[image]

kkasperczyk-no, 2021-06-25:
Hi,
1. As far as I know for now you can't. In general you can read some clusters value by invoking `zclread <cluster> <attribute> <nodeid> <endpoint> <groupid>`, but ZCL OnOff cluster supports only commands controlling the state, but does not support reads that could provide you with that information.
2. After pressing the button firmware updates state of a ZCL cluster, so it contain information consistent with actual device state, but any reports are not sent.
3. Those messages are result of some workaround that was used some time ago. Few months ago we didn't have in Matter a way to perform Operational Discovery and learn device IP address on the controller side, so we just had to broadcast this address over UDP and listen to this messages on the controller side. Now we have support for Operational Discovery in Python CHIP Controller, but not so long ago we still didn't have it for mobile platforms like Android, so that workaround was still needed for some cases. Honestly I don't know the current status of mobile platform and I need to check that, as it's possible that we may don't need to publish those messages anymore.

568760310, 2021-06-28:
@kkasperczyk-no  Thanks for your reply.

I didn't get your idea for question 2.
 Do you mean for CHIP spec, when a device changes its ZCL state, it should not report the state and why do that?
 if someone subscribes to this ZCL cluster or node? How are those devices synchronizing information of the cluster state?

closed 2022-09-20 by stale[bot]