project-chip/connectedhomeip #7860: chip-tool tests TestCluster is broken after "Enable CASE session establishment (#7666)"

mspang, 2021-06-24:
#### Problem

chip-tool tests TestClusters is broken:

Initial pairing:
```
[1624494664.846274][1749867] CHIP:IN: TransportMgr initialized
[1624494664.846299][1749867] CHIP:DIS: Init admin pairing table with server storage
[1624494664.846399][1749867] CHIP:IN: Loading certs from KVS
[1624494664.846409][1749867] CHIP:IN: local node id is 0x000000000001B669
[1624494664.847036][1749867] CHIP:DL: MDNS failed to join multicast group on enx00e04c002cc7 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1624494664.847073][1749867] CHIP:ZCL: Using ZAP configuration...
[1624494664.847098][1749867] CHIP:ZCL: deactivate report event
[1624494664.847105][1749867] CHIP:CTL: Getting operational keys
[1624494664.847109][1749867] CHIP:CTL: Generating credentials
[1624494664.847223][1749867] CHIP:CTL: Loaded credentials successfully
[1624494664.847320][1749875] CHIP:DL: CHIP task running
[1624494664.847375][1749867] CHIP:SC: Assigned local session key ID 25
[1624494664.847438][1749867] CHIP:IN: Message appended to BLE send queue
[1624494664.847444][1749867] CHIP:SC: Sent PBKDF param request
[1624494664.847903][1749876] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1624494664.848022][1749871] CHIP:DL: TRACE: Bus acquired for name C-b36b
[1624494664.851782][1749875] CHIP:DL: PlatformBlueZInit init success
[1624494664.853991][1749871] CHIP:BLE: BLE removing known devices.
[1624494664.854010][1749871] CHIP:BLE: BLE initiating scan.
[1624494664.989681][1749871] CHIP:BLE: New device scanned: 67:E9:7E:06:AF:6B
[1624494664.989705][1749871] CHIP:BLE: Device discriminator match. Attempting to connect.
[1624494664.993670][1749871] CHIP:BLE: Scan complete notification without an active scan.
[1624494665.156233][1749871] CHIP:DL: ConnectDevice complete
[1624494666.137502][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service0006
[1624494666.137527][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137535][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service0006
[1624494666.137540][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137547][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service0006
[1624494666.137552][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137562][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137567][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137576][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137581][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137588][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137593][1749871] CHIP:DL: Char1 /org/bluez/hci0/dev_67_E9_7E_06_AF_6B/service000a
[1624494666.137601][1749871] CHIP:DL: New BLE connection 0x7f4a000324c0, device 67:E9:7E:06:AF:6B, path /org/bluez/hci0/dev_67_E9_7E_06_AF_6B
[1624494666.137665][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16385
[1624494666.137687][1749875] CHIP:IN: BleConnectionComplete: endPoint 0x5598d75208e0
[1624494666.520203][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1624494666.617871][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16388
[1624494666.617891][1749875] CHIP:BLE: subscribe complete, ep = 0x5598d75208e0
[1624494666.618372][1749871] CHIP:DL: Indication received, conn = 0x7f4a000324c0
[1624494666.618435][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1624494666.618460][1749875] CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
[1624494666.618471][1749875] CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
[1624494666.618480][1749875] CHIP:BLE: local and remote recv window size = 3
[1624494666.618541][1749875] CHIP:IN: BLE EndPoint 0x5598d75208e0 Connection Complete
[1624494666.715024][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1624494667.252066][1749871] CHIP:DL: Indication received, conn = 0x7f4a000324c0
[1624494667.252167][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1624494667.300451][1749871] CHIP:DL: Indication received, conn = 0x7f4a000324c0
[1624494667.300516][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1624494667.397588][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1624494667.398522][1749871] CHIP:DL: Indication received, conn = 0x7f4a000324c0
[1624494667.398582][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1624494667.398970][1749871] CHIP:DL: Indication received, conn = 0x7f4a000324c0
[1624494667.399041][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1624494667.495051][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1624494667.495907][1749871] CHIP:DL: Indication received, conn = 0x7f4a000324c0
[1624494667.495969][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1624494667.496008][1749875] CHIP:EM: Received message of type 33 and protocolId 0 on exchange 39017
[1624494667.496024][1749875] CHIP:SC: Received PBKDF param response
[1624494667.497168][1749875] CHIP:SC: Sent spake2p msg1
[1624494667.592513][1749875] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[... 426 lines cut ...]
[1624494697.332565][1750248] CHIP:ZCL: deactivate report event
[1624494697.332571][1750248] CHIP:CTL: Getting operational keys
[1624494697.332573][1750248] CHIP:CTL: Generating credentials
[1624494697.332663][1750248] CHIP:CTL: Loaded credentials successfully
[1624494697.332752][1750256] CHIP:DL: CHIP task running
[1624494697.332791][1750248] CHIP:TOO: Test Cluster - Send Test Command: Sending command...
[1624494697.332806][1750248] CHIP:DMG: ICR moving to [Initialize]
[1624494697.332810][1750248] CHIP:DMG: ICR moving to [AddCommand]
[1624494697.333051][1750248] CHIP:SC: Sent SigmaR1 msg
[1624494697.333071][1750248] CHIP:-: CHIP Error 4072 (0x00000FE8): Not connected at ../../src/app/CommandSender.cpp:75
[1624494697.333074][1750248] CHIP:DMG: ICR moving to [Uninitiali]
[1624494697.333078][1750248] CHIP:TOO: TestCluster: CHIP Error 4072 (0x00000FE8): Not connected
[1624494697.333085][1750248] CHIP:TOO: Run command failure: CHIP Error 4072 (0x00000FE8): Not connected
[1624494697.333216][1750248] CHIP:CTL: Shutting down the commissioner
[1624494697.333223][1750248] CHIP:CTL: Shutting down the controller
[1624494697.333225][1750248] CHIP:DL: Inet Layer shutdown
[1624494697.333254][1750255] CHIP:IN: Async DNS worker thread woke up.
[1624494697.333277][1750255] CHIP:IN: Async DNS worker thread exiting.
[1624494697.333293][1750254] CHIP:IN: Async DNS worker thread woke up.
[1624494697.333314][1750254] CHIP:IN: Async DNS worker thread exiting.
[1624494697.333511][1750248] CHIP:DL: BLE layer shutdown
[1624494697.333518][1750248] CHIP:DL: System Layer shutdown
```

@pan-apple @vivien-apple

mspang, 2021-06-24:
tagging https://github.com/project-chip/connectedhomeip/issues/7860 since the subject didn't do it

mspang, 2021-06-24:
tagging #7666 since the subject didn't do it

mspang, 2021-06-24:
Same issue with the other tests under `chip-tool tests` as well

pan-apple, 2021-06-24:
I'll take a look.

PR #7865 (2021-06-24): Use GetConnectedDevice in TestCommand to wait for CASE session
#### Problem
Fixes #7860 chip-tool tests TestCluster is broken after "Enable CASE session establishment

#### Change overview
Use `GetConnectedDevice()` API to get a callback on CASE session establishment, for test commands.

#### Testing
Ran the tests using `chip-tool tests TestCluster`
Note: need this PR to fix some packet buffer and memory leaks. https://github.com/project-chip/connectedhomeip/pull/7852
Without #7852, the device may reboot after a few CASE session establishments due to memory allocation failures.
closed 2021-06-24 by mspang