project-chip/connectedhomeip #7887: Lock App keeps Crashing on EFR32 MG21  with the latest Master branch

saurabhst, 2021-06-24:
#### Problem
Lock App keeps Crashing on EFR32 MG21  with the latest Master branch.

Crash Logs -
```Process: JLinkExe
<efr32 > ==================================================
<efr32 > chip-efr32-lock-example starting
<efr32 > ==================================================
<efr32 > Init CHIP Stack
<detail> [-] Initializing key value store
<detail> [-] KeyValueStore init complete: active keys 2, deleted keys 2, sectors 4, logical sector size 8192 bytes
<info  > [DL] Setting device name to : "EFR32_LOCK"
<efr32 > Initializing OpenThread stack
<info  > [DL] OpenThread started: OK
<info  > [DL] Setting OpenThread device type to ROUTER
<efr32 > Starting Platform Manager Event Loop
<efr32 > Starting OpenThread task
<efr32 > Starting App Task
<efr32 > Starting FreeRTOS scheduler
<error > HardFault:
SCB->CFSR   0x01000000
SCB->HFSR   0x40000000
SCB->MMFAR  0xe000ed34
SCB->BFAR   0xe000ed38
SP          0x20009618
R0          0x2000968c
R1          0x00000002
R2          0x000163af
R3          0xa5a5a5a5
R12         0xa5a5a5a5
LR          0x000172a9
PC          0x0001657c
PSR         0x0100000```

 #### Proposed Solution
Lock app should be able to run successfully on EFR32 MG21 platform.

jmartinez-silabs, 2021-06-25:
@saurabhst I could not reproduce with ToT can you pull and retry ? If you still get this please give me some more info. With this log  it seems that right after boot up it hard faults. I was able to do the full commissionning flow.

saurabhst, 2021-06-25:
@jmartinez-silabs thanks for checking on this. Let me retry and report back the result .

closed 2022-01-26 by woody-apple