project-chip/connectedhomeip #7918: Resolve Command on python controller unable to resolve when run inside a docker container.

wahajsyed, 2021-06-25:
#### Problem
Python Controller running inside a docker container errors out when trying to run resolve command. 

System Configuration:
- Ubuntu 20.10 docker container running on a RasPi 4

Steps to reproduce: 
- Install the chip sdk .whl package inside the docker container using the docker file [here](https://github.com/chip-csg/connectedhomeip/blob/csg/test_event_3.1/src/controller/python/test_client/Dockerfile)
- Run connect command (successful)
- Run resolve command (fails)

Log output: 
```
chip-device-ctrl > resolve 0 12344321
OS Error 2099 (0x00000833): Cannot assign requested address
```

Other things I tried:
- The same set of commands when run on the host machine (RasPi 4) run fine without any issues, only fails when run from inside a docker container. 

Use Case: 
We need to be able to run the chip controller from inside a docker contianer to support the test harness and the test cases.

bzbarsky-apple, 2021-07-07:
@wahajsyed What server is the python controller connecting to?  What do the logs (if any) look like?  Ideally from a chip-device-ctrl compiled with detail logging enabled....

(More detailed steps to reproduce, showing the exact commands to run at a shell prompt, might be handy as well, because a lot of this might depend on the exact way the docker container is being created and started...)

wahajsyed, 2021-07-07:
@bzbarsky-apple - I am trying to check if I can repro this with TE4 branch, will update you if the issue still persists

wahajsyed, 2021-07-07:
Controller Side Logs:

```console
ubuntu@pixie:~/repos/connectedhomeip/src/controller/python/test_client$ docker run --rm -it --entrypoint bash --mount type=bind,source=/var/run/dbus,target=/var/run/dbus -p 5050:5000 --privileged=true 660345c43a33
appuser@2aa3dd0219dd:/chip_tool$ chip-device-ctrl 
[1625694064.750101][8:8] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsCAKey
[1625694064.750492][8:8] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsICAKey
[1625694064.756360][8:8] CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
[1625694064.756541][8:8] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1625694064.757014][8:8] CHIP:DL: writing settings to file (/tmp/chip_config.ini.tmp)
[1625694064.757191][8:8] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1625694064.757587][8:8] CHIP:DL: writing settings to file (/tmp/chip_counters.ini.tmp)
[1625694064.757777][8:8] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1625694064.758785][8:8] CHIP:IN: TransportMgr initialized
[1625694064.758864][8:8] CHIP:DIS: Init admin pairing table with server storage
[1625694064.758945][8:8] CHIP:IN: local node id is 0x000000000001B669
[1625694064.760413][8:8] CHIP:ZCL: Using ZAP configuration...
[1625694064.760642][8:8] CHIP:ZCL: deactivate report event
[1625694064.760733][8:8] CHIP:CTL: Getting operational keys
[1625694064.761170][8:8] CHIP:CTL: Getting root certificate for the controller from the issuer
[1625694064.762558][8:8] CHIP:CTL: Generating operational certificate for the controller
[1625694064.764049][8:8] CHIP:CTL: Getting intermediate CA certificate from the issuer
[1625694064.764516][8:8] CHIP:CTL: GetIntermediateCACertificate returned 0
[1625694064.764921][8:8] CHIP:CTL: SyncSetKeyValue on CHIPAdmin0
[1625694064.765000][8:8] CHIP:CTL: Generating credentials
[1625694064.765510][8:8] CHIP:CTL: Loaded credentials successfully
[1625694064.782380][8:16] CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > 
chip-device-ctrl > connect -ble 3840 20202021 12344321
Device is assigned with nodeid = 12344321
[1625694176.686737][8:8] CHIP:SC: Assigned local session key ID 1
[1625694176.687188][8:8] CHIP:IN: [1625694176.687282][8:12] CHIP:DL: TRACE: Bus acquired for name C-0008
Message appended to BLE send queue
[1625694176.687680][8:8] CHIP:SC: Sent PBKDF param request
[1625694176.688081][8:8] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1625694176.688112][8:1106] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1625694176.698996][8:16] CHIP:DL: PlatformBlueZInit init success
[1625694176.706165][8:12] CHIP:BLE: BLE removing known devices.
[1625694176.706288][8:12] CHIP:BLE: BLE initiating scan.
[1625694176.727127][8:12] CHIP:BLE: Device 73:84:F4:68:34:15 does not look like a CHIP device.
[1625694176.734285][8:12] CHIP:BLE: Device 36:E6:F9:69:A3:9E does not look like a CHIP device.
[1625694176.759297][8:12] CHIP:BLE: Device 5C:CB:6B:DD:C4:07 does not look like a CHIP device.
[1625694176.768165][8:12] CHIP:BLE: Device 4D:06:AD:32:7A:4E does not look like a CHIP device.
[1625694176.769952][8:12] CHIP:BLE: Device F0:18:98:34:BA:8A does not look like a CHIP device.
[1625694176.802252][8:12] CHIP:BLE: Device 63:69:3E:0D:81:7C does not look like a CHIP device.
[1625694176.804408][8:12] CHIP:BLE: Device 18:33:60:F9:0F:EF does not look like a CHIP device.
[1625694176.827213][8:12] CHIP:BLE: Device 77:29:BD:69:E4:18 does not look like a CHIP device.
[1625694176.830580][8:12] CHIP:BLE: Device 37:C9:33:7F:BB:4B does not look like a CHIP device.
[1625694176.840922][8:12] CHIP:BLE: Device 55:74:7F:F7:00:87 does not look like a CHIP device.
[1625694176.883667][8:12] CHIP:BLE: Device 02:79:3F:AD:18:47 does not look like a CHIP device.
[1625694176.907281][8:12] CHIP:BLE: Device 73:34:43:66:0D:37 does not look like a CHIP device.
[1625694176.919938][8:12] CHIP:BLE: Device 7C:F7:94:80:3B:12 does not look like a CHIP device.
[1625694176.930416][8:12] CHIP:BLE: Device 4B:16:A4:83:4F:A2 does not look like a CHIP device.
[1625694176.966019][8:12] CHIP:BLE: Device 57:55:22:AA:CD:D2 does not look like a CHIP device.
[1625694177.000822][8:12] CHIP:BLE: Device 60:3C:CB:71:86:E5 does not look like a CHIP device.
[1625694177.003227][8:12] CHIP:BLE: Device 42:3E:B1:7A:CB:61 does not look like a CHIP device.
[1625694177.196244][8:12] CHIP:BLE: Device 28:5E:D8:86:20:C9 does not look like a CHIP device.
[1625694177.222989][8:12] CHIP:BLE: Device 7D:98:8F:EC:BA:E8 does not look like a CHIP device.
[1625694177.309406][8:12] CHIP:BLE: Device 7A:3F:05:0A:92:16 does not look like a CHIP device.
[1625694177.572998][8:12] CHIP:BLE: Device 40:92:88:A3:0B:B3 does not look like a CHIP device.
[1625694177.580091][8:12] CHIP:BLE: New device scanned: DC:A6:32:BE:E6:8B
[1625694177.580192][8:12] CHIP:BLE: Device discriminator match. Attempting to connect.
[1625694177.585018][8:12] CHIP:BLE: Scan complete notification without an active scan.
[1625694181.544733][8:12] CHIP:DL: ConnectDevice complete
[1625694182.934463][8:12] CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_BE_E6_8B/service0010
[1625694182.934540][8:12] CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_BE_E6_8B/service0013
[1625694182.934592][8:12] CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_BE_E6_8B/service0010
[1625694182.934637][8:12] CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_BE_E6_8B/service0013
[... 521 lines cut ...]
[1625694285.967948][8:12] CHIP:DL: Indication received, conn = 0xffffac0633e0
[... cut ...]
chip-device-ctrl > [1625694302.593353][8:16] CHIP:DL: HandlePlatformSpecificBLEEvent 16388
[1625694302.593524][8:16] CHIP:BLE: no endpoint for unsub complete
[1625694302.593636][8:12] CHIP:DL: Bluez disconnected
[1625694302.593782][8:12] CHIP:DL: Bluez notify CHIPoBluez connection disconnected

chip-device-ctrl > 
chip-device-ctrl > resolve 0 12344321
[1625694322.643777][8:8] CHIP:DIS: mDNS broadcast success
[1625694322.644141][8:8] CHIP:DIS: Attempt to mDNS broadcast failed:  OS Error 2099 (0x00000833): Cannot assign requested address
```

wahajsyed, 2021-07-07:
Device Side Logs:

```console
wahaj@blackpi:~/repos/connectedhomeip/examples/lighting-app/linux$ sudo out/debug/chip-lighting-app --wifi 
[sudo] password for wahaj: 
[1625694142.910340][2063:2063] CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
[1625694142.910567][2063:2063] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1625694142.910887][2063:2063] CHIP:DL: writing settings to file (/tmp/chip_config.ini.tmp)
[1625694142.911027][2063:2063] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1625694142.911261][2063:2063] CHIP:DL: writing settings to file (/tmp/chip_counters.ini.tmp)
[1625694142.911401][2063:2063] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1625694142.911642][2063:2063] CHIP:DL: Device Configuration:
[1625694142.911680][2063:2063] CHIP:DL:   Serial Number: TEST_SN
[1625694142.911708][2063:2063] CHIP:DL:   Vendor Id: 9050 (0x235A)
[1625694142.911736][2063:2063] CHIP:DL:   Product Id: 65279 (0xFEFF)
[1625694142.911765][2063:2063] CHIP:DL:   Product Revision: 1
[1625694142.911794][2063:2063] CHIP:DL:   Setup Pin Code: 20202021
[1625694142.911824][2063:2063] CHIP:DL:   Setup Discriminator: 3840 (0xF00)
[1625694142.911853][2063:2063] CHIP:DL:   Manufacturing Date: (not set)
[1625694142.911881][2063:2063] CHIP:DL:   Device Type: 257 (0x101)
[1625694142.911975][2063:2063] CHIP:SVR: SetupQRCode: [MT:YNJV7VSC00KA0648G00]
[1625694142.912012][2063:2063] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1625694142.912039][2063:2063] CHIP:SVR: https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3AYNJV7VSC00KA0648G00
[1625694142.912084][2063:2063] CHIP:SVR: Manual pairing code: [34970112332]
[1625694142.912785][2063:2063] CHIP:ZCL: Using ZAP configuration...
[1625694142.913029][2063:2063] CHIP:ZCL: deactivate report event
[1625694142.913088][2063:2063] CHIP:ZCL: OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pairing table.
[1625694142.913120][2063:2063] CHIP:DIS: Set the admin pairing table delegate
[1625694142.913148][2063:2063] CHIP:ZCL: OpCreds: Call to writeAdminsIntoFabricsListAttribute
[1625694142.913179][2063:2063] CHIP:ZCL: OpCreds: Stored 0 admins in fabrics list attribute.
[1625694142.913218][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913251][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913282][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913313][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913344][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913384][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913416][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913446][2063:2063] CHIP:ZCL: Unknown cluster ID: 28
[1625694142.913485][2063:2063] CHIP:ZCL: Unknown cluster ID: 300
[1625694142.913515][2063:2063] CHIP:ZCL: Unknown cluster ID: 300
[1625694142.913546][2063:2063] CHIP:ZCL: Unknown cluster ID: 300
[1625694142.913577][2063:2063] CHIP:ZCL: Unknown cluster ID: 406
[1625694142.913608][2063:2063] CHIP:ZCL: Unknown cluster ID: 406
[1625694142.913911][2063:2063] CHIP:DL: writing settings to file (/tmp/chip_server_kvs.tmp)
[1625694142.914090][2063:2063] CHIP:DL: renamed tmp file to file (/tmp/chip_server_kvs)
[1625694142.914227][2063:2063] CHIP:DIS: Init admin pairing table with server storage
[1625694142.914527][2063:2063] CHIP:IN: TransportMgr initialized
[1625694142.914787][2063:2063] CHIP:DL: writing settings to file (/tmp/chip_counters.ini.tmp)
[1625694142.915548][2063:2063] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1625694142.915613][2063:2063] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 1000 (0x3E8)
[1625694142.915649][2063:2063] CHIP:IN: local node id is 0x0000000000BC5C01
[1625694142.915722][2063:2063] CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
[1625694142.915841][2063:2063] CHIP:SC: Assigned local session key ID 0
[1625694142.915880][2063:2063] CHIP:SC: Waiting for PBKDF param request
[1625694142.915911][2063:2063] CHIP:DIS: Start dns-sd server
[1625694142.918478][2063:2063] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1625694142.918857][2063:2063] CHIP:DIS: CHIP minimal mDNS started advertising.
[1625694142.919596][2063:2067] CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
[1625694142.919781][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.919825][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.919858][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920023][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920061][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920093][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920263][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920303][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920336][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920500][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920540][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[1625694142.920568][2063:2063] CHIP:DIS: Replying to DNS-SD service listing request
[... 497 lines cut ...]
[1625694265.735232][2063:2067] CHIP:BLE: Indication confirmation, 0xffff8804ac50
[... cut ...]
[1625694293.570703][2063:2067] CHIP:BLE: Write request received debug 0xffff8804ac50
[1625694296.155274][2063:2067] CHIP:BLE: Indication confirmation, 0xffff8804ac50
[1625694298.640457][2063:2067] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1625694298.640848][2063:2067] CHIP:BLE: Write request received debug 0xffff8804ac50
[1625694302.837320][2063:2067] CHIP:DL: Bluez disconnected
[1625694302.837418][2063:2067] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1625694302.837929][2063:2063] CHIP:IN: Clearing BLE pending packets.
```

wahajsyed, 2021-07-07:
List of commands: 
```
connect -ble 3840 20202021 12344321
zcl NetworkCommissioning AddWiFiNetwork 12344321 1 0 ssid=str:Sadaath_Restricted credentials=str:Password1234 breadcrumb=0 timeoutMs=1000
zcl NetworkCommissioning EnableNetwork 12344321 1 0 networkID=str:Sadaath_Restricted breadcrumb=0 timeoutMs=1000
```

wahajsyed, 2021-07-07:
Setup : 
Raspi 1: 
- Clone the repo
- Checkout test_event_4
- Activate the env and build using:
```
gn gen out/debug
ninja -C out/debug
``` 
- Create a docker container using the docker [image](https://github.com/chip-csg/connectedhomeip/blob/csg/test_event_3.1/src/controller/python/test_client/Dockerfile) 
- Run the docker image using 
```
docker run --rm -it --entrypoint bash --mount type=bind,source=/var/run/dbus,target=/var/run/dbus -p 5050:5000 --privileged=true 660345c43a33
```
- Ensure Raspi 2 is up and running by following the below steps
- Run the following list of commands
```
connect -ble 3840 20202021 12344321
zcl NetworkCommissioning AddWiFiNetwork 12344321 1 0 ssid=str:Sadaath_Restricted credentials=str:Password1234 breadcrumb=0 timeoutMs=1000
zcl NetworkCommissioning EnableNetwork 12344321 1 0 networkID=str:Sadaath_Restricted breadcrumb=0 timeoutMs=1000
resolve 0 12344321
```

Raspi 2: 

- Clone the repo
- Checkout test_event_4
- Activate the env
- Navigate to the examples/lighting-app/linux
- Build using:
```
gn gen out/debug
ninja -C out/debug
``` 
- Run using
```
 sudo out/debug/chip-lighting-app --wifi
 ```

bzbarsky-apple, 2021-07-08:
> Create a docker container using the docker image

Just to check, what are the exact steps here?

bzbarsky-apple, 2021-07-08:
So the controller log is showing that we manage to broadcast successfully on one interface, then fail on a different one.  @wahajsyed could you apply the [attached patch](https://github.com/project-chip/connectedhomeip/files/6781435/test-patch.txt) to log more information and try the controller side, to see which interface/addrtype it is that's failing?

That said, it sounds like we are broadcasting correctly one one interface; the question is why we get no replies....  Do we need to expose port 5353 from the docker container?

wahajsyed, 2021-07-08:
Got this log from the patch
```
chip-device-ctrl > 
chip-device-ctrl > resolve 0 12344321
[1625773112.418847][8:8] CHIP:DIS: mDNS broadcast success on eth0 for address type IPv4
[1625773112.419160][8:8] CHIP:DIS: Attempt to mDNS broadcast on eth0 for address type IPv6 failed:  OS Error 2099 (0x00000833): Cannot assign requested address
```

Is there something special with the 5353 port?

wahajsyed, 2021-07-08:
I think it might be related to enabling ipv6 within docker container, same interface but fails for ipv6 and not for ipv4

wahajsyed, 2021-07-09:
https://github.com/chip-csg/chip-certification-tool/pull/214#pullrequestreview-703032335
Solves the issue, the IPV6 network needed to be enabled.

wahajsyed, 2021-07-09:
Logs from the device: 
```console
[1625835975.675615][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.675735][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.675842][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.675903][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835975.676281][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.676366][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.676467][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.676525][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835975.676872][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.676955][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.677055][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835975.677165][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835976.681475][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835976.681584][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835976.681687][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835976.681748][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835977.159337][10355:10358] CHIP:BLE: Indication confirmation, 0xffff980444e0
[1625835979.644265][10355:10358] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1625835979.644547][10355:10358] CHIP:BLE: Write request received debug 0xffff980444e0
[1625835979.692521][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835979.692626][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835979.692720][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835979.692773][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835981.651645][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835981.652093][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835981.652393][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835982.229493][10355:10358] CHIP:BLE: Indication confirmation, 0xffff980444e0
[1625835982.656012][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835983.224668][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.224888][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.225309][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.225520][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.225781][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.225965][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.279142][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.282215][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.282500][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.285762][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.288785][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.289272][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.291238][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.292478][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835983.292599][10355:10355] CHIP:DIS: Directly sending mDns reply to peer on port 5353
[1625835984.281148][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835984.284403][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835984.284940][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835984.714315][10355:10358] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1625835984.714593][10355:10358] CHIP:BLE: Write request received debug 0xffff980444e0
[1625835985.663915][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835987.291206][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835987.294569][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835987.295085][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835987.348056][10355:10358] CHIP:BLE: Indication confirmation, 0xffff980444e0
[1625835988.710977][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835988.711083][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835988.711191][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625835988.711252][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835989.833164][10355:10358] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1625835989.833446][10355:10358] CHIP:BLE: Write request received debug 0xffff980444e0
[1625835992.418254][10355:10358] CHIP:BLE: Indication confirmation, 0xffff980444e0
[1625835994.758707][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835994.762194][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835994.765454][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835994.765955][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625835995.833769][10355:10358] CHIP:DL: Bluez disconnected
[... cut ...]
[1625839792.519177][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625839799.226317][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625839859.121405][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625839859.121600][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625839859.121714][10355:10355] CHIP:DIS: Replying to DNS-SD service listing request
[1625839859.121777][10355:10355] CHIP:DIS: Broadcasting mDns reply
[1625839898.700821][10355:10355] CHIP:DIS: Broadcasting mDns reply
```

bzbarsky-apple, 2021-07-09:
> Is there something special with the 5353 port?

That's the port mdns uses, right?

wahajsyed, 2021-07-09:
Yep from the logs: 
```
[1625850052.468081][13233] CHIP:DIS: Directly sending mDns reply to peer on port 5353
```

bzbarsky-apple, 2021-07-09:
Hence my question: is port 5353 traffic able to transit in/out of the container onto the broader network correctly in this case?

wahajsyed, 2021-07-09:
I tried mapping the 5353 port from the host onto the container, is there a way to verify that the port traffic is able to make it out of the container?

bzbarsky-apple, 2021-07-10:
I don't know.....  Could try wireshark to see what's transiting the actual network between the raspis?  That will at least tell us whether the outgoing broadcasts are happening, though not whether the incoming responses land inside the container....

wahajsyed, 2021-07-13:
Issue resolved. 

TE4 branch with updated endpoint commands, built with the platform flag and using the 5Ghz SSID instead of 2.4 GHz seems to be working. 

@bzbarsky-apple  - I did notice that the address is being resolved to an IPv4, is that expected here?

bzbarsky-apple, 2021-07-13:
> I did notice that the address is being resolved to an IPv4, is that expected here?

It's pretty random what it resolves to.  The advertiser advertises all valid IPs, and the resolver kinda grabs the first one it gets a response for.  So yes, this is expected for now.

closed 2021-07-14 by wahajsyed