project-chip/connectedhomeip #7935: Maybe -Wformat-type-confusion to the build as much as we can

bzbarsky-apple, 2021-06-26:
#### Problem
We are not getting all possible format confusion warnings right now.

 #### Proposed Solution
See https://github.com/project-chip/connectedhomeip/pull/7921#issuecomment-868983724 -- add `-Wformat-type-confusion` to catch more.

I tried adding it in `build/config/compiler/BUILD.gn` but then I get errors in third-party code like so:
```
../../third_party/lwip/repo/lwip/src/core/ipv6/ip6.c:1173:21: error: format specifies type 'unsigned short' but the argument has type 'u8_t' (aka 'unsigned char') [-Werror,-Wformat-type-confusion]
                    IP6H_NEXTH(ip6hdr),
                    ^~~~~~~~~~~~~~~~~~~
```

@mspang is there somewhere we can add this that will get used for our code but not third-party code?

mspang, 2021-06-28:
We actually already are disabling warnings in lwip

```
  config("${lwip_target_name}_warnings") {
    cflags = [
      "-Wno-address",
      "-Wno-format",
      "-Wno-type-limits",
      "-Wno-unused-variable",
      "-Wno-maybe-uninitialized",
    ]
  }
```

We can split the warnings_default config into warnings_default and warnings_third_party if we want but it's a little bit tricky to do it without introducing coupling with the name & organization of the core build rules (which I'm really trying to minimize, so that you can drop in CHIP into a project with different versions of them which is something that we are doing).

One way we could do it is https://github.com/project-chip/connectedhomeip/pull/7959

bzbarsky-apple, 2021-06-29:
> We actually already are disabling warnings in lwip

Perfect, thank you!

closed 2021-06-29 by woody-apple