project-chip/connectedhomeip #7939: Move IM standalone test run inside chip thread.

yunhanw-google, 2021-06-28:
#### Problem
we see race happens in IM test, since shutdown happens after OnWriteResponse mpDelegate->WriteResponseProcessed(this); inside WriteResponseProcessed, https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/integration/chip_im_initiator.cpp#L291 has been called to unblock the next test run, but shutdown has not yet run.

in cirque test, we are seeing the flaky for IM test
Send write request message to Node: 12344321
[1624643323.443755][157] CHIP:DMG: WriteClient moving to [AddAttribu]
[1624643323.443785][157] CHIP:IN: Secure message was encrypted: Msg ID 17
[1624643323.443793][157] CHIP:IN: Encrypted message 0x7fffffffe2c0 from 0x000000000001B669 to 0x0000000000BC5C01 of type 6 and protocolId 5 on exchange 39025.
[1624643323.443798][157] CHIP:IN: Sending msg 0x7fffffffe2c0 to 0x0000000000BC5C01 at utc time: 1160308 msec
[1624643323.443801][157] CHIP:IN: Sending secure msg on generic transport
[1624643323.443822][157] CHIP:IN: Secure msg send status No Error
[1624643323.443831][157] CHIP:DMG: WriteClient moving to [AwaitingRe]
[1624643323.644432][167] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 0
[1624643323.644475][167] CHIP:EM: Received message of type 7 and protocolId 5 on exchange 39025
[1624643323.644488][167] CHIP:EM: Sending Standalone Ack for MsgId:00000009
[1624643323.644495][167] CHIP:IN: Secure message was encrypted: Msg ID 18
[1624643323.644498][167] CHIP:IN: Encrypted message 0x7fffe2ffd330 from 0x000000000001B669 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39025.
[1624643323.644502][167] CHIP:IN: Sending msg 0x7fffe2ffd330 to 0x0000000000BC5C01 at utc time: 1160509 msec
[1624643323.644504][167] CHIP:IN: Sending secure msg on generic transport
[1624643323.644526][167] CHIP:IN: Secure msg send status No Error
[1624643323.644534][167] CHIP:EM: Flushed pending ack for MsgId:00000009
[1624643323.644541][167] CHIP:DMG: WriteResponse =
[1624643323.644544][167] CHIP:DMG: {
[1624643323.644546][167] CHIP:DMG: 	AttributeStatusList =
[1624643323.644548][167] CHIP:DMG: 	[
[1624643323.644551][167] CHIP:DMG: 		AttributeStatusElement =
[1624643323.644553][167] CHIP:DMG: 		{
[1624643323.644555][167] CHIP:DMG: 			AttributePath =
[1624643323.644558][167] CHIP:DMG: 			{
[1624643323.644560][167] CHIP:DMG: 				FieldTag = 0x4,
[1624643323.644562][167] CHIP:DMG: 				NodeId = 0x1,
[1624643323.644565][167] CHIP:DMG: 				ClusterId = 0x3,
[1624643323.644567][167] CHIP:DMG: 				EndpointId = 0x2,
[1624643323.644569][167] CHIP:DMG: 			}
[1624643323.644572][167] CHIP:DMG: 				
[1624643323.644574][167] CHIP:DMG: 			StatusElement =
[1624643323.644576][167] CHIP:DMG: 			{
[1624643323.644578][167] CHIP:DMG: 				GeneralCode = 0x0,
[1624643323.644580][167] CHIP:DMG: 				ProtocolId = 0x0,
[1624643323.644583][167] CHIP:DMG: 				protocolCode = 0x0,
[1624643323.644584][167] CHIP:DMG: 			},
[1624643323.644587][167] CHIP:DMG: 			
[1624643323.644589][167] CHIP:DMG: 		},
[1624643323.644592][167] CHIP:DMG: 		
[1624643323.644594][167] CHIP:DMG: 	],
[1624643323.644597][167] CHIP:DMG: 	
[1624643323.644599][167] CHIP:DMG: }
Write Response: 2/2(100.00%) time=0.201ms
ChipCommandSender failed: CHIP Error 4011 (0x00000FAB): No memory
[1624643323.645142][167] CHIP:DMG: WriteClient moving to [Uninitiali]
[1624643323.645142][167] CHIP:DMG: WriteClient moving to [Uninitiali]
[Thread 0x7fffedffd700 (LWP 165) exited]
[Thread 0x7fffe2ffe700 (LWP 167) exited]
[Thread 0x7fffe3fff700 (LWP 166) exited]
[Thread 0x7fffeeffe700 (LWP 164) exited]
[Thread 0x7fffeffff700 (LWP 163) exited]
[Thread 0x7ffff5f24700 (LWP 162) exited]
[Thread 0x7ffff6f25700 (LWP 161) exited]
[Inferior 1 (process 157) exited with code 01]
No stack.
(gdb) quit

yunhanw-google, 2021-07-07:
we may apply the way similar to https://github.com/project-chip/connectedhomeip/pull/8100

yufengwangca, 2021-07-12:
Closed with #8174

closed 2021-07-12 by yufengwangca