project-chip/connectedhomeip #7941: chip-device-ctrl - resolve command gives error with all-clusters-app on mac

PSONALl, 2021-06-28:
#### Problem
chip-device-ctrl - resolve command gives error with all-clusters-app on mac
 * expected behavior : resolve dns and update address of the node in the device controller after command
      `resolve 0 135246`
 * actual behavior - resolve command gives error with all-clusters-app on mac like:
 ```
chip-device-ctrl > resolve 0 135246
[1624866923276] [0x828ff] CHIP: [DIS] Node ID resolved for 0x000000000002104E to fe80::c6dd:57ff:fe5b:f7f0
Node address has been updated
[1624866923276] [0x828ff] CHIP: [CTL] OperationalDiscoveryComplete for device ID 135246
Current address: unknown
chip-device-ctrl > [1624866923277] [0x828ff] CHIP: [CTL] LoadSecureSessionParameters returning error 2022
[1624866923277] [0x828ff] CHIP: [CTL] Device connection failed. Error OS Error 2022 (0x000007E6): Invalid argument 
```
>
 * system configuration
    - platform : macOS
    - kit : esp32 devkitC
 *  logs
    - device side : [device-side.txt](https://github.com/project-chip/connectedhomeip/files/6724599/device-side.txt)

    - python controller side : [python-ctrl.txt](https://github.com/project-chip/connectedhomeip/files/6724601/python-ctrl.txt)

bzbarsky-apple, 2021-06-28:
@PSONALl Which exact revision are you seeing this on?

Also, that failure does not look like it's coming from resolve per se... Just out of curiosity:

1) Is this reliably reproducible?
2) Does issuing `zcl OnOff Toggle 135246 1 0` or something else that would actually try to use the connection to the device work after seeing the above error?

PSONALl, 2021-06-28:
>Which exact revision are you seeing this on?

I am using esp-idf release/v4.2 branch and latest chip head

> 2\. `zcl OnOff Toggle 135246 1 0` or something else that would actually try to use the connection to the device work after seeing the above error?

No, `zcl OnOff Toggle 135246 1 0` is also not working after the above error

bzbarsky-apple, 2021-06-28:
> nd latest chip head

So revision `17dbc199973f3532d360d26ecbe4f13a4ecafebc`?

Just to make sure we are looking at similar timing, are you manually typing in the `chip-device-ctrl` commands?  Or just pasting them all in at once so the next one runs as soon as the previous one completes?

I'll see if I can reproduce, but I haven't been able to so far.  Given that, if you can reproduce reliably, are you willing/able to step through in a debugger and see what's going on?  Or add more logging and do likewise?  In particular, it looks from those logs like `pychip_DeviceController_GetAddressAndPort` in `src/controller/python/ChipDeviceController-ScriptBinding.cpp` is returning an error.  So it would be useful to check whether that's true and if so which of the calls there fails, exactly, and why.

Also, would be useful to have a stack to the "LoadSecureSessionParameters returning error" log line happening; that's in `Device::LoadSecureSessionParameters` in `src/controller/CHIPDevice.cpp`.

PSONALl, 2021-06-28:
> So revision `17dbc199973f3532d360d26ecbe4f13a4ecafebc`?

Yes!

> Just to make sure we are looking at similar timing, are you manually typing in the `chip-device-ctrl` commands? Or just pasting them all in at once so the next one runs as soon as the previous one completes?

I have manually typed the `chip-device-ctrl` commands

> I'll see if I can reproduce, but I haven't been able to so far. Given that, if you can reproduce reliably, are you willing/able to step through in a debugger and see what's going on? Or add more [...]

Ok, Thanks @bzbarsky-apple , I will check

PSONALl, 2021-07-16:
I have updated logs for the above issue, still facing error for my macOS BigSur 11.1, please check

PSONALl, 2021-07-16:
Here are the details:
I am using:
  - esp-idf : tag v4.3 (head at c9646ff0beffc86d2c6d1bfbad34da16e328e0e3 )
  - chip : master ( head 6d613877186c73372e1c17298d46e6dd910bebae )
  - platform: macOS BigSur 11.1
  - kit - esp32 devkit

Currently facing issue for resolve command like:
```
chip-device-ctrl > resolve 0 135246
[1626426434869] [45189:287982] CHIP: [DIS] Node ID resolved for 0x000000000002104E to [fe80::c6dd:57ff:fe5b:f7f0]:11097
Node address has been updated
[1626426434869] [45189:287982] CHIP: [CTL] OperationalDiscoveryComplete for device ID 135246
[1626426434870] [45189:287982] CHIP: [CTL] LoadSecureSessionParameters returning error 2000016
[1626426434870] [45189:287982] CHIP: [CTL] Device connection failed. Error OS Error 0x02000016: Invalid argument
Failed to commission: 33554454
```
Detailed logs:
  - device side: [device-logs.txt](https://github.com/project-chip/connectedhomeip/files/6829411/udated-device-logs.txt)
  - controller side: [python-ctrl-logs.txt](https://github.com/project-chip/connectedhomeip/files/6829417/updated-python-ctrl-logs.txt)

Also while tracing the error, I have found that error is returned while sending IP packet. The function `IPEndPointBasis::SendMsg`  from `src/inet/IPEndPointBasis.cpp` is returning error, In the  function `IPEndPointBasis::SendMsg` At following point `sendmsg` returns -1, thus lenSent becomes -1, so it returns error

```
    // Send IP packet.
    const ssize_t lenSent = sendmsg(mSocket.GetFD(), &msgHeader, 0);
    if (lenSent == -1)
        return chip::System::MapErrorPOSIX(errno);
```

bzbarsky-apple, 2021-07-29:
> Also while tracing the error, I have found that error is returned while sending IP packet

@PSONALl Which packet is that?  That is, what does the stack to that `IPEndPointBasis::SendMsg` look like when this fails?

PSONALl, 2021-08-04:
> @PSONALl Which packet is that? That is, what does the stack to that `IPEndPointBasis::SendMsg` look like when this fails?

This is how the stack looks like:

Device::EstablishConnectivity (src/controller/CHIPDevice.cpp)

Device::LoadSecureSessionParametersIfNeeded (src/controller/CHIPDevice.cpp)

Device::LoadSecureSessionParameters (src/controller/CHIPDevice.cpp)

Device::WarmupCASESession (src/controller/CHIPDevice.cpp)

CASESession::EstablishSession (src/protocols/secure_channel/CASESession.cpp)

CASESession::SendSigmaR1 (src/protocols/secure_channel/CASESession.cpp)

 ExchangeContext::SendMessage (src/messaging/ExchangeContext.cpp)

ExchangeMessageDispatch::SendMessage (src/messaging/ExchangeMessageDispatch.cpp)

TransportMgrBase::SendMessage (src/transport/TransportMgrBase.cpp)

UDP::SendMessage (src/transport/raw/UDP.cpp)

UDPEndPoint::SendMsg (src/inet/UDPEndPoint.cpp)
UDP point

IPEndPointBasis::SendMsg (src/inet/IPEndPointBasis.cpp)

bzbarsky-apple, 2021-08-05:
OK, so the very first packet we try to send is failing...

At the place where we print this:
```
[1626426434869] [45189:287982] CHIP: [DIS] Node ID resolved for 0x000000000002104E to [fe80::c6dd:57ff:fe5b:f7f0]:11097
```
a few questions:
1) Is that IP address correct, presumably?
2) Can you ping6 that address from the machine that is ending up with the error stack above, given the right `-I` value?
3) Is our mdns resolution finding the right interface id in this case?

PSONALl, 2021-08-06:
This issue is solved now with recent master branch commit head  `3d8d542b4dcc69cfa1b39af6df882687968fb67c` .  Closing this issue.

closed 2021-08-06 by PSONALl