project-chip/connectedhomeip #7944: Execute "export.sh" for esp32 in branch of release/4.3 failed

Hugh-Bao, 2021-06-28:
#### Problem
PC: macOS v10.15.6
bash: v5.1.8, installed by "brew"
When I using export.sh to setup my idf enviroment, it will be failed at the end of command.

PM80266089:esp-idf 80266089$ . ./export.sh 
Detecting the Python interpreter
Checking "python" ...
Python 3.9.1
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /Users/80266089/work/chip/esp/toolchains/python_env/idf4.3_py3.9_env/bin/python
Checking if Python packages are up to date...
Python requirements from /Users/80266089/work/chip/esp/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
 /Users/80266089/work/chip/esp/esp-idf/components/esptool_py/esptool
 /Users/80266089/work/chip/esp/esp-idf/components/espcoredump
 /Users/80266089/work/chip/esp/esp-idf/components/partition_table
 /Users/80266089/work/chip/esp/esp-idf/components/app_update
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
 idf.py build
-bash: complete: nosort: invalid option name
WARNING: Failed to load shell autocompletion!

juejuegongsui, 2021-07-02:
如果没有解决，可以看看我的，小伙子你这个错误很神奇，没遇到过，但是有类似的错误，也是bash的，不保证有用哈，去你的export.sh下，159行增加LANG=en，我的bash问题是这样解决的 @Hugh-Bao

Hugh-Bao, 2021-07-02:
MacOS环境会出这个问题，也在export.sh 159行添加了LANG=en，但是目前不影响使用。（I have added "LANG=en" at line 159 in expot.sh, and this problem still exists, but it has no impaction for my testing for now）

closed 2022-01-26 by woody-apple