project-chip/connectedhomeip #7972: `WriteHandler::ProcessAttributeDataList` likewise reads node ids from the TLV.  This does not make sense: writes are always targeted at the message recipient node (and the node id may not even be known, if the write is issued to a group).

yunhanw-google, 2021-06-28:


yunhanw-google, 2021-06-28:
Need to file ticket to check if we can remove nodeId from attribute path.

bzbarsky-apple, 2021-06-29:
In the spec, attribute path for a write on the encoding level should not include a node id, yes.  @mrjerryjohns

mrjerryjohns, 2021-07-01:
The spec is already compliant in this regard right? 

From the encoding spec:

> The Node ID MAY be omitted if the target node of the AttributePath matches DestNodeID in the message header, as it’s only relevant when interacting with proxies, where the subject Node ID in the path may not be the same as the Node ID of the proxy receiving the message. In this regard, this omission can be viewed as a compression scheme to reduce redundant information. Logically speaking, it is still implicitly present.

bzbarsky-apple, 2021-07-01:
This says the node id _MAY_ be omitted.  But it _MAY_ also be present, yes?

But for a write including it never makes any sense: you never write through a proxy, so for writes the node id in the path must always match the DestNodeID, right?  In which case, why allow to include it at all?

The difference is whether the receiver side needs to do "read the node id, if not present use message header, if present use that and compare to my own" vs "just use the message header".

bzbarsky-apple, 2022-09-07:
We are no longer reading node ids from TLV in WriteHandler.

closed 2022-09-07 by bzbarsky-apple