project-chip/connectedhomeip #7976: controller: APIs for CASE with previously unknown device on same fabric

mspang, 2021-06-28:
#### Problem

Currently to connect to via a device you must
- use PairDevice() to run the commissioning procedure, or
- Have a persisted local state from a previous successful commissioning session via PairDevice

With CASE it should be possible to instead establish a session with a previously unknown device on the same fabric. Currently, there is no API that would allow this.

 #### Proposed Solution

Enhance CHIPDeviceController to support establishing CASE session with a previously unknown device.

mspang, 2021-06-28:
@pan-apple FYI

tcarmelveilleux, 2021-08-03:
This is the fundamental assumption of CASE: that's what CASE is for. The nodes that will need to communicate over CASE to other nodes will, in large proportion of cases, not be the commissioners, but rather nodes that were told by other administrator nodes that they have a relations with device X (e.g. data model bindings, OTA request to OTA provider, time sync, etc)

mspang, 2021-09-21:
I made this hack for this,
https://github.com/mspang/connectedhomeip/commit/7d6534f0622f8c687e2e1d94a4758ec50d045610

Haven't opened a PR, it's very low road and unless other folks need this it isn't the solution we want to pursue.

holbrookt, 2021-09-21:
> I made this hack for this,
> [mspang@7d6534f](https://github.com/mspang/connectedhomeip/commit/7d6534f0622f8c687e2e1d94a4758ec50d045610)
> 
> Haven't opened a PR, it's very low road and unless other folks need this it isn't the solution we want to pursue.

Just want to say that I imagined the actual solution to this issue would not require use of `DeviceCommissioner`.

I would have considered using this to unblock OTA if not for that facet.

closed 2021-10-21 by woody-apple