project-chip/connectedhomeip #8000: Adjust mdns to support arbitrary number of fabrics.

cecille, 2021-06-29:
#### Problem
Need to adjust mdns to support the number of fabrics the device actually supports.

see: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/2551

 #### Proposed Solution
Template response sender, get setting from wherever that info comes into the SDK.

cecille, 2021-07-28:
There was also a request to support arbitrary numbers of fabrics for the more capable devices - see PR #8610. In general, this is something we will need to tackle on a larger scale - what's CHIP's philosophy on static vs. dynamic allocation and do we support multiple configs of this?

cecille, 2021-07-30:
Note: possible good solution would be to hook into the in-progress pool impl. https://github.com/project-chip/connectedhomeip/pull/8299

cecille, 2022-04-19:
@andy31415 - I had an issue opened for this. I think you've done this already? I know you were working on it. If so, would you mind closing this.

andy31415, 2022-05-19:
Maximum number of records returned is configurable - #16535 added that functionality.

andy31415, 2022-05-19:
Marking as closed: combination of #16535 and #16181

closed 2022-05-19 by andy31415