project-chip/connectedhomeip #8006: Make number of command/read/write client/handler configurable

yunhanw-google, 2021-06-29:
We can add macro in different platform to make it configurable.

Context:
> Why did we choose 4? Is there a better path forward here for constrained devices? @bzbarsky-apple ?

@bzbarsky-apple @woody-apple I think we have clear TODO "Make number of command/read/write client/handler configurable"

_Originally posted by @yunhanw-google in https://github.com/project-chip/connectedhomeip/issues/7967#issuecomment-870181926_

bzbarsky-apple, 2021-06-29:
Please see https://github.com/project-chip/connectedhomeip/pull/7967#issuecomment-870761884 for what I think should actually happen here.

erjiaqing, 2021-06-30:
> Please see [#7967 (comment)](https://github.com/project-chip/connectedhomeip/pull/7967#issuecomment-870761884) for what I think should actually happen here.

> * ReadHandler/WriteHandler/CommandHandler: 1, possibly allocated on the stack, not living in bss.  Questionable whether these objects should even exist in their current form, especially CommandHandler (which has a bunch of unused members, etc).

Given the fact that there might be trunked data in read / write, there might be more than 1 active ReadHandler and WriteHandler. And as for the case of CommandHandler, it could be 1 if all command transaction are short, however, if there are commands have some background processing, there might also be more than one CommandHandler.

> * Constrained device ReadClient/WriteClient/CommandClient: Depends on the device, between 0 and 1.

The same, we should be able to having more than one active read / write / command transactions since they are async operations even on constrained devices. We can let the project to choose the actual number, and yes, no Read/Write/Command client should be OK -- They can also be a switch to disable related code.

> * Unconstrained device ReadClient/WriteClient/CommandClient: infinite, bounded by available RAM (and maybe not even that, for devices with swap).

Agree, malloc and other similar operations is not limited, we don't have to limit the number of Read/Write/Command clients.

bzbarsky-apple, 2021-06-30:
> Given the fact that there might be trunked data in read / write

Ah, chunking.  Yes, ok, good point.

> however, if there are commands have some background processing, there might also be more than one CommandHandler.

Yes, but the point is this depends on the exact implementation details of the device and its commands, and it's those implementation details that should control the state we need to capture and where it's stored.

PR #13248 (2022-01-20): [IM] Move lifecycle management of WriteClient to application
#### Problem

Fixes #8006 -- when there is no WriteClient / CommandSender instances, the related code should be optimized out.

Unlike ReadClient and CommandSender, management of the lifecycle of WriteClient is handled by interaction model engine.

#### Change overview
Update the code to make it managed by application.

#### Testing
- sanitiser should make sure that there is no memory leak in this PR.
closed 2021-08-04 by yunhanw-google