project-chip/connectedhomeip #8010: Unify state machines in DeviceController

bzbarsky-apple, 2021-06-29:
#### Problem
Right now we have three somewhat different state machines in DeviceController:
* `mIsIPRendezvous` true, `CONFIG_USE_CLUSTERS_FOR_IP_COMMISSIONING` true
* `mIsIPRendezvous` true, `CONFIG_USE_CLUSTERS_FOR_IP_COMMISSIONING` false
* `mIsIPRendezvous` false.

One of these, the "`mIsIPRendezvous` true, `CONFIG_USE_CLUSTERS_FOR_IP_COMMISSIONING` false" one, is exercised in CI as far as I can tell.

These state machines perform important operations (like "tear down the existing connection") at totally different points in the process, send various notifications in different orders, and some of them perform some steps (e.g. sending a "Commissioning Complete" command) that others do not.

This leads to problems like #7982.

 #### Proposed Solution
1. Get rid of `CONFIG_USE_CLUSTERS_FOR_IP_COMMISSIONING`.  We should have only one way to do on-network rendezvous.
2. Unify the `mIsIPRendezvous` and `!mIsIPRendezvous` flows as much as possible.  This implies deciding who drives the commissioning process and decides when to move to the next step.  Right now for `mIsIPRendezvous` it's _mostly_ our internal state machine, but for BLE rendezvous it's the API consumer, as far as I can tell.  We need to figure out what API hooks consumers actually need here and implement them, then unify the various flows around those hooks.

@cecille @pan-apple @msandstedt @tcarmelveilleux

cecille, 2021-06-29:
Fully agree - the flag and separation of flows was intended to be removed once we got some testing in.

Suggestion for this - the BLE can easily go through the clusters once I add in the states to do network setup. However, I think we should consider separating the PASE connection from the start of commissioning. ie, instead of PairDevice kicking commissioning and opcreds automatically, that we have a secondary function like "AutomaticCommissioning" or something to that effect, that kicks the commissioning process. We can sequence the two in the python controller, or provide single-access to the function to establish the PASE connection for folks that are looking to tests flows manually. That way, there are no surprises for folks looking to bring up devices.

bzbarsky-apple, 2021-06-29:
> Test events use the other two.

Turns out this is not quite true; test events compile the python controller with `CONFIG_USE_CLUSTERS_FOR_IP_COMMISSIONING` set to false, apparently.

Past that, I agree that we just need to decide what the entrypoints/callbacks here should be and then do it...

bzbarsky-apple, 2022-03-31:
Both `CONFIG_USE_CLUSTERS_FOR_IP_COMMISSIONING` and `mIsIPRendezvous` are gone now.  @cecille is this fixed?  It seems fixed...

cecille, 2022-04-01:
Yes, this is done

closed 2022-04-01 by bzbarsky-apple