project-chip/connectedhomeip #8029: Add chunking support for IM write response

yunhanw-google, 2021-06-30:


mrjerryjohns, 2021-11-19:
This will be needed if we want ACL cluster to work correctly.

@bzbarsky-apple @tcarmelveilleux @mlepage-google

erjiaqing, 2021-12-07:
With some quick thought, and given the fact that the MCU should have the capability to handle chunked write, I think we may have a quick and dirty solution.

IIRC chained buffer cannot be used for handling chunked write, since reading the data between gaps can be a issue.
Then my idea is:
- maintain a buffer on the server side, and then just do exactly the same as buffered read.
- Only support only one chunked write transaction
- The client side also need to maintain a buffer, to hold the data since we need to hold the ownership of the data.

However, maintaining a buffer on the server side seems not affordable on a MCU, so another plan is:

- On the client side, maintain a shortlived buffer, to hold the ownership of data.
- On the server side, add the support to list AppendItem operation

This path is a bit simplier, however, each cluster must support handling the append operation, and it might have some issues related to message integrity(?).

Also, in this case, the client will receive more than one status report (infact, it is 1 + # of elements in the list), seems we can defer the response, and notify the application if the write transaction is finished.

Both the client and server should be aware of potential failures. Can we add another callback to write handler to notify the cluster it is safe to "commit" the data received?

PR #13641 (2022-02-10): [IM] Write Chunking Support
#### Problem
- Support write chunking
- Fixes #8029

#### Change overview
- Depend on https://github.com/project-chip/connectedhomeip/pull/13248
- Support write chunking
  - Implement ChunkedWriteCallback
  - Update ListLongOctetString
  - Update various cluster (test cluster, acl, server list)

#### Testing
- Python Tests  -- write a large attribute
- Test Suite -- writing a large attribute and verify it
- Unit test with various attribute length
closed 2022-02-10 by mrjerryjohns