project-chip/connectedhomeip #8043: Add Invoke Command Suppress response support

yunhanw-google, 2021-06-30:
#### Problem
there is spec conflict regarding suppress response flag in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/3536, after the above issue is resolved, we would decide how to add Invoke Command Suppress response support.

mrjerryjohns, 2022-01-31:
Was resolved a while back.

bzbarsky-apple, 2022-02-01:
No, this is not supported.  We parse it, but do absolutely nothing with it.

Rajashreekalmane, 2022-06-03:
@bzbarsky-apple @mrjerryjohns In Spec mentioned suppress response is mandatory for invokerequest and write request message 
[image]
Could you please let me know how to check whether DUT  suppress its response when TH sends Invoke/write request with suppress response parameter set to bool.

bzbarsky-apple, 2022-06-03:
> Could you please let me know how to check whether DUT suppress its response

Does the TH get a response?  If so, the DUT did not suppress it.

sankarselvam, 2022-06-07:
@bzbarsky-apple From my observation, DUT is not suppressing the response. 
@Rajashreekalmane , Can you please attach the logs?

bzbarsky-apple, 2022-06-08:
> @bzbarsky-apple From my observation, DUT is not suppressing the response.

Right, that's why this issue is open!

>  @Rajashreekalmane , Can you please attach the logs?

There is no need to attach logs here.

Rajashreekalmane, 2022-06-08:
@bzbarsky-apple what should I do in this case, Is this cert-blocker  or V1.0 out of scope issue.

bzbarsky-apple, 2022-06-08:
So far it's being treated as out of scope.

vivien-apple, 2022-12-16:
Just to make it clear, the `chip-tool` syntax for such a command is:
```
$ ./out/debug/standalone/chip-tool onoff toggle 0x12344321 1 --suppressResponse true
```

It does not work as of today because the stack does not supports suppress response. It can be observed that the value is set properly in the `InvokeRequestMessage` that is logged.

VaishaliAvhale, 2023-09-27:
With the above MR is this issue resolved?

bzbarsky-apple, 2023-09-28:
@VaishaliAvhale I don't know what you mean by "above MR".  This issue is not resolved.

VaishaliAvhale, 2023-09-28:
I was talking about https://github.com/project-chip/connectedhomeip/pull/27024 this MR.
I wanted to know if the issue has been resolved using the above MR.

VaishaliAvhale, 2023-09-28:
@bzbarsky-apple Please ignore the comments. Sorry for the confusion.