project-chip/connectedhomeip #8076: ESP32: Compilation of ipv6only-app and lock-app esp32 fails when pigweed is enabled.

sweetymhaiske, 2021-07-02:
#### Problem
* I am using esp-idf tag v4.3
* Compilation of ipv6only-app and lock-app fails when pigweed is enabled.
* Both apps are using the pigweed library for compiling services.
* Pigweed used `std=c++17` for compilation, hence it leads to an issue.
* Logs:

-std=gnu++17 -Os -DLWIP_IPV6_SCOPES=0 -DCHIP_HAVE_CONFIG_H -D_GNU_SOURCE -DESP_PLATFORM -I/home/sweety/Espressif/connectedhomeip/examples/ipv6only-app/esp32/build/config -g -fdiagnostics-color=always -fno-common **-std=c++17** -Wno-register -include /home/sweety/Espressif/connectedhomeip/examples/ipv6only-app/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_polyfill/language_features.h -DNDEBUG -DCHIP_HAVE_CONFIG_H=1 -I../../../../../../config/esp32/third_party/connectedhomeip/src/include -I../../../../../../config/esp32/third_party/connectedhomeip/src -Igen/include -I../../../../../../config/esp32/third_party/connectedhomeip/src/lib -I../../../../../../config/esp32/third_party/connectedhomeip/third_party/nlassert/repo/include -I../../../../../../config/esp32/third_party/connectedhomeip/third_party/nlio/repo/include -I../../../../../../config/esp32/third_party/connectedhomeip/third_party/nlfaultinjection/repo/include -c ../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandHandler.cpp -o obj/third_party/connectedhomeip/src/app/libCHIPDataModel.CommandHandler.cpp.o
In file included from /home/sweety/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/string.h:12,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/transport/raw/MessageHeader.h:27,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/app/util/basic-types.h:29,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandPathParams.h:21,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/app/Command.h:27,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandHandler.h:27,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandHandler.cpp:25:
`/home/sweety/Espressif/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:318:1: error: static assertion failed: portGET_ARGUMENT_COUNT() result does not match for 0 arguments
 _Static_assert(portGET_ARGUMENT_COUNT() == 0, "portGET_ARGUMENT_COUNT() result does not match for 0 arguments");`
 ^

sweetymhaiske, 2021-07-02:
@rgoliver

juejuegongsui, 2021-07-02:
很简单，去你的错误的文件下，修改该行代码，修改##__VA__ARGS__为0 __VA_OPT__(,) __VA_ARGS__，啥，你不认识中文吗？那就翻译吧，全是英文烦死了 @sweetymhaiske

wqx6, 2021-07-02:
I have added
 `chip_gn_arg_append("remove_default_configs"  "[\"//third_party/connectedhomeip/third_party/pigweed/repo/pw_build:cpp17\"]")` in `config/esp32/components/chip/CMakeLists.txt`, it solved the compilation failure in pigweed-app. I will figure out why it does not work in lock-app and ipv6only-app.

sweetymhaiske, 2021-07-19:
The respective PR for fixing this is merged. #8129

closed 2021-07-19 by sweetymhaiske