project-chip/connectedhomeip #8097: thread devices could not be provisioned using chip-tool

cjandhyala, 2021-07-02:
with ToT '2478e02ffb100ab17758972e32e4261d295e4bf6' , Couldn't provision the thread accessory (lock app running on nordic board). Looks like accessory couldn't parse the thread network dataset . Accessory log shows 
"
D: 354201 [ZCL]Received Cluster Command: Cluster=31 Command=6 Endpoint=0
D: 354207 [ZCL]Failed to parse Thread operational dataset: CHIP Error 4047 (0x00000FCF): Invalid argument
D: 354217 [ZCL]AddThreadNetwork: 19
D: 354220 [DMG]ICR moving to [AddCommand]
D: 354224 [IN]Secure message was encrypted: Msg ID 4
"

With the same dataset using python controller, we could provision the thread accessory and send zcl commands.

CMD used 
chaitanyajandhyala@chaitanyas-Mac-mini-2 chip-tool % sudo ./out/debug/chip-tool pairing ble-thread 0e080000000000000000000300000f35060004001fffe0020811111111222222220708fd23abec7ac0ed06051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f01021234041061e1206d2c2b46e079eb775f41fc72190c0402a0fff8 12344321 20202021 3000

Attached logs from chip-tool and accessory
[Nordic_lock_applog.txt](https://github.com/project-chip/connectedhomeip/files/6757385/Nordic_lock_applog.txt)
[chip_tool_log.txt](https://github.com/project-chip/connectedhomeip/files/6757387/chip_tool_log.txt)

bluebin14, 2021-07-03:
Note: should use "hex:" prefix for the dataset otherwise chip-tool considers it a string. That fixes the invalid argument error. Commissioning still fails though because chip-tool hardcodes the Thread networkId to 0123456789ABCDEF which does not match the above dataset. Also, chip-tool expects the hardcoded 5544332211 fabricId as the argument following the dataset.

bluebin14, 2021-07-05:
PR fix: https://github.com/project-chip/connectedhomeip/pull/8111

bluebin14, 2021-07-10:
@cjandhyala fixed and merged by PR #8245, see also notes above about correcting command line.