project-chip/connectedhomeip #8104: sourcing script activate.sh fails with error indicating an unexpected python version

AymenLA, 2021-07-04:
#### Problem
when trying to setup build environment, sourcing script "scripts/activate.sh" fails with error indicating that the Python version in unexpected : 

[image]

 #### Proposed Solution
at least, the error message should suggest to install another version of Python.

pin2-sakariya, 2021-07-05:
Agree with the proposed solution, I also got this error and no clue.

Below steps solved my issue, 

1. sudo apt install python3.8-dev
2. sudo apt install python3.8-venv
3. sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.8 <priority>

The print statement need to update in the below file,
Location: third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py, Line:128

AymenLA, 2021-12-05:
it's been 5 months and no answer for this issue
I'm closing it

although, I still have it :

[image]

AymenLA, 2021-12-05:
issue is not solved and no follow-up was made

closed 2021-12-05 by AymenLA