project-chip/connectedhomeip #8146: On test_event_4 branch, observing 'Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3'

Garik098, 2021-07-06:
While trying to establish pairing between controller and accessory on test_event_4 branch, I am observing the 'Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3' issue. I see that the BLE pairing is going through but the error is coming at CASE session establishment. 

Below are the logs,

Controller side log:
MacBook-Pro chip-tool % ./out/debug/chip-tool pairing ble-wifi WifiAP WifiPWD 0 20202021 3840
[1625613734595] [31013:1232304] CHIP: [DL] _Init
[1625613734595] [31013:1232304] CHIP: [IN] TransportMgr initialized
[1625613734595] [31013:1232304] CHIP: [DIS] Init admin pairing table with server storage
[1625613734595] [31013:1232304] CHIP: [IN] local node id is 0x000000000001B669
[1625613734596] [31013:1232304] CHIP: [ZCL] Using ZAP configuration...
[1625613734596] [31013:1232304] CHIP: [ZCL] deactivate report event
[1625613734596] [31013:1232304] CHIP: [CTL] Getting operational keys
[1625613734596] [31013:1232304] CHIP: [CTL] Getting root certificate for the controller from the issuer
[1625613734596] [31013:1232304] CHIP: [CTL] Generating operational certificate for the controller
[1625613734597] [31013:1232304] CHIP: [CTL] Getting intermediate CA certificate from the issuer
[1625613734597] [31013:1232304] CHIP: [CTL] GetIntermediateCACertificate returned 0
[1625613734597] [31013:1232304] CHIP: [CTL] Generating credentials
[1625613734597] [31013:1232304] CHIP: [CTL] Loaded credentials successfully
[1625613734598] [31013:1232308] CHIP: [BLE] NewConnection
[1625613734599] [31013:1232308] CHIP: [SC] Assigned local session key ID 1
[1625613734599] [31013:1232308] CHIP: [IN] Message appended to BLE send queue
[1625613734599] [31013:1232308] CHIP: [SC] Sent PBKDF param request
[1625613734652] [31013:1232309] CHIP: [BLE] CBManagerState: ON
[1625613739374] [31013:1232308] CHIP: [BLE] Connecting to device with discriminator: 3840
[1625613749058] [31013:1232308] CHIP: [IN] BleConnectionComplete: endPoint 0x100d1f950
[1625613749956] [31013:1232308] CHIP: [BLE] subscribe complete, ep = 0x100d1f950
[1625613750003] [31013:1232309] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1625613750003] [31013:1232309] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1625613750003] [31013:1232309] CHIP: [BLE] local and remote recv window size = 3
[1625613750003] [31013:1232309] CHIP: [IN] BLE EndPoint 0x100d1f950 Connection Complete
[1625613750678] [31013:1232309] CHIP: [EM] Received message of type 33 and protocolId 0 on exchange 44249
[1625613750679] [31013:1232309] CHIP: [SC] Received PBKDF param response
[1625613750679] [31013:1232309] CHIP: [SC] Sent spake2p msg1
[1625613751579] [31013:1232309] CHIP: [EM] Received message of type 35 and protocolId 0 on exchange 44249
[1625613751579] [31013:1232309] CHIP: [SC] Received spake2p msg2
[1625613751579] [31013:1232309] CHIP: [SC] Peer assigned session key ID 0
[1625613751579] [31013:1232309] CHIP: [SC] Sent spake2p msg3
[1625613751579] [31013:1232309] CHIP: [IN] New secure session created for device 0x0000000000BC5C01, key 0!!
[1625613751579] [31013:1232309] CHIP: [CTL] Remote device completed SPAKE2+ handshake
[1625613751579] [31013:1232309] CHIP: [CTL] Sending root certificate to the device
[1625613751579] [31013:1232309] CHIP: [DMG] ICR moving to [Initialize]
[1625613751579] [31013:1232309] CHIP: [DMG] ICR moving to [AddCommand]
[1625613751579] [31013:1232309] CHIP: [IN] Secure message was encrypted: Msg ID 0
[1625613751579] [31013:1232309] CHIP: [IN] Encrypted message 0x700001afec20 from 0x000000000001B669 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 44250.
[1625613751579] [31013:1232309] CHIP: [IN] Sending msg 0x700001afec20 to 0x0000000000BC5C01 at utc time: 942238083 msec
[1625613751579] [31013:1232309] CHIP: [IN] Sending secure msg on generic transport
[1625613751579] [31013:1232309] CHIP: [IN] Secure msg send status No Error
[1625613751579] [31013:1232309] CHIP: [DMG] ICR moving to [   Sending]
[1625613751579] [31013:1232309] CHIP: [CTL] Sent root certificate to the device
[1625613752613] [31013:1232308] CHIP: [IN] Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1625613752613] [31013:1232308] CHIP: [IN] Setting fabricID BC5C01 on admin.
[1625613752613] [31013:1232308] CHIP: [IN] Since admin was modified, persisting changes to KVS
[1625613752614] [31013:1232308] CHIP: [EM] Received message of type 9 and protocolId 5 on exchange 44250
[1625613752614] [31013:1232308] CHIP: [DMG] InvokeCommand =
[1625613752614] [31013:1232308] CHIP: [DMG] {
[1625613752614] [31013:1232308] CHIP: [DMG] 	CommandList =
[1625613752614] [31013:1232308] CHIP: [DMG] 	[
[1625613752614] [31013:1232308] CHIP: [DMG] 		CommandDataElement =
[1625613752614] [31013:1232308] CHIP: [DMG] 		{
[1625613752614] [31013:1232308] CHIP: [DMG] 			CommandPath =
[1625613752614] [31013:1232308] CHIP: [DMG] 			{
[1625613752614] [31013:1232308] CHIP: [DMG] 				EndpointId = 0x0,
[1625613752614] [31013:1232308] CHIP: [DMG] 				ClusterId = 0x3e,
[1625613752614] [31013:1232308] CHIP: [DMG] 				CommandId = 0xa1,
[1625613752614] [31013:1232308] CHIP: [DMG] 			},
[1625613752614] [31013:1232308] CHIP: [DMG] 			
[1625613752614] [31013:1232308] CHIP: [DMG] 			StatusElement =
[1625613752614] [31013:1232308] CHIP: [DMG] 			{
[1625613752614] [31013:1232308] CHIP: [DMG] 				GeneralCode = 0x0,
[... 784 lines cut ...]
[1625613760.637421][5538:5541] CHIP:BLE: Indication confirmation, 0xffff98045600
[1625613763.111866][5538:5541] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
[1625613763.112142][5538:5541] CHIP:BLE: Write request received debug 0xffff98045600
[1625613763.295995][5538:5538] CHIP:EM: Retransmit MsgId:00000000 Send Cnt 1
[1625613763.319829][5538:5538] CHIP:EM: Received message of type 48 and protocolId 0 on exchange 44255
[1625613763.319946][5538:5538] CHIP:EM: Sending Standalone Ack for MsgId:00000000
[1625613763.320109][5538:5538] CHIP:EM: Flushed pending ack for MsgId:00000000
[1625613763.320179][5538:5538] CHIP:IN: CASE Session establishment failed: CHIP Error 4042 (0x00000FCA): Invalid message type
[1625613763.551963][5538:5538] CHIP:EM: Retransmit MsgId:00000000 Send Cnt 2
[1625613763.573871][5538:5538] CHIP:EM: Received message of type 48 and protocolId 0 on exchange 44255
[1625613763.573968][5538:5538] CHIP:EM: Dropping unexpected message 00000000:48 ACDF MsgId:00000000
[1625613763.616261][5538:5538] CHIP:EM: Sending Standalone Ack for MsgId:00000000
[1625613763.807861][5538:5538] CHIP:EM: Retransmit MsgId:00000000 Send Cnt 3
[1625613763.827213][5538:5538] CHIP:EM: Received message of type 48 and protocolId 0 on exchange 44255
[1625613763.827311][5538:5538] CHIP:EM: Dropping unexpected message 00000000:48 ACDF MsgId:00000000
[1625613763.871596][5538:5538] CHIP:EM: Sending Standalone Ack for MsgId:00000000
[1625613764.064077][5538:5538] CHIP:EM: Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3
[1625613765.677563][5538:5541] CHIP:BLE: Indication confirmation, 0xffff98045600
[1625613768.151776][5538:5541] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
[1625613768.152051][5538:5541] CHIP:BLE: Write request received debug 0xffff98045600
[1625613768.649377][5538:5541] CHIP:DL: Bluez disconnected
[1625613768.649457][5538:5541] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1625613768.649655][5538:5538] CHIP:IN: Clearing BLE pending packets.
[1625613782.971471][5538:5542] CHIP:DL: Got IP address on interface: wlan0 IP: 192.168.0.49
[1625613782.972156][5538:5538] CHIP:DL: HandlePlatformSpecificBLEEvent 32770

yunhanw-google, 2021-07-07:
@pan-apple could you take a look?

wissn84, 2021-08-04:
In our test result, in specific Routers reproduce issue 100%. 
- Router Name (DLINK 8501 and IPTIME).

closed 2022-01-26 by woody-apple