project-chip/connectedhomeip #8186: Some Network Commissioning calls not completing as expected

SaileeG, 2021-07-08:
Accessory logs show "CHIP:ZCL: Failed to dispatch command" When Running a few Network Commissioning calls like:
- ScanNetworks: chip-device-ctrl > zcl NetworkCommissioning ScanNetworks 1 0 0 ssid=str:“UncharteredTerretory” breadcrumb=0 timeoutMs=1000
Controller logs:
```
chip-device-ctrl > zcl NetworkCommissioning ScanNetworks 1 0 0 ssid=str:“UncharteredTerretory” breadcrumb=0 timeoutMs=1000
[1625706531.221704][47355:47355] CHIP:IN: Encrypted message 0xffffc3ea0a80 from 0x000000000001B669 to 0x0000000000000001 of type 8 and protocolId 5 on exchange 39022.
[1625706531.221785][47355:47355] CHIP:IN: Sending msg 0xffffc3ea0a80 to 0x0000000000000001 at utc time: 7180154 msec
[1625706531.221830][47355:47355] CHIP:IN: Sending secure msg on generic transport
[1625706531.221998][47355:47355] CHIP:IN: Secure msg send status No Error
[1625706531.796047][47355:47362] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1625706531.796238][47355:47362] CHIP:EM: Received message of type 9 and protocolId 5 on exchange 39022
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1625706531.797048][47355:47362] CHIP:ZCL: DefaultResponse:
[1625706531.797115][47355:47362] CHIP:ZCL:   Transaction: 0xffff8efafab8
[1625706531.797169][47355:47362] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
Received command status response:
Container: 
    ProtocolId = 0
    ProtocolCode = 133
    EndpointId = 0
    ClusterId = 49
    CommandId = 0
    CommandIndex = 1

```
Accessory Logs:
```
[1625706531.253384][5280:5280] CHIP:IN: Secure transport received message destined to fabric 0, node 0x0000000000000001. Key ID 0
[1625706531.253514][5280:5280] CHIP:EM: Received message of type 8 and protocolId 5 on exchange 39022
[1625706531.253590][5280:5280] CHIP:EM: ec id: 39022, Delegate: 0x0xaaaacac8d860
[1625706531.253671][5280:5280] CHIP:DMG: ICR moving to [Initialize]
[1625706531.253744][5280:5280] CHIP:DMG: InvokeCommand =
[1625706531.253803][5280:5280] CHIP:DMG: {
[1625706531.253850][5280:5280] CHIP:DMG: 	CommandList =
[1625706531.253901][5280:5280] CHIP:DMG: 	[
[1625706531.253950][5280:5280] CHIP:DMG: 		CommandDataElement =
[1625706531.254011][5280:5280] CHIP:DMG: 		{
[1625706531.254066][5280:5280] CHIP:DMG: 			CommandPath =
[1625706531.254124][5280:5280] CHIP:DMG: 			{
[1625706531.254186][5280:5280] CHIP:DMG: 				EndpointId = 0x0,
[1625706531.254259][5280:5280] CHIP:DMG: 				ClusterId = 0x31,
[1625706531.254321][5280:5280] CHIP:DMG: 				CommandId = 0x0,
[1625706531.254385][5280:5280] CHIP:DMG: 			},
[1625706531.254461][5280:5280] CHIP:DMG: 			
[1625706531.254529][5280:5280] CHIP:DMG: 			CommandData = 
[1625706531.254587][5280:5280] CHIP:DMG: 			{
[1625706531.254713][5280:5280] CHIP:DMG: 				0x0 = [
[1625706531.254801][5280:5280] CHIP:DMG: 					0xe2, 0x80, 0x9c, 0x55, 0x6e, 0x63, 0x68, 0x61, 0x72, 0x74, 0x65, 0x72, 0x65, 0x64, 0x54, 0x65, 0x72, 0x72, 0x65, 0x74, 0x6f, 0x72, 0x79, 0xe2, 0x80, 0x9d, 0x0, 
[1625706531.254874][5280:5280] CHIP:DMG: 			]
[1625706531.254944][5280:5280] CHIP:DMG: 				0x1 = 0, 
[1625706531.255009][5280:5280] CHIP:DMG: 				0x2 = 1000, 
[1625706531.255070][5280:5280] CHIP:DMG: 			},
[1625706531.255133][5280:5280] CHIP:DMG: 		},
[1625706531.255198][5280:5280] CHIP:DMG: 		
[1625706531.255247][5280:5280] CHIP:DMG: 	],
[1625706531.255305][5280:5280] CHIP:DMG: 	
[1625706531.255350][5280:5280] CHIP:DMG: }
[1625706531.255418][5280:5280] CHIP:ZCL: Received Cluster Command: Cluster=31 Command=0 Endpoint=0
[1625706531.255493][5280:5280] CHIP:DMG: ICR moving to [AddCommand]
[1625706531.255540][5280:5280] CHIP:ZCL: Failed to dispatch command, 3/3 arguments parsed, TLVError=0, UnpackError=0 (last decoded tag = 2
[1625706531.255675][5280:5280] CHIP:IN: Secure message was encrypted: Msg ID 5
[1625706531.255740][5280:5280] CHIP:IN: Encrypted message 0xffffec5053c0 from 0x0000000000000001 to 0x000000000001B669 of type 9 and protocolId 5 on exchange 39022.
```
- UpdateNetwork: chip-device-ctrl > zcl NetworkCommissioning UpdateWiFiNetwork 1 0 0 ssid=str:<test_ssid> credentials=str:“<test_password> breadcrumb=0 timeoutMs=1000
Controller Logs:
```
chip-device-ctrl > zcl NetworkCommissioning UpdateWiFiNetwork 1 0 0 ssid=str:<test_ssid> credentials=str:“<test_password> breadcrumb=0 timeoutMs=1000
[1625706684.478281][47355:47355] CHIP:IN: Encrypted message 0xffffc3ea09f0 from 0x000000000001B669 to 0x0000000000000001 of type 8 and protocolId 5 on exchange 39023.
[1625706684.478337][47355:47355] CHIP:IN: Sending msg 0xffffc3ea09f0 to 0x0000000000000001 at utc time: 7333411 msec
[1625706684.478363][47355:47355] CHIP:IN: Sending secure msg on generic transport
[1625706684.478482][47355:47355] CHIP:IN: Secure msg send status No Error

[1625706685.068177][47355:47362] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1625706685.068364][47355:47362] CHIP:EM: Received message of type 9 and protocolId 5 on exchange 39023
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1625706685.069228][47355:47362] CHIP:ZCL: DefaultResponse:
[1625706685.069301][47355:47362] CHIP:ZCL:   Transaction: 0xffff8efafab8
[1625706685.069356][47355:47362] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
Received command status response:
Container: 
    ProtocolId = 0
    ProtocolCode = 133
    EndpointId = 0
    ClusterId = 49
    CommandId = 4
    CommandIndex = 1
```
Accessory Logs:
```
[1625706684.486560][5280:5280] CHIP:EM: Received message of type 8 and protocolId 5 on exchange 39023
[1625706684.486690][5280:5280] CHIP:EM: ec id: 39023, Delegate: 0x0xaaaacac8d860
[1625706684.486776][5280:5280] CHIP:DMG: ICR moving to [Initialize]
[1625706684.486850][5280:5280] CHIP:DMG: InvokeCommand =
[1625706684.486907][5280:5280] CHIP:DMG: {
[1625706684.486953][5280:5280] CHIP:DMG: 	CommandList =
[1625706684.487004][5280:5280] CHIP:DMG: 	[
[1625706684.487062][5280:5280] CHIP:DMG: 		CommandDataElement =
[1625706684.487133][5280:5280] CHIP:DMG: 		{
[1625706684.487199][5280:5280] CHIP:DMG: 			CommandPath =
[1625706684.487268][5280:5280] CHIP:DMG: 			{
[1625706684.487350][5280:5280] CHIP:DMG: 				EndpointId = 0x0,
[1625706684.487451][5280:5280] CHIP:DMG: 				ClusterId = 0x31,
[1625706684.487548][5280:5280] CHIP:DMG: 				CommandId = 0x4,
[1625706684.487620][5280:5280] CHIP:DMG: 			},
[1625706684.487694][5280:5280] CHIP:DMG: 			
[1625706684.487750][5280:5280] CHIP:DMG: 			CommandData = 
[1625706684.487808][5280:5280] CHIP:DMG: 			{
[1625706684.487866][5280:5280] CHIP:DMG: 				0x0 = [
[1625706684.487966][5280:5280] CHIP:DMG: 					0xe2, 0x80, 0x9c, 0x50, 0x69, 0x78, 0x65, 0x6c, 0x5f, 0x35, 0x37, 0x32, 0x31, 0xe2, 0x80, 0x9d, 0x0, 
[1625706684.488059][5280:5280] CHIP:DMG: 			]
[1625706684.488133][5280:5280] CHIP:DMG: 				0x1 = [
[1625706684.488231][5280:5280] CHIP:DMG: 					0xe2, 0x80, 0x9c, 0x73, 0x61, 0x69, 0x6c, 0x65, 0x65, 0x31, 0x32, 0xe2, 0x80, 0x9d, 0x0, 
[1625706684.488318][5280:5280] CHIP:DMG: 			]
[1625706684.488397][5280:5280] CHIP:DMG: 				0x2 = 0, 
[1625706684.488489][5280:5280] CHIP:DMG: 				0x3 = 1000, 
[1625706684.488577][5280:5280] CHIP:DMG: 			},
[1625706684.488655][5280:5280] CHIP:DMG: 		},
[1625706684.488739][5280:5280] CHIP:DMG: 		
[1625706684.488800][5280:5280] CHIP:DMG: 	],
[1625706684.488871][5280:5280] CHIP:DMG: 	
[1625706684.488927][5280:5280] CHIP:DMG: }
[1625706684.489010][5280:5280] CHIP:ZCL: Received Cluster Command: Cluster=31 Command=4 Endpoint=0
[1625706684.489102][5280:5280] CHIP:DMG: ICR moving to [AddCommand]
[1625706684.489150][5280:5280] CHIP:ZCL: Failed to dispatch command, 4/4 arguments parsed, TLVError=0, UnpackError=0 (last decoded tag = 3
[1625706684.489304][5280:5280] CHIP:IN: Secure message was encrypted: Msg ID 6
[1625706684.489376][5280:5280] CHIP:IN: Encrypted message 0xffffec5053c0 from 0x0000000000000001 to 0x000000000001B669 of type 9 and protocolId 5 on exchange 39023.
```

Similar issues seen when running:
- DisableNetwork
- RemoveNetwork

 * Branch tested with: Master
 * Setup: 2 Raspi(1 Raspi running Controller, 1 Raspi running Linux Lighting app)

 * steps to reproduce
 - from the chip device controller cli run connect -ble
 - Run Add and Enable Wifi network commissioning commands
 - the try above mentioned Network Commissioning Commands

SaileeG, 2021-07-08:
Also if these Cluster commands are run using ZclSend API they show that the command passed(do not raise an Error)
zcl_scan_network: {'status': 0, 'result': 'Container: \n    ProtocolId = 0\n    ProtocolCode = 65535\n    EndpointId = 0\n    ClusterId = 49\n    CommandId = 0\n    CommandIndex = 1', 'error': None}
zcl_update_network: {'status': 0, 'result': 'Container: \n    ProtocolId = 0\n    ProtocolCode = 65535\n    EndpointId = 0\n    ClusterId = 49\n    CommandId = 4\n    CommandIndex = 1', 'error': None}
zcl_disable_network: {'status': 0, 'result': 'Container: \n    ProtocolId = 0\n    ProtocolCode = 65535\n    EndpointId = 0\n    ClusterId = 49\n    CommandId = 14\n    CommandIndex = 1', 'error': None}
zcl_remove_network: {'status': 0, 'result': 'Container: \n    ProtocolId = 0\n    ProtocolCode = 65535\n    EndpointId = 0\n    ClusterId = 49\n    CommandId = 10\n    CommandIndex = 1', 'error': None}

Sample Accessory logs for one of the calls:
[1625701668.371766][1985] CHIP:EM: Received message of type 8 and protocolId 327680 on exchange 39026
[1625701668.371838][1985] CHIP:EM: ec id: 39026, Delegate: 0xdaa04a40
[1625701668.371909][1985] CHIP:DMG: ICR moving to [Initialize]
[1625701668.371975][1985] CHIP:DMG: InvokeCommand =
[1625701668.372028][1985] CHIP:DMG: {
[1625701668.372075][1985] CHIP:DMG: 	CommandList =
[1625701668.372132][1985] CHIP:DMG: 	[
[1625701668.372186][1985] CHIP:DMG: 		CommandDataElement =
[1625701668.372257][1985] CHIP:DMG: 		{
[1625701668.372314][1985] CHIP:DMG: 			CommandPath =
[1625701668.372385][1985] CHIP:DMG: 			{
[1625701668.372457][1985] CHIP:DMG: 				EndpointId = 0x0,
[1625701668.372543][1985] CHIP:DMG: 				ClusterId = 0x31,
[1625701668.372626][1985] CHIP:DMG: 				CommandId = 0x4,
[1625701668.372708][1985] CHIP:DMG: 			},
[1625701668.372782][1985] CHIP:DMG: 			
[1625701668.372840][1985] CHIP:DMG: 			CommandData = 
[1625701668.372909][1985] CHIP:DMG: 			{
[1625701668.372980][1985] CHIP:DMG: 				0x0 = [
[1625701668.373076][1985] CHIP:DMG: 					0xe2, 0x80, 0x9c, 0x50, 0x69, 0x78, 0x65, 0x6c, 0x5f, 0x35, 0x37, 0x32, 0x31, 0xe2, 0x80, 0x9d, 0x0, 
[1625701668.373161][1985] CHIP:DMG: 			]
[1625701668.373234][1985] CHIP:DMG: 				0x1 = [
[1625701668.373327][1985] CHIP:DMG: 					0xe2, 0x80, 0x9c, 0x73, 0x61, 0x69, 0x6c, 0x65, 0x65, 0x31, 0x32, 0xe2, 0x80, 0x9d, 0x0, 
[1625701668.373410][1985] CHIP:DMG: 			]
[1625701668.373482][1985] CHIP:DMG: 				0x2 = 0, 
[1625701668.373568][1985] CHIP:DMG: 				0x3 = 1000, 
[1625701668.373670][1985] CHIP:DMG: 			},
[1625701668.373744][1985] CHIP:DMG: 		},
[1625701668.373812][1985] CHIP:DMG: 		
[1625701668.373864][1985] CHIP:DMG: 	],
[1625701668.373927][1985] CHIP:DMG: 	
[1625701668.373976][1985] CHIP:DMG: }
[1625701668.374047][1985] CHIP:ZCL: Received Cluster Command: Cluster=31 Command=4 Endpoint=0
[1625701668.374127][1985] CHIP:DMG: ICR moving to [AddCommand]
[1625701668.374180][1985] CHIP:ZCL: Failed to dispatch command, 4/4 arguments parsed, TLVError=0, UnpackError=0 (last decoded tag = 3
[1625701668.374320][1985] CHIP:IN: Secure message was encrypted: Msg ID 8
[1625701668.374381][1985] CHIP:IN: Sending msg from 0x0000000000000001 to 0x000000000001B669 at utc time: 2325335 msec
[1625701668.374433][1985] CHIP:IN: Sending secure msg on generic transport
[1625701668.374596][1985] CHIP:IN: Secure msg send status No Error
[1625701668.374702][1985] CHIP:DMG: ICR moving to [   Sending]
[1625701668.374750][1985] CHIP:DMG: ICR moving to [Uninitiali]
[1625701668.386689][1985] CHIP:DIS: Replying to DNS-SD service listing request

SaileeG, 2021-07-08:
Blocked by: https://github.com/project-chip/connectedhomeip/issues/5619

erjiaqing, 2021-12-21:
Please verify if it works with Linux all-clusters-app
Follow up: https://github.com/project-chip/connectedhomeip/issues/13170

closed 2021-12-21 by erjiaqing