project-chip/connectedhomeip #8207: CASE session fails when commissioning second device in a row

kkasperczyk-no, 2021-07-08:
#### Problem
When commissioning the accessory device for the first time CASE session is established fine, however after factory resetting the device and commissioning it once again with different node ID it fails on the controller side due to no memory error.

Steps to reproduce:
1. Commission the accessory with Python CHIP controller and some node ID (e.g. 1):

`connect -ble 3840 20202021 1
zcl NetworkCommissioning AddThreadNetwork 1 0 0 operationalDataset=hex:000300000f02081111111122222222051000112233445566778899aabbccddeeff01021234 breadcrumb=0 timeoutMs=1000
zcl NetworkCommissioning EnableNetwork 1 0 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000
close-ble`

2. Resolve IP address with `resolve 5544332211 1` - it will succeed.
3. Factory reset the device.
4. Commission the accessory once again with different node ID (e.g. 2):

`connect -ble 3840 20202021 2
zcl NetworkCommissioning AddThreadNetwork 2 0 0 operationalDataset=hex:000300000f02081111111122222222051000112233445566778899aabbccddeeff01021234 breadcrumb=0 timeoutMs=1000
zcl NetworkCommissioning EnableNetwork 2 0 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000
close-ble`

5. Resolve IP address with `resolve 5544332211 2` again - it will fail.

Logs on the accessory side:
`'I: 72134 [IN]CASE Server received SigmaR1 message. Starting handshake. EC 0x20003898\r\n'
'D: 72144 [SC]Waiting for SigmaR1 msg\r\n'
'D: 72148 [SC]Received SigmaR1 msg\r\n'
'D: 72151 [SC]Peer assigned session key ID 12\r\n'
'D: 72209 [SC]Sent SigmaR2 msg\r\n'
'I: 72310 [EM]Received message of type 63 and protocolId 0 on exchange 56472\r\n'
'E: 72317 [SC]Received error (255) during CASE pairing process\r\n'
'I: 72324 [IN]CASE Session establishment failed: CHIP Error 4084 (0x00000FF4): Invalid CASE parameter\r\n'`

Logs on the controller side:
`[1625463797.529033][459:658] CHIP:DIS: Node ID resolved for 0x0000000000000009 to [fd11:22::7cf8:11af:95ed:a205]:11097
[1625463797.529059][459:658] CHIP:CTL: SyncSetKeyValue on PairedDevice9
Node address has been updated
[1625463797.529367][459:658] CHIP:CTL: OperationalDiscoveryComplete for device ID 9
[1625463797.529378][459:658] CHIP:CTL: Enabling CASE session establishment for the device
[1625463797.529391][459:658] CHIP:CTL: SyncSetKeyValue on PairedDevice9
[1625463797.529395][459:658] CHIP:IN: Marking old secure session for device 0x0000000000000009 as expired
[1625463797.529409][459:658] CHIP:CTL: SyncSetKeyValue on PairedDevice9
[1625463797.529415][459:658] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1625463797.529698][459:658] CHIP:SC: Sent SigmaR1 msg
[1625463797.747842][459:658] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 56458
[1625463797.747889][459:658] CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1625463797.747901][459:658] CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
[1625463797.747929][459:658] CHIP:SC: Received SigmaR2 msg
[1625463797.747939][459:658] CHIP:SC: Peer assigned session key ID 1
[1625463797.748543][459:658] CHIP:EM: Piggybacking Ack for MsgId:00000000 with msg
[1625463797.748686][459:658] CHIP:CTL: Device connection failed. Error CHIP Error 4011 (0x00000FAB): No memory
Failed to commission: 4011`

 #### Proposed Solution
<suggested fix, suggested enhancement>

pan-apple, 2021-07-08:
@kkasperczyk-no, thank you for the detailed analysis and the logs.

I think the main issue is that we are loading the peer's operational cert in the local credentials set to validate it. We should update the validation code to not load the peer's cert in the local set (so that the local credentials are not mutated with each CASE session).

Each device (including controllers) will have one set of credentials per fabric. So a controller should have only one of these (per fabric) to communicate with any number of peer devices (on the same fabric).

cc: @jpk233

jpk233, 2021-07-09:
> @kkasperczyk-no, thank you for the detailed analysis and the logs.
>
> I think the main issue is that we are loading the peer's operational cert in the local credentials set to validate it. We should [...]

I agree with this, and after testing this solution it seems to resolve the issue. Will roll the fix into our open CASE changes.

pan-apple, 2021-07-09:
@jpk233 , it'll make it harder to support concurrent CASE session setups in future.

pan-apple, 2021-07-13:
> @jpk233 , it'll make it harder to support concurrent CASE session setups in future.

Filed https://github.com/project-chip/connectedhomeip/issues/8342 to track this.

tcarmelveilleux, 2021-07-27:
The cost of loading necessary credentials (in time), versus the cost of keeping them in memory (in space) is mis-balanced.

Loading a cert is somewhat "costly" (but not egregiously so), in comparison with keeping the decoded certs in memory for the entire duration of a session establishment.

Overall, the operational credentials are used in *a single message handling step*.

Here are the messages and their use of credentials:

- Sending Sigma1: requires access to root public key from root cert. Could argue for separately caching root public keys away from root certs, after validation of install of a root.
- Processing Sigma1: requires access to all root public keys and the fabric+node IDs of each NOC. Again, this is data that could have been pre-cached at NOC install in a "secondary" way, since a fabric+node ID is 128 bits total
- Sending Sigma2: requires sending NOC/ICAC array for only current NOC, and signing with NOC's operational private key (i.e. does not require loading certs for the NOC). The NOC/ICAC data can be copied as is without actually parsing anything.
- Processing Sigma2: requires validating NOC/ICAC/RCAC chain. This is the only time a full validation is needed. The NOC and ICAC are actually already in memory in the incoming message, so there is only need to load the RCAC cert (unless you assume it's prevalidated, which it is, then it's not needed to actually load it as long as the root public key is available to validate the ICAC
- Sending Sigma3/Processing Sigma3: same as for Sigma2 in terms of cert access, but reverse roles of which peer has to validate.

Overall, the lifetime requirement of access for NOC chain is *a single message processing or construction step*, and the rest only needs access to the root public key of the shared fabric determined at Sigma1. I would recommend refactoring CASESession.cpp to only load the needed certificates at the right time, and also to consider caching some metadata about the installed certificates so that if only one item (e.g. root public key) is needed, it can be more easily looked-up than having the load the "certificate sets". 

The current methodology of access will not scale to large numbers of fabrics for multi-admin, even if single CASE session is allowed at one time.

PR #8137 (2021-07-28): CASE spec refresh
#### Problem
* CASE does not currently follow the current spec with message handling.

* Fixes #5934 (along with the previously submitted CASE TLV changes)
* Fixes #8207

#### Change overview
* Added TrustedRootId parameter to CASESession’s EstablishSession method

* Added new methods to Find and Generate Destination IDs (New CASE Spec updates)

* Updated some CASESession method’s parameters that were using raw input buffers with ByteSpan and MutableByteSpan objects

* Updated CASESession Nonce and Info vectors (new CASE Spec updates)

* Updated SigmaR1 message construction m [... cut]
closed 2021-07-28 by hawk248