project-chip/connectedhomeip #8227: Add ability to advertise multiple node IDs on mDNS

sagar-apple, 2021-07-08:
#### Problem
Our accessories are currently unable to advertise more than one node ID on mDNS. 
This means that only the first Fabric is able to discover an accessory, all subsequent Fabrics won't be able to discover the accessory if they assigned a different node ID (spec compliant requirement is to advertise Fabric+NodeID but that's a separate issue for now). 

 #### Proposed Solution
On the accessory side we to iterate over all the provisioned operational certs and advertise all assigned node IDs.

sagar-apple, 2021-07-26:
That's a good question. 

>  Currently, I believe all in-tree examples advertise hard-coded identities.

So I'm not sure this is exactly right, the FabricID is hardcoded but they advertise the NodeID assigned to them by the commissioner. This is at least true for the M5Stack. 

I think the fix here should allow accessories to advertise the identities stored in their certs, i.e the designated Fabrics and the NodeID on those Fabrics.

tcarmelveilleux, 2021-07-26:
Actually, the advertisements should not contain the Fabric ID from the NOC, but rather the "compressed fabric identifier", which is derived from the root public key for the NOC's RCAC and the fabric ID.

msandstedt, 2021-07-26:
> That's a good question.
> 
> > Currently, I believe all in-tree examples advertise hard-coded identities.
> 
> So I'm not sure this is exactly right, the FabricID is hardcoded but they advertise the NodeID assigned to them by the commissioner. This is at least true for the M5Stack.
> 

Did this change sometime after July 16th?  I'm seeing the m5stack still advertising `0000000000000000-0000000000BC5C01`.

msandstedt, 2021-07-26:
> Did this change sometime after July 16th? I'm seeing the m5stack still advertising `0000000000000000-0000000000BC5C01`.

@sagar-apple , @cecille 

Unfortunately this is not quite the case:

```
    // TODO: Remove temporary code once AddOptCert is implemented
    if (packetHeader.GetDestinationNodeId().HasValue())
    {
        localNodeId = packetHeader.GetDestinationNodeId().Value();
        if (localNodeId != kUndefinedNodeId && fabric->GetNodeId() != localNodeId)
        {
            fabric->SetNodeId(localNodeId);
            ChipLogProgress(Inet, "Setting nodeID %" PRIX64 " on fabric.", fabric->GetNodeId());
            modifiedFabric = true;
        }
    }
```

Accessories are still inferring node ID from random incoming traffic, not certs.  Likewise, commissioners (at least the in-tree ones) are also ignoring node IDs embedded in certs, and instead assuming the node IDs they pass to the OpCert delegate will be honored (not a reasonable expectation).  So in effect, this only works because we have bugs / hacks on both sides.  Any code that tries to do the correct thing and expects node IDs to actually come from certs is likely to be incompatible.

The description text here does infer the accessory-side part of the issue I'm raising will be solved when we implement the proposed solution:

> On the accessory side we to iterate over all the provisioned operational certs and advertise all assigned node IDs.

So I will leave this at that and assume this is the case.

CC @samadDotDev, @kianooshkarami , @z-michel, @saurabhst

closed 2021-08-12 by bzbarsky-apple