project-chip/connectedhomeip #8235: Setting up OTBR on standalone Raspi is failing

ramakrishnanandyala, 2021-07-08:
Following this provided link to setup my Raspi-4 as OTBR 
https://groups.zigbee.org/wg/chip-cert-sg/document/23206

Device Details : 
Raspi - 4
Ubuntu - 21.04

Cloned "https://github.com/openthread/ot-br-posix" and ran  ./script/bootstrap but I see "libjsoncpp-dev" package is not found

Below is the error
+ sudo apt-get install --no-install-recommends -y libjsoncpp1 libjsoncpp-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libjsoncpp1

I also tried "sudo apt-get install libjsoncpp1" but didn't help

simonlingoogle, 2021-07-09:
Found the bug at https://bugs.launchpad.net/ubuntu/+source/libjsoncpp/+bug/1928695

I think we only need to install `libjsoncpp-dev` on Ubuntu since it will automatically install the correct `libjsoncpp<Version>` package.

**Ubuntu 21.04:**
```
# apt install libjsoncpp-dev
The following NEW packages will be installed:
  libjsoncpp-dev libjsoncpp24
```

**Ubuntu 18.04:**
```
# apt install libjsoncpp-dev
The following NEW packages will be installed:
  libjsoncpp-dev libjsoncpp1
```

@ramakrishnanandyala As a temporary workaround, you can remove `libjsoncpp1` from the install script.

ramakrishnanandyala, 2021-07-09:
@simonlingoogle with temporary workaround it worked

closed 2021-07-22 by jwhui