project-chip/connectedhomeip #8244: stack overflow in task CHIP while adding M5stack CHIP accessories to home

shah-kartik, 2021-07-09:
***ERROR*** A stack overflow in task CHIP has been detected.

Backtrace:0x40090a8e:0x3ffddb30 0x40091059:0x3ffddb50 0x40091306:0x3ffddb70 0x40091edc:0x3ffddbf0 0x400913fc:0x3ffddc10 0x400913b2:0x0df15cce |<-CORRUPTED
0x40090a8e: panic_abort at /Users/apple/tools/esp-idf/components/esp_system/panic.c:341

0x40091059: esp_system_abort at /Users/apple/tools/esp-idf/components/esp_system/system_api.c:106

0x40091306: vApplicationStackOverflowHook at /Users/apple/tools/esp-idf/components/freertos/xtensa/port.c:474

0x40091edc: vTaskSwitchContext at /Users/apple/tools/esp-idf/components/freertos/tasks.c:2770

0x400913fc: _frxt_dispatch at /Users/apple/tools/esp-idf/components/freertos/xtensa/portasm.S:431

0x400913b2: _frxt_int_exit at /Users/apple/tools/esp-idf/components/freertos/xtensa/portasm.S:231

ELF file SHA256: 08a43fa0300f1773

Rebooting...
ets Jul 29 2019 12:21:46

ESP-IDF Setup:
-----------
M5Stack connected to Mac mini 

git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout release/v4.2
git fetch --all
git reset --hard d9ec7df39
git submodule update --init
pip install virtualenv

full logs attached here:
[M5Stack_CHIP_StackOverflow_Logs.pdf](https://github.com/project-chip/connectedhomeip/files/6790007/M5Stack_CHIP_StackOverflow_Logs.pdf)

MarkMk1, 2021-07-26:
Also experienced this issue with a devkitc, identical error message

npc-mark, 2021-07-26:
Also experienced this issue with a devkitc, identical error message

dhrishi, 2021-07-26:
@shah-kartik @MarkMk1 CHIP uses ESP-IDF v4.3 release tag. Please refer to the instructions in the all-clusters-app.

If you are using IDF v4.3 with latest CHIP code and facing the issue, please share the console logs. Is this reproducible everytime at your end with the all-clusters-app? Can you please share any other steps/deviations from the default example along with the sdkconfig file?

npc-mark, 2021-07-26:
You're correct in my case. Other issues with 4.3 (I was running 4.2), but there's no stack overflow at least

closed 2022-01-26 by woody-apple