project-chip/connectedhomeip #8273: ZCLread Basic Attributes returning empty Value or not a valid AttributeValue

SaileeG, 2021-07-10:
When running ZCLread for Basic cluster not all Attributes are returning Value or valid AttributeValue

List of Attributes returning Null/Empty Value filed:
- ProductName
- VendorName
- UserLabel
- Location
- SoftwareVersionString
- ManufacturingDate
- PartNumber
- ProductURL
- ProductLabel
- SerialNumber

Controller logs:
**VendorName:**
```
chip-device-ctrl > zclread Basic VendorName 1 0 0
[1625878813.435173][113310:113310] CHIP:IN: Encrypted message 0x25ce2138 from 0x000000000001B669 to 0x0000000000000001 of type 2 and protocolId 5 on exchange 56414.
[1625878813.435280][113310:113310] CHIP:IN: Sending msg 0x25ce2138 to 0x0000000000000001 at utc time: 22133786 msec
[1625878813.435340][113310:113310] CHIP:IN: Sending secure msg on generic transport
[1625878813.435559][113310:113310] CHIP:IN: Secure msg send status No Error
[1625878813.458134][113310:113317] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 3
[1625878813.458292][113310:113317] CHIP:EM: Received message of type 5 and protocolId 5 on exchange 56414
[1625878813.458525][113310:113317] CHIP:ZCL: ReadAttributesResponse:
[1625878813.458776][113310:113317] CHIP:ZCL:   ClusterId: 0x0028
[1625878813.458826][113310:113317] CHIP:ZCL:   attributeId: 0x0001
[1625878813.458878][113310:113317] CHIP:ZCL:   status: Success                (0x0000)
[1625878813.458937][113310:113317] CHIP:ZCL:   attribute TLV Type: 0x0c
[1625878813.458980][113310:113317] CHIP:ZCL:   value: 
[1625878813.459054][113310:113317] CHIP:ZCL:   attributeValue: (span of length 0) 
[1625878813.459424][113310:113317] CHIP:IN: Encrypted message 0xffff82fddef0 from 0x000000000001B669 to 0x0000000000000001 of type 16 and protocolId 0 on exchange 56414.
[1625878813.459535][113310:113317] CHIP:IN: Sending msg 0xffff82fddef0 to 0x0000000000000001 at utc time: 22133811 msec
[1625878813.459585][113310:113317] CHIP:IN: Sending secure msg on generic transport
chip-device-ctrl > [1625878813.459792][113310:113317] CHIP:IN: Secure msg send status No Error
```

**ProductName:**
```
chip-device-ctrl > zclread Basic ProductName 1 0 0
[1625878900.304690][113310:113310] CHIP:IN: Encrypted message 0x25ce2138 from 0x000000000001B669 to 0x0000000000000001 of type 2 and protocolId 5 on exchange 56415.
[1625878900.304785][113310:113310] CHIP:IN: Sending msg 0x25ce2138 to 0x0000000000000001 at utc time: 22220656 msec
[1625878900.304838][113310:113310] CHIP:IN: Sending secure msg on generic transport
[1625878900.305043][113310:113310] CHIP:IN: Secure msg send status No Error
[1625878900.320995][113310:113317] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 3
[1625878900.321125][113310:113317] CHIP:EM: Received message of type 5 and protocolId 5 on exchange 56415
[1625878900.321310][113310:113317] CHIP:ZCL: ReadAttributesResponse:
[1625878900.321388][113310:113317] CHIP:ZCL:   ClusterId: 0x0028
[1625878900.321423][113310:113317] CHIP:ZCL:   attributeId: 0x0003
[1625878900.321452][113310:113317] CHIP:ZCL:   status: Success                (0x0000)
[1625878900.321482][113310:113317] CHIP:ZCL:   attribute TLV Type: 0x0c
[1625878900.321512][113310:113317] CHIP:ZCL:   value: 
[1625878900.321547][113310:113317] CHIP:ZCL:   attributeValue: (span of length 0) 
[1625878900.321762][113310:113317] CHIP:IN: Encrypted message 0xffff82fddef0 from 0x000000000001B669 to 0x0000000000000001 of type 16 and protocolId 0 on exchange 56415.
[1625878900.321813][113310:113317] CHIP:IN: Sending msg 0xffff82fddef0 to 0x0000000000000001 at utc time: 22220673 msec
[1625878900.321841][113310:113317] CHIP:IN: Sending secure msg on generic transport
chip-device-ctrl > [1625878900.322011][113310:113317] CHIP:IN: Secure msg send status No Error
```

**UserLabel:**
```
chip-device-ctrl > zclread Basic UserLabel 1 0 0
[1625878968.489257][113310:113310] CHIP:IN: Encrypted message 0x25ce2138 from 0x000000000001B669 to 0x0000000000000001 of type 2 and protocolId 5 on exchange 56417.
[1625878968.489342][113310:113310] CHIP:IN: Sending msg 0x25ce2138 to 0x0000000000000001 at utc time: 22288840 msec
[1625878968.489387][113310:113310] CHIP:IN: Sending secure msg on generic transport
[1625878968.489579][113310:113310] CHIP:IN: Secure msg send status No Error
[1625878968.500133][113310:113317] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 3
[1625878968.500260][113310:113317] CHIP:EM: Received message of type 5 and protocolId 5 on exchange 56417
[1625878968.500446][113310:113317] CHIP:ZCL: ReadAttributesResponse:
[1625878968.500486][113310:113317] CHIP:ZCL:   ClusterId: 0x0028
[... 95 lines cut ...]
[1625879155.808409][113310:113317] CHIP:IN: Sending secure msg on generic transport
chip-device-ctrl > [1625879155.808633][113310:113317] CHIP:IN: Secure msg send status No Error
```

**PartNumber**
```
chip-device-ctrl > zclread Basic PartNumber 1 0 0
[1625879265.385086][113310:113310] CHIP:IN: Encrypted message 0x25ce2138 from 0x000000000001B669 to 0x0000000000000001 of type 2 and protocolId 5 on exchange 56424.
[1625879265.385190][113310:113310] CHIP:IN: Sending msg 0x25ce2138 to 0x0000000000000001 at utc time: 22585736 msec
[1625879265.385251][113310:113310] CHIP:IN: Sending secure msg on generic transport
[1625879265.385479][113310:113310] CHIP:IN: Secure msg send status No Error
[1625879265.401691][113310:113317] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 3
[1625879265.401822][113310:113317] CHIP:EM: Received message of type 5 and protocolId 5 on exchange 56424
[1625879265.402006][113310:113317] CHIP:ZCL: ReadAttributesResponse:
[1625879265.402042][113310:113317] CHIP:ZCL:   ClusterId: 0x0028
[1625879265.402076][113310:113317] CHIP:ZCL:   attributeId: 0x000c
[1625879265.402111][113310:113317] CHIP:ZCL:   status: Success                (0x0000)
[1625879265.402157][113310:113317] CHIP:ZCL:   attribute TLV Type: 0x0c
[1625879265.402204][113310:113317] CHIP:ZCL:   value: 
[1625879265.402296][113310:113317] CHIP:ZCL:   attributeValue: (span of length 0) 
[1625879265.402568][113310:113317] CHIP:IN: Encrypted message 0xffff82fddef0 from 0x000000000001B669 to 0x0000000000000001 of type 16 and protocolId 0 on exchange 56424.
[1625879265.402653][113310:113317] CHIP:IN: Sending msg 0xffff82fddef0 to 0x0000000000000001 at utc time: 22585754 msec
[1625879265.402699][113310:113317] CHIP:IN: Sending secure msg on generic transport
[1625879265.402892][113310:113317] CHIP:IN: Secure msg send status No Error
```

yunhanw-google, 2021-07-12:
should be fixed in https://github.com/project-chip/connectedhomeip/pull/8241

erjiaqing, 2021-07-13:
Fixed via #8241

closed 2021-07-13 by erjiaqing