project-chip/connectedhomeip #8308: Copy/pasted non-spec-compliant code in subscribe message definitions

bzbarsky-apple, 2021-07-12:
#### Problem
See https://github.com/project-chip/connectedhomeip/pull/8285#issuecomment-878633570 -- ` SubscribeRequest::Parser::GetMinIntervalMs` is not getting the interval (!) and should be treating the value it gets as _seconds_ per spec.

 #### Proposed Solution
Probably revert #8285 and also fix the code to get the right value.  And fix the tests to actually test the code.

bzbarsky-apple, 2021-07-12:
Also, I have no idea what `FinalSyncInterval` is.  It's not in the actual message definitions in the IM spec; "8.5.3. Subscribe Response Action" has "MinInterval" and "MaxInterval".  It's in the encoding spec, of course, but all that means is that the encoding spec can't encode what the actual messages are supposed to be in its current state...

@mrjerryjohns @CamWms

yunhanw-google, 2021-12-15:
fixed in latest code

closed 2021-12-15 by yunhanw-google