project-chip/connectedhomeip #8321: Accessory device is not able to communicate with controller after reset

kkasperczyk-no, 2021-07-13:
#### Problem
As in the title. The device looks like having a problem with finding the key to communicate with controller after reset. It was tested using:
- nrfconnect platform and lock-app example
- repo revision: 0e19e0dc441b2481c48022856211f3471f9a6600

Steps to reproduce the problem:
1. Commission the accessory device with python controller.
2. Resolve address. (`resolve <fabric-id> <node-id>`)
3. Verify that communication with accessory using OnOff commands works fine (e.g. `zcl OnOff On 3 1 0`, `zcl OnOff Off 3 1 0`)
4. Reset the device with the button (soft reset, not factory reset)
5. Verify that communication with accessory using OnOff commands doesn't work.

Controller logs:
```
chip-device-ctrl > zcl OnOff Off 3 1 0
[1626161000.714107][59535:59535] CHIP:DMG: ICR moving to [Initialize]
[1626161000.714159][59535:59535] CHIP:DMG: ICR moving to [AddCommand]
[1626161000.714322][59535:59535] CHIP:IN: Secure message was encrypted: Msg ID 4
[1626161000.714348][59535:59535] CHIP:IN: Encrypted message 0x28d7298 from 0x000000000001B669 to 0x0000000000000003 of type 8 and protocolId 5 on exchange 39026.
[1626161000.714377][59535:59535] CHIP:IN: Sending msg 0x28d7298 to 0x0000000000000003 at utc time: 5266386 msec
[1626161000.714392][59535:59535] CHIP:IN: Sending secure msg on generic transport
[1626161000.714537][59535:59535] CHIP:IN: Secure msg send status No Error
[1626161000.714579][59535:59535] CHIP:DMG: ICR moving to [ Sending]
[1626161005.649890][59535:59543] CHIP:IN: Sending msg 0x28d7298 to 0x0000000000000003 at utc time: 5271322 msec
[1626161005.649939][59535:59543] CHIP:IN: Sending secure msg on generic transport
[1626161005.650078][59535:59543] CHIP:IN: Secure msg send status No Error
[1626161005.650096][59535:59543] CHIP:EM: Retransmit MsgId:00000004 Send Cnt 1
[1626161005.905593][59535:59543] CHIP:IN: Sending msg 0x28d7298 to 0x0000000000000003 at utc time: 5271577 msec
[1626161005.905645][59535:59543] CHIP:IN: Sending secure msg on generic transport
[1626161005.905805][59535:59543] CHIP:IN: Secure msg send status No Error
[1626161005.905825][59535:59543] CHIP:EM: Retransmit MsgId:00000004 Send Cnt 2
[1626161006.161315][59535:59543] CHIP:IN: Sending msg 0x28d7298 to 0x0000000000000003 at utc time: 5271833 msec
[1626161006.161364][59535:59543] CHIP:IN: Sending secure msg on generic transport
[1626161006.161497][59535:59543] CHIP:IN: Secure msg send status No Error
[1626161006.161516][59535:59543] CHIP:EM: Retransmit MsgId:00000004 Send Cnt 3
[1626161006.417944][59535:59543] CHIP:EM: Failed to Send CHIP MsgId:00000004 sendCount: 3 max retries: 3
[1626161012.716877][59535:59543] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39026
[1626161012.717044][59535:59543] CHIP:ZCL: DefaultResponse:
[1626161012.717065][59535:59543] CHIP:ZCL: Transaction: 0x7f4dbc4c0b78
[1626161012.717095][59535:59543] CHIP:ZCL: status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1626161012.717117][59535:59543] CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback
[1626161012.717129][59535:59543] CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
[1626161012.717167][59535:59543] CHIP:DMG: ICR moving to [Uninitiali]
Failed to receive command response: None
```

Device logs:
```
E: 13795 [IN]Data received on an unknown connection (1). Dropping it!!
E: 13801 [EM]Error receiving message from UDP:2001:db8:1::1:11098: CHIP Error 4127 (0x0000101F): Key not found error code received from peer
E: 18732 [IN]Data received on an unknown connection (1). Dropping it!!
E: 18739 [EM]Error receiving message from UDP:2001:db8:1::1:11098: CHIP Error 4127 (0x0000101F): Key not found error code received from peer
E: 18988 [IN]Data received on an unknown connection (1). Dropping it!!
E: 18995 [EM]Error receiving message from UDP:2001:db8:1::1:11098: CHIP Error 4127 (0x0000101F): Key not found error code received from peer
E: 19244 [IN]Data received on an unknown connection (1). Dropping it!!
E: 19251 [EM]Error receiving message from UDP:2001:db8:1::1:11098: CHIP Error 4127 (0x0000101F): Key not found error code received from peer
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

kkasperczyk-no, 2021-07-20:
As far as I understand problem was solved, so I can close that one.

Damian-Nordic, 2021-07-27:
It is not properly solved. There's currently no automatic mechanism of recovering a CASE session whose one end lost the connection. We have a workaround, but let's keep it open :)

closed 2022-09-22 by stale[bot]