project-chip/connectedhomeip #8324: EFR32 lighting-app Resolve address error

hainahouwei, 2021-07-13:
@jmartinez-silabs

#### Problem

   EFR32 lighting-app Resolve address error

#### Steps to reproduce the problem:

##### platform:   

-   (Raspberry1)  OT Border Router  run on  Raspberry Pi OS(32-bit),  base on commit 80736a878493a2e5a2a0845e5ed4719c318db569

-   (Raspberry 2)  Python Controller run on the ubuntu Desktop 21.04 (RPi4/400),  base on commit    13888a6b862fb749d72aad503c62b5fdb381237f

-    lighting-app run on the efr32 board BRD4161A   base on commit    13888a6b862fb749d72aad503c62b5fdb381237f

   

1. ​    run command as follow on (Raspberry1)  OT Border Router 

   ```
   sudo service otbr-agent restart
   sudo ot-ctl factoryreset
   sudo ot-ctl prefix add 2001:db8::/64 paros
   sudo ip addr add dev wlan0 2002::2/64
   sudo ot-ctl ifconfig down
   sudo ot-ctl panid 0xface
   sudo ot-ctl extpanid dead00beef00cafe
   sudo ot-ctl networkkey 00112233445566778899aabbccddeeff
   sudo ot-ctl channel 11
   sudo ot-ctl networkname simon-openThread
   sudo ot-ctl ifconfig up
   sudo ot-ctl thread start
   ```

2. Raspberry2 connect to Raspberry1's AP   via wifi

3.  run command as follow on (Raspberry2) 

   ```
   sudo ifconfig enp0s3 inet6 add 2002::1/64
   sudo ip route add 2001:db8:0:0::/64 via 2002::2
   sudo ip route add fdde:ad00:beef::/64 via 2002::2
   ```

4. test ping 

   on Raspberry1 OT Border Router

   ```
   pi@raspberrypi:~/ot-br-posix $ sudo ot-ctl
   > ipaddr
   2001:db8:0:0:b237:78e2:9615:2d8d
   fdde:ad00:beef:0:0:ff:fe00:2803
   fdde:ad00:beef:0:3dff:e568:2d68:1b07
   fe80:0:0:0:a82a:a98b:517:d78e
   Done
   
   ```

   on Raspberry2 Python Controller

   ```
   wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
           inet 192.168.123.109  netmask 255.255.255.0  broadcast 192.168.123.255
           inet6 fd6a:4a79:580b:0:f86a:743b:121b:3681  prefixlen 64  scopeid 0x0<global>
           inet6 fe80::e603:450a:ddac:7dbe  prefixlen 64  scopeid 0x20<link>
           inet6 fd6a:4a79:580b:0:970d:268c:75f1:d18d  prefixlen 64  scopeid 0x0<global>
           inet6 fd45:f14f:e90:0:56d0:52fa:5a43:e186  prefixlen 64  scopeid 0x0<global>
           inet6 2002::1  prefixlen 64  scopeid 0x0<global>
[... 978 lines cut ...]
00> <detail> [DL]    FDDE:AD00:BEEF:0:AF6E:1643:699C:3AF0 Thread mesh-local address)
00> <detail> [DL]    FDDE:AD00:BEEF::FF:FE00:FC00 Thread mesh-local address)
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <detail> [IN] Clearing BLE pending packets.
00> <info  > [BLE] Releasing end point's BLE connection back to application.
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
```

jmartinez-silabs, 2021-07-13:
@hainahouwei Thank you for all the details

The EFR32 device didn't join the BR Thread Network you can see this on the EFR32 logs that it becomes leader his own network so not connected to anything else, and wont be able to advertisse the SRP service
```
00> <detail> [DL] OpenThread State Changed (Flags: 0x100012a5)
00> <detail> [DL]    Device Role: LEADER
00> <detail> [DL]    Partition Id: 0xDC7E081
00> <detail> [DL]    Thread Unicast Addresses:
00> <detail> [DL]         FDDE:AD00:BEEF::FF:FE00:FC00/64 valid
00> <detail> [DL]         FDDE:AD00:BEEF::FF:FE00:E000/64 valid rloc
00> <detail> [DL]         FDDE:AD00:BEEF:0:AF6E:1643:699C:3AF0/64 valid
00> <detail> [DL]         FE80::3876:1C0F:644B:AE79/64 valid preferred

```

This points to and error when providing the thread credential to the device.

You showed that your OTBR active dataset is

```
dataset active -x
000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef0000051000112233445566778899aabbccddeeff031073696d6f6e2d6f70656e5468726561640102face04106d1a1587bc960e420620b4f3e8adf0be0c0402a0fff80e080000000000000000
```

But the dataset you provided in the AddthreadNetwork doesn't match.

`zcl NetworkCommissioning AddThreadNetwork 1234 0 0 operationalDataset=hex:**0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef0000051085691ff0090eba917a03593fe59d0e92031073696d6f6e2d6f70656e5468726561640102face0410f0ea5abf113ce95462e41ba38401268b0c0402a0fff8**  breadcrumb=0 timeoutMs=3000`

Please try again making sure you copied the dataset value into AddThreadNetwork comment arguments and let me know

hainahouwei, 2021-07-14:
@jmartinez-silabs Thank you very much for your advices.

I  had try  several times again,   dataset value is right  .  However  the `resove address` process still not go ahead as expected. 

 And I sure  the EFR32 device  joined the BR Thread Network become a router  as follow command :

#### on OT Border Router

```
> state
leader
Done
> nameworkname
Error 35: InvalidCommand
> networkname
openThread
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
| 20 | 0x5000 |       63 |         0 |     3 |      3 |  10 | 6eca4697863ddc66 |    1 |
| 37 | 0x9400 |       63 |         0 |     0 |      0 |   0 | aa2aa98b0517d78e |    0 |

Done

```

#### on EFR32

```
> eui64
000b57fffed8f0d7
Done
> state
router
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |               Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+              ------+
| 20 | 0x5000 |       63 |         0 |     0 |      0 |   0 | 6eca4697863ddc66 |                  0 |
| 37 | 0x9400 |       63 |         0 |     3 |      3 |  18 | aa2aa98b0517d78e |                  1 |

Done
> networkname
openThread
Done
>
```

#### On BR Web

[image]

#### dataset on OT Border Router

```
> dataset active -x
0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef0000051000112233445566778899aabbccddeeff030a6f70656e5468726561640102face04106d1a1587bc960e420620b4f3e8adf0be0c0402a0fff8
Done
> extpanid
dead00beef00cafe
Done
>

```

#### log on python controller

```
(python_env) simon@simon-desktop:~/connectedhomeip$ chip-device-ctrl --bluetooth-adapter hci1
[... 1423 lines cut ...]
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26

```

jmartinez-silabs, 2021-07-14:
@hainahouwei  Hi, ok this time it joined the thread network succesfully but there is a weird behaviour that I see with your thread network. Your OTBR seems to crash/reset mulitple times. You can see in the EFR32 Logs that the thread state keeps changing from child -> router  with the right IP but then drops to Leader of his own network with only Link Local addresses.

So you might want to check your OT-BR and What commit hash it uses. I suggest using 
`commit 15d556e94b72144db117b7d602418fe8f09a642f`

But the resolve command you are using won't work for EFR32. The fabric ID on our device is 0 current
so the command should be `resolve 0 <Your_nodeID>`
The `get-fabricid` command gives you the fabric ID of the controller.

You can see what the SRP service advertisse with either avahi-browse or if you are on MAC Discovery
[image]

hainahouwei, 2021-07-16:
[@jmartinez-silabs](https://github.com/jmartinez-silabs) Thank you very much for your advices!

I  tried to use `commit 15d556e94b72144db117b7d602418fe8f09a642f`  on my OT-BR  .

And changed  resolve command as `esolve 0 <my_nodeID>`

But it show erro as fllow: 

```
chip-device-ctrl > resolve 1234  1234
[1626426298.318597][335383:335383] CHIP:DIS: mDNS broadcast success
[1626426298.318865][335383:335383] CHIP:DIS: mDNS broadcast success
[1626426298.319061][335383:335383] CHIP:DIS: mDNS broadcast success
[1626426298.319262][335383:335383] CHIP:DIS: mDNS broadcast success
[1626426611.776544][335383:335390] CHIP:DIS: Node ID resolved for 0x00000000000004D2 to [2001:db8::2cdc:4f1e:e09b:97a7]:11097
[1626426611.777974][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
Failed to update node address: 33554533
Exception ignored on calling ctypes callback function: <function ChipDeviceController.__init__.<locals>.HandleAddressUpdateComplete at 0xffff92554d30>
Traceback (most recent call last):
  File "/home/simon/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 114, in HandleAddressUpdateComplete
    self.state = DCState.IDLEHandleCommissioningComplete
  File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
    raise AttributeError(name) from None
AttributeError: IDLEHandleCommissioningComplete
[1626426611.779838][335383:335390] CHIP:CTL: OperationalDiscoveryComplete for device ID 1234
[1626426611.780983][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
[1626426611.781072][335383:335390] CHIP:CTL: Device connection failed. Error OS Error 0x02000065: Network is unreachable
Failed to commission: 33554533
simon debug,do_resolve,err=%d 33554533
[1626426611.781495][335383:335390] CHIP:DIS: Node ID resolved for 0x00000000000004D2 to [2001:db8::2cdc:4f1e:e09b:97a7]:11097
chip-device-ctrl > [1626426611.782458][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
Failed to update node address: 33554533
Exception ignored on calling ctypes callback function: <function ChipDeviceController.__init__.<locals>.HandleAddressUpdateComplete at 0xffff92554d30>
Traceback (most recent call last):
  File "/home/simon/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 114, in HandleAddressUpdateComplete
    self.state = DCState.IDLEHandleCommissioningComplete
  File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
    raise AttributeError(name) from None
AttributeError: IDLEHandleCommissioningComplete
[1626426611.784547][335383:335390] CHIP:CTL: OperationalDiscoveryComplete for device ID 1234
[1626426611.785751][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
[1626426611.785862][335383:335390] CHIP:CTL: Device connection failed. Error OS Error 0x02000065: Network is unreachable
Failed to commission: 33554533
[1626426611.786375][335383:335390] CHIP:DIS: Node ID resolved for 0x00000000000004D2 to [2001:db8::2cdc:4f1e:e09b:97a7]:11097
[1626426611.787518][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
Failed to update node address: 33554533
Exception ignored on calling ctypes callback function: <function ChipDeviceController.__init__.<locals>.HandleAddressUpdateComplete at 0xffff92554d30>
Traceback (most recent call last):
  File "/home/simon/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 114, in HandleAddressUpdateComplete
    self.state = DCState.IDLEHandleCommissioningComplete
  File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
    raise AttributeError(name) from None
AttributeError: IDLEHandleCommissioningComplete
[1626426611.788632][335383:335390] CHIP:CTL: OperationalDiscoveryComplete for device ID 1234
[1626426611.789388][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
[1626426611.789446][335383:335390] CHIP:CTL: Device connection failed. Error OS Error 0x02000065: Network is unreachable
Failed to commission: 33554533
[1626426611.789702][335383:335390] CHIP:DIS: Node ID resolved for 0x00000000000004D2 to [2001:db8::2cdc:4f1e:e09b:97a7]:11097
[1626426611.790288][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
Failed to update node address: 33554533
Exception ignored on calling ctypes callback function: <function ChipDeviceController.__init__.<locals>.HandleAddressUpdateComplete at 0xffff92554d30>
Traceback (most recent call last):
  File "/home/simon/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 114, in HandleAddressUpdateComplete
    self.state = DCState.IDLEHandleCommissioningComplete
  File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
    raise AttributeError(name) from None
AttributeError: IDLEHandleCommissioningComplete
[1626426611.791354][335383:335390] CHIP:CTL: OperationalDiscoveryComplete for device ID 1234
[1626426611.792088][335383:335390] CHIP:CTL: LoadSecureSessionParameters returning error OS Error 0x02000065: Network is unreachable
[1626426611.792146][335383:335390] CHIP:CTL: Device connection failed. Error OS Error 0x02000065: Network is unreachable
[... 1500 lines cut ...]
[... cut ...]
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
```

jmartinez-silabs, 2021-07-19:
@hainahouwei Your controller seems to hit mutiple errors like
```
Exception ignored on calling ctypes callback function: <function ChipDeviceController.__init__.<locals>.HandleAddressUpdateComplete at 0xffff92554d30>
Traceback (most recent call last):
  File "/home/simon/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 114, in HandleAddressUpdateComplete
    self.state = DCState.IDLEHandleCommissioningComplete
  File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
    raise AttributeError(name) from None
AttributeError: IDLEHandleCommissioningComplete
```

I have never seen those and that should not be happen in a correct setup.

On what device are you running the python controller?

I validated curent master version was succesful with my setup.

Make sure to build the controller on the same commit has the EFR32 example you are building to avoid any conflicts

hainahouwei, 2021-07-19:
@jmartinez-silabs 
I run  Python Controller  on the ubuntu Desktop 21.04 (RPi4/400), base on commit 13888a6b862fb749d72aad503c62b5fdb381237f .,and  EFR32 example on the same commit .
My setup steps are 1 through 11 as shown in the first question above. But I not sure the steps is right or not .Could you please guide me through the detailed steps of your experiment?

hainahouwei, 2021-07-19:
> @jmartinez-silabs
> I run Python Controller on the ubuntu Desktop 21.04 (RPi4/400), base on commit [13888a6](https://github.com/project-chip/connectedhomeip/commit/13888a6b862fb749d72aad503c62b5fdb381237f) .,and EFR32 example on the same commit .
> My setup steps are 1 through 11 as shown in the first question above. But I not sure the steps is right or not .Could you please guide me through the detailed steps of your experiment?

jmartinez-silabs, 2021-07-19:
Or I went back up and re read your setup. I do not use any wifi AP.

Here is what I Do:
OTBR build on raspi 4 with EFR32 OT-rcp
```
cd ot-br-posix
sudo ./script/bootstrap
sudo BACKBONE_ROUTER=1 INFRA_IF_NAME=eth0 ./script/setup

sudo vim /etc/default/otbr-agent
   replace with 
   OTBR_AGENT_OPTS="-I wpan0 -B eth0 spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=460800"

reboot

sudo ot-ctl
factoryreset
prefix add fd11:22::/64 paros
thread stop
ifconfig down
ifconfig up
thread start
dataset active -x
dataset active
```

PYTHON CONTROLLER on another raspberrypi 4
_Note that my device already ahve ipv6 native addressing else you need to create an ipv6 addresses on both devices like you did in. your steps_

```
sudo ip -6 route add fd11:22::/64 dev eth0
scripts/build_python.sh -m platform
source ./out/python_env/bin/activate
chip-device-ctrl

connect -ble 3840 73141520 1345

zcl NetworkCommissioning AddThreadNetwork 1345 0 0 operationalDataset=hex:0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef000005105091ac26d43900b7a26ca2a035cda5e6030a4f70656e54687265616401027ab40410101279bd37d2da81b788e85f703d1a930c0402a0fff8 breadcrumb=0 timeoutMs=1000

zcl NetworkCommissioning EnableNetwork 1345 0 0 networkID=hex:dead00beef00cafe breadcrumb=0 timeoutMs=1000

close-ble

resolve 0 1345

zcl OnOff Toggle 1345 1 0
```

hainahouwei, 2021-07-20:
@jmartinez-silabs 
According to your steps , I successfully resolve address .But failed to send zcl command  as follow.

Have you ever seen this kind of error?

#### python contrller log

```
chip-device-ctrl > resolve 0  1234
[1626749749.460865][40088:40095] CHIP:DL: Avahi resolve found
[1626749749.461592][40088:40095] CHIP:DIS: Node ID resolved for 0x00000000000004D2 to [fd03:a829:7269:6c5:3c6f:616f:e0e2:f9d1]:5540
Node address has been updated
[1626749749.462923][40088:40095] CHIP:CTL: OperationalDiscoveryComplete for device ID 1234
[1626749749.760039][40088:40095] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 39023
Commissioning complete
Current address: fd03:a829:7269:6c5:3c6f:616f:e0e2:f9d1:5540
chip-device-ctrl > [1626749754.710974][40088:40095] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 39023
[1626749754.711284][40088:40095] CHIP:EM: Dropping unexpected message 00000000:49 986F MsgId:00000000

chip-device-ctrl > zcl OnOff Toggle 1234 1 0
[1626749804.206941][40088:40088] CHIP:IN: Encrypted message 0x259e95e8 from 0x000000000001B669 to 0x00000000000004D2 of type 8 and protocolId 5 on exchange 39024.
[1626749804.207025][40088:40088] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9584473 msec
[1626749804.207071][40088:40088] CHIP:IN: Sending secure msg on generic transport
[1626749804.207260][40088:40088] CHIP:IN: Secure msg send status No Error
[1626749809.174718][40088:40095] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9589440 msec
[1626749809.174814][40088:40095] CHIP:IN: Sending secure msg on generic transport
[1626749809.175074][40088:40095] CHIP:IN: Secure msg send status No Error
[1626749809.426577][40088:40095] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9589692 msec
[1626749809.426651][40088:40095] CHIP:IN: Sending secure msg on generic transport
[1626749809.426837][40088:40095] CHIP:IN: Secure msg send status No Error
[1626749809.682475][40088:40095] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9589948 msec
[1626749809.682556][40088:40095] CHIP:IN: Sending secure msg on generic transport
[1626749809.682757][40088:40095] CHIP:IN: Secure msg send status No Error
[1626749809.938235][40088:40095] CHIP:EM: Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3
[1626749816.207949][40088:40095] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39024
[1626749816.208182][40088:40095] CHIP:ZCL: DefaultResponse:
[1626749816.208251][40088:40095] CHIP:ZCL:   Transaction: 0xffff8e44fc18
Failed to receive command response: None
[1626749816.208307][40088:40095] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
chip-device-ctrl > zcl OnOff Toggle 1234 1 0
[1626749898.422628][40088:40088] CHIP:IN: Encrypted message 0x259e95e8 from 0x000000000001B669 to 0x00000000000004D2 of type 8 and protocolId 5 on exchange 39025.
[1626749898.422714][40088:40088] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9678688 msec
[1626749898.422761][40088:40088] CHIP:IN: Sending secure msg on generic transport
[1626749898.422956][40088:40088] CHIP:IN: Secure msg send status No Error
[1626749903.382433][40088:40095] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9683648 msec
[1626749903.382531][40088:40095] CHIP:IN: Sending secure msg on generic transport
[1626749903.382794][40088:40095] CHIP:IN: Secure msg send status No Error
[1626749903.634300][40088:40095] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9683900 msec
[1626749903.634376][40088:40095] CHIP:IN: Sending secure msg on generic transport
[1626749903.634566][40088:40095] CHIP:IN: Secure msg send status No Error
[1626749903.891084][40088:40095] CHIP:IN: Sending msg 0x259e95e8 to 0x00000000000004D2 at utc time: 9684157 msec
[1626749903.891161][40088:40095] CHIP:IN: Sending secure msg on generic transport
[1626749903.891362][40088:40095] CHIP:IN: Secure msg send status No Error
[1626749904.146831][40088:40095] CHIP:EM: Failed to Send CHIP MsgId:00000001 sendCount: 3 max retries: 3
[1626749910.423606][40088:40095] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39025
[1626749910.423836][40088:40095] CHIP:ZCL: DefaultResponse:
[1626749910.423904][40088:40095] CHIP:ZCL:   Transaction: 0xffff8e44fc18
Failed to receive command response: None
[1626749910.423960][40088:40095] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
```

#### EFR32 Log

```
00> <efr32 > ==================================================
00> <efr32 > chip-efr32-lighting-example starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <detail> [-] Initializing key value store
00> <detail> [-] KeyValueStore init complete: active keys 2, deleted keys 2, sectors 4, logical sector size 2048 bytes
[... 798 lines cut ...]
00> <detail> [DL]     dest FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1, port 5540
00> <error > [IN] Data received on an unknown connection (1). Dropping it!!
00> <error > [EM] Error receiving message from UDP:FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF:5541: CHIP Error 0x0000007F: Key not found error code received from peer
00> <detail> [DL] Thread packet RCVD: UDP, len 116
00> <detail> [DL]     src  FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF, port 5541
00> <detail> [DL]     dest FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1, port 5540
00> <error > [IN] Data received on an unknown connection (1). Dropping it!!
[... cut ...]
00> <error > [EM] Error receiving message from UDP:FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF:5541: CHIP Error 0x0000007F: Key not found error code received from peer
00> <detail> [DL] Thread packet RCVD: UDP, len 116
00> <detail> [DL]     src  FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF, port 5541
00> <detail> [DL]     dest FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1, port 5540
00> <error > [IN] Data received on an unknown connection (1). Dropping it!!
00> <error > [EM] Error receiving message from UDP:FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF:5541: CHIP Error 0x0000007F: Key not found error code received from peer

```

jmartinez-silabs, 2021-07-20:
```
00> <detail> [SC] Sent SigmaR2 msg
00> <error > [DL] Long dispatch time: 225 ms, for event type 3
00> <detail> [DL] Thread packet SENT: UDP, len 554
00> <detail> [DL]     src  FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1, port 5540
00> <detail> [DL]     dest FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF, port 5541
00> <detail> [EM] Retransmit MsgId:00000000 Send Cnt 1
00> <detail> [DL] Thread packet RCVD: UDP, len 66
00> <detail> [DL]     src  FD14:7B01:4444:0:83BB:2C11:1CB2:9EFF, port 5541
00> <detail> [DL]     dest FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1, port 5540
00> <info  > [EM] Received message of type 16 and protocolId 0 on exchange 39023
00> <detail> [EM] Rxd Ack; Removing MsgId:00000000 from Retrans Table
00> <detail> [EM] Removed CHIP MsgId:00000000 from RetransTable
00> <error > [SC] CASESession timed out while waiting for a response from the peer. Expected message type was hu
00> <info  > [IN] CASE Session establishment failed: CHIP Error 0x00000032: Timeout
```

The case session timed out failing too receive a response t the SigmaR2 msg sent by the device to the controller. This can happens when the routing is working properly.

I see that the device is communicating with the IPV6 ADDR
`FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1` and not the FD11

```
00> <detail> [DL] LwIP Thread interface addresses updated
00> <detail> [DL]    FE80::54C0:CD6F:3D8F:868F IPv6 link-local address, preferred)
00> <detail> [DL]    FDDE:AD00:BEEF:0:DC0F:FB5E:9918:2F17 Thread mesh-local address)
00> <detail> [DL]    FD03:A829:7269:6C5:3C6F:616F:E0E2:F9D1 IPv6 unique local address, preferred)
00> <detail> [DL]    FD11:22::393E:9A70:8956:420B IPv6 unique local address, preferred)
```
Both of them are IPV6 unique local I am not sure how you are configured to have two of them. But if you did the route for FD11 but it use FD03 that might be a reason for it to fail

hainahouwei, 2021-07-20:
@jmartinez-silabs 
Yes, I did the route for FD11 . But when I change the route to FD03 and it use FD03 too , it still failed.

Do you use two raspberry connected to the same router ? Or connected directly? 
Should I set anything about IPv6 on the router?

#### python controller log

```
chip-device-ctrl > resolve 0 1234
[1626790692.934844][1715:1722] CHIP:DL: Avahi resolve found
[1626790692.935012][1715:1722] CHIP:DIS: Node ID resolved for 0x00000000000004D2 to [fd03:22::a730:4e49:9012:2f1e]:5540
Node address has been updated
[1626790692.935658][1715:1722] CHIP:CTL: OperationalDiscoveryComplete for device ID 1234
[1626790693.251723][1715:1722] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 39023
Commissioning complete
Current address: fd03:22::a730:4e49:9012:2f1e:5540
chip-device-ctrl > [1626790698.241343][1715:1722] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 39023
[1626790698.241616][1715:1722] CHIP:EM: Dropping unexpected message 00000000:49 986F MsgId:00000000
[1626790698.497861][1715:1722] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 39023
[1626790698.756725][1715:1722] CHIP:EM: Received message of type 49 and protocolId 0 on exchange 39023

chip-device-ctrl > zcl OnOff Toggle 1234 1 0
[1626790730.553354][1715:1715] CHIP:IN: Encrypted message 0x1438fd18 from 0x000000000001B669 to 0x00000000000004D2 of type 8 and protocolId 5 on exchange 39024.
[1626790730.553460][1715:1715] CHIP:IN: Sending msg 0x1438fd18 to 0x00000000000004D2 at utc time: 826222 msec
[1626790730.553521][1715:1715] CHIP:IN: Sending secure msg on generic transport
[1626790730.553748][1715:1715] CHIP:IN: Secure msg send status No Error
[1626790735.492314][1715:1722] CHIP:IN: Sending msg 0x1438fd18 to 0x00000000000004D2 at utc time: 831161 msec
[1626790735.492410][1715:1722] CHIP:IN: Sending secure msg on generic transport
[1626790735.492637][1715:1722] CHIP:IN: Secure msg send status No Error
[1626790735.745173][1715:1722] CHIP:IN: Sending msg 0x1438fd18 to 0x00000000000004D2 at utc time: 831414 msec
[1626790735.745246][1715:1722] CHIP:IN: Sending secure msg on generic transport
[1626790735.745401][1715:1722] CHIP:IN: Secure msg send status No Error
[1626790736.000907][1715:1722] CHIP:IN: Sending msg 0x1438fd18 to 0x00000000000004D2 at utc time: 831670 msec
[1626790736.000983][1715:1722] CHIP:IN: Sending secure msg on generic transport
[1626790736.001151][1715:1722] CHIP:IN: Secure msg send status No Error
[1626790736.257636][1715:1722] CHIP:EM: Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3
[1626790742.558411][1715:1722] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39024
[1626790742.558675][1715:1722] CHIP:ZCL: DefaultResponse:
[1626790742.558752][1715:1722] CHIP:ZCL:   Transaction: 0xffff7ee42c08
[1626790742.558824][1715:1722] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
```

#### EFR32 Log

```
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32778
00> <detail> [DL] OpenThread State Changed (Flags: 0x00000064)
00> <detail> [DL]    Device Role: ROUTER
00> <detail> [DL] Thread packet RCVD: UDP, len 203
00> <detail> [DL]     src  FDC0:D401:6FB:0:DAB9:A605:DE6E:F79D, port 5541
00> <detail> [DL]     dest FD03:22::A730:4E49:9012:2F1E, port 5540
00> <info  > [EM] Received message of type 48 and protocolId 0 on exchange 39023
00> <detail> [EM] ec id: 39023, Delegate: 0x0x20012574
00> <info  > [IN] CASE Server received SigmaR1 message. Starting handshake. EC 0x200137d0
00> <detail> [SC] Waiting for SigmaR1 msg
00> <detail> [SC] Received SigmaR1 msg
00> <detail> [SC] Peer assigned session key ID 2
00> <detail> [EM] Piggybacking Ack for MsgId:00000000 with msg
00> <detail> [DL] Thread packet SENT: UDP, len 554
00> <detail> [DL]     src  FD03:22::A730:4E49:9012:2F1E, port 5540
00> <detail> [DL]     dest FDC0:D401:6FB:0:DAB9:A605:DE6E:F79D, port 5541
00> <detail> [SC] Sent SigmaR2 msg
00> <error > [DL] Long dispatch time: 227 ms, for event type 3
00> <detail> [DL] Thread packet SENT: UDP, len 554
00> <detail> [DL]     src  FD03:22::A730:4E49:9012:2F1E, port 5540
00> <detail> [DL]     dest FDC0:D401:6FB:0:DAB9:A605:DE6E:F79D, port 5541
00> <detail> [EM] Retransmit MsgId:00000000 Send Cnt 1
[... 79 lines cut ...]
        inet6 fdde:ad00:beef:0:1185:6938:21db:7950  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:7000  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fd03:22::6272:8c1b:5c7a:33df  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::201f:ad07:90f5:2a55  prefixlen 64  scopeid 0x20<link>
        inet6 fdde:ad00:beef::ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10  bytes 1680 (1.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

#### IP on EFR32

```
ipaddr
fd03:22:0:0:a730:4e49:9012:2f1e
fdde:ad00:beef:0:0:ff:fe00:7001
fdde:ad00:beef:0:2c2d:e1b6:eddb:ea4b
fe80:0:0:0:3cb0:af3b:99ae:1674
Done
```

hainahouwei, 2021-07-23:
@jmartinez-silabs Thank you very much for you help .  I can resolve address and control EFR32 light app now. But I find that it's easily time out in ` CASE Session establishment` process.  Many I have any way to avoid it?

```
00> <detail> [DL]    Device Role: ROUTER
00> <detail> [DL] Thread packet RCVD: UDP, len 203
00> <detail> [DL]     src  FD1F:D297:8B0B:0:7E44:857:7A9:5E1F, port 5541
00> <detail> [DL]     dest FD03:22::F368:C737:5EF8:8E4B, port 5540
00> <info  > [EM] Received message of type 48 and protocolId 0 on exchange 39023
00> <detail> [EM] ec id: 39023, Delegate: 0x0x20012574
00> <info  > [IN] CASE Server received SigmaR1 message. Starting handshake. EC 0x200137d0
00> <detail> [SC] Waiting for SigmaR1 msg
00> <detail> [SC] Received SigmaR1 msg
00> <detail> [SC] Peer assigned session key ID 2
00> <detail> [EM] Piggybacking Ack for MsgId:00000000 with msg
00> <detail> [DL] Thread packet SENT: UDP, len 554
00> <detail> [DL]     src  FD03:22::F368:C737:5EF8:8E4B, port 5540
00> <detail> [DL]     dest FD1F:D297:8B0B:0:7E44:857:7A9:5E1F, port 5541
00> <detail> [SC] Sent SigmaR2 msg
00> <error > [DL] Long dispatch time: 226 ms, for event type 3
00> <detail> [DL] Thread packet SENT: UDP, len 554
00> <detail> [DL]     src  FD03:22::F368:C737:5EF8:8E4B, port 5540
00> <detail> [DL]     dest FD1F:D297:8B0B:0:7E44:857:7A9:5E1F, port 5541
00> <detail> [EM] Retransmit MsgId:00000000 Send Cnt 1
00> <detail> [DL] Thread packet RCVD: UDP, len 66
00> <detail> [DL]     src  FD1F:D297:8B0B:0:7E44:857:7A9:5E1F, port 5541
00> <detail> [DL]     dest FD03:22::F368:C737:5EF8:8E4B, port 5540
00> <info  > [EM] Received message of type 16 and protocolId 0 on exchange 39023
00> <detail> [EM] Rxd Ack; Removing MsgId:00000000 from Retrans Table
00> <detail> [EM] Removed CHIP MsgId:00000000 from RetransTable
00> <error > [SC] CASESession timed out while waiting for a response from the peer. Expected message type was hu
00> <info  > [IN] CASE Session establishment failed: CHIP Error 0x00000032: Timeout
```

In addition, if the CASE is successfully established, the response of the `OnOff Toggle` is sometimes fast and sometimes slow. Is it the network environment? Is there a way to solve it?

jmartinez-silabs, 2021-07-23:
It should be really fast. I am not sure what in the network could slow it down. The only time i noticed CASESession time outs is when my routing wasn't working. It can happens to drop a UDP package but we do have a retry mechanism in Matter. So a full timeout usually happens when the address is not reachable for some reason

hainahouwei, 2021-07-24:
@jmartinez-silabs Are there any  retry mechanism  about receive SigmaR3 msg ?  I find that  it can't  receive SigmaR3 msg again ,if SigmaR1 msg and SigmaR2 msg  was received but SigmaR3 not.   So even though the network become better ,it still can't establish  CASE .

closed 2021-08-12 by hainahouwei

hainahouwei, 2021-08-12:
OTBR and python controller  should both have global ipv6 address.

   setting in master router

[image]

enable ipv6 in   dhcpcd of OTBR

```
 sudo nano /etc/dhcpcd.conf
```

[image]