project-chip/connectedhomeip #8342: Support concurrent CASE handshake sessions

pan-apple, 2021-07-13:
#### Problem
Currently, a device/controller can only have an ongoing CASE handshake. The following use case is not achievable.
1. Node A starts a CASE handshake with Node B
2. Node C starts another CASE handshake with Node A, while handshake with Node B is in progress

 #### Proposed Solution

There are multiple changes needed to support this.

- [ ] Update CASEServer to be able to track multiple ongoing CASE handshakes
- [ ] Don't mutate local credential/certificate set to validate peer's credentials

Add other changes to the issue as detected while enabling this functionality.

pan-apple, 2021-07-28:
Referring the comment from @tcarmelveilleux, so not to lose track of it.
https://github.com/project-chip/connectedhomeip/issues/8207#issuecomment-887547995

msandstedt, 2021-09-22:
When we implement this, we will also likely want to move to a dynamically-sized unauthenticated session pool. In such case, we will also need to add a garbage collection mechanism for stale sessions (ref count 0), as eviction no longer works in such an approach.

CC @kghost , @samadDotDev , @kianooshkarami

woody-apple, 2022-02-07:
Keeping in v1 for now.

PR #17069 (2022-04-15): Introduce ExchangeAcceptor for unsolicited message handler
#### Problem
Path to fix #8342

#### Change overview
The unsolicited message handler should take an exchange acceptor instead of an exchange delegate. Instead of associating the same delegate to all exchanges, the exchange acceptor is able to assign a delegate to each new exchange when an unsolicited message arrives.

There are several usage of unsolicited message handler in CHIP
1. IM: the newly create exchange should associate to ReadHandler/ReadClient when possible. W/o doing this, there is a bug the `OnResponseTimeout` is not handl [... cut]
closed 2022-04-15 by kghost