project-chip/connectedhomeip #8360: [pase] PBKDFParamRequest payload is incomplete

turon, 2021-07-14:
#### Problem

The spec defines PBKDFParamRequest message as:
```
struct {
  byte randomFromInitiator[32];
  byte initiatorSessionId[2];
  byte passcodeID[2];
  boolean hasPBKDFParameters;
} PBKDFParamRequest
```

Yet the implementation in `PASESession::SendPBKDFParamRequest` only constructs the first field.
 
 #### Proposed Solution
Fix implementation to match spec.

yufengwangca, 2022-02-01:
@emargolis  this seems related to the work you recently did, could you help to take a look?

emargolis, 2022-02-02:
From my inspection of the latest SDK code this issue was already addressed.

@pan-apple @bzbarsky-apple  please confirm

bzbarsky-apple, 2022-02-04:
Yes, this looks good now.

closed 2022-02-04 by bzbarsky-apple