project-chip/connectedhomeip #8387: Is should be allowed to initialize ChipCertificateSet with Zero Length Decode Buffer

emargolis, 2021-07-14:
#### Problem
When certificates loaded into `ChipCertificateSet` don't need to calculate Hash, the decode buffer shouldn't be used. However, current implementation requires non-zero buffer allocation also for these case.

 #### Proposed Solution
Allow `ChipCertificateSet::Init()` to be called with zero length decode buffer.

bzbarsky-apple, 2021-07-14:
So the idea here is that if all the `LoadCert` calls will not pass `kGenerateTBSHash` then we can have a zero second arg to `Init()`?

Maybe the right answer is that the cert decode flags should be passed to `Init`, not the individual `LoadCert` calls?

emargolis, 2021-07-14:
> So the idea here is that if all the `LoadCert` calls will not pass `kGenerateTBSHash` then we can have a zero second arg to `Init()`?

Yes
 
> Maybe the right answer is that the cert decode flags should be passed to `Init`, not the individual `LoadCert` calls?

cert decode flags are individual per certificate so there is no unique set of flags to be passed to `Init` method

emargolis, 2022-01-26:
resolved with #8418

closed 2022-01-26 by emargolis