project-chip/connectedhomeip #8405: ESP32 flashing script crashes

mspang, 2021-07-15:
#### Problem
```
yuif:43297:~/connectedhomeip5/examples/all-clusters-app/esp32% idf.py build flashing_script
Executing action: all (aliases: build)
Running ninja in directory /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build
Executing "ninja all"...
[1/8] Performing build step for 'bootloader'
ninja: no work to do.
[2/6] Performing build step for 'chip_gn'
ninja: no work to do.
Executing action: flashing_script
Running ninja in directory /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build
Executing "ninja flashing_script"...
ninja: no work to do.
Done
```

```
./build/chip-all-clusters-app.flash.py 
Traceback (most recent call last):
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/./build/chip-all-clusters-app.flash.py", line 23, in <module>
    sys.exit(esp32_firmware_utils.Flasher(**DEFAULTS).flash_command(sys.argv))
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/firmware_utils.py", line 397, in flash_command
    return self.parse_argv(argv).actions().status()
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp32_firmware_utils.py", line 440, in actions
    if self.flash(flash).err:
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp32_firmware_utils.py", line 351, in flash
    return self.run_tool(
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/firmware_utils.py", line 204, in run_tool
    arguments = self.format_command(arguments, opt=option_map)
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/firmware_utils.py", line 328, in format_command
    result += self.format_command(i, args, opt)
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/firmware_utils.py", line 328, in format_command
    result += self.format_command(i, args, opt)
  File "/ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/firmware_utils.py", line 362, in format_command
    raise ValueError('Unknown: {}'.format(template))
ValueError: Unknown: None

```

I'm not sure what's wrong but as we need this tool for our interop testing, some assistance would be greatly appreciated.

@kpschoedel 
@sweetymhaiske

mspang, 2021-07-15:
@cecille apparently on it?

cecille, 2021-07-15:
Sorry - missed the notifications. https://github.com/project-chip/connectedhomeip/issues/8357 and https://github.com/project-chip/connectedhomeip/pull/8358. Just need a few to fix up the fails on the PR.

closed 2022-01-26 by woody-apple