project-chip/connectedhomeip #8406: Ubuntu 20.04.1 TLS source ./scripts/bootstrap.sh ERROR subprocess.CalledProcessError:Command [...]returned non-zero exit status 1.

736676938, 2021-07-15:
zhaoxu@ubuntu:/mnt/hgfs/shear/ceshi/connectedhomeip$ source ./scripts/bootstrap.sh 

  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[\]^[|]
b'[P19632 01:12:12.939 client.go:306 W] RPC failed transiently. Will retry in 1s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 203.111.254.117:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"1s"}\n[P19632 01:12:36.963 client.go:306 W] RPC failed transiently. Will retry in 2s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 203.111.254.117:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"2s"}\n[P19632 01:13:00.011 client.go:306 W] RPC failed transiently. Will retry in 4s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 203.111.254.117:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"4s"}\n[P19632 01:13:25.083 client.go:306 W] RPC failed transiently. Will retry in 8s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 203.111.254.117:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"8s"}\n[P19632 01:13:56.107 client.go:306 W] RPC failed transiently. Will retry in 16s   {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 104.244.46.85:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"16s"}\n[P19632 01:14:33.158 client.go:306 W] RPC failed transiently. Will retry in 32s   {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 104.244.46.85:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"32s"}\n[P19632 01:15:28.210 client.go:306 W] RPC failed transiently. Will retry in 1m4s  {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 104.244.46.85:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"1m4s"}\n[P19632 01:16:53.312 client.go:306 W] RPC failed transiently. Will retry in 2m8s  {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 104.244.46.85:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"2m8s"}\n[P19632 01:19:22.397 client.go:306 W] RPC failed transiently. Will retry in 4m16s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 75.126.115.192:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"4m16s"}\n[P19632 01:23:59.546 client.go:306 W] RPC failed transiently. Will retry in 8m32s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 162.125.2.5:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"8m32s"}\n[P19632 01:32:52.692 client.go:350 W] RPC failed permanently: rpc error: code = Internal desc = prpc: when sending request: Post "https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix": dial tcp 108.160.167.174:443: connect: connection refused {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 108.160.167.174:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository"}\nError: prpc: when sending request: Post "https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix": dial tcp 108.160.167.174:443: connect: connection refused.\n'
Traceback (most recent call last):
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 647, in <module>
    sys.exit(main())
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 639, in main
    return EnvSetup(**vars(parse())).setup()
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 345, in setup
    result = step(spin)
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 432, in cipd
    spin=spin):
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 199, in update
    if not check_auth(cipd, package_files, spin):
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 116, in check_auth
    inaccessible_paths = _check_all_paths()
  File "/mnt/hgfs/shear/ceshi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 100, in _check_all_paths
    [cipd, 'ls', path], stderr=subprocess.STDOUT).decode()
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/cipd', 'ls', 'gn/gn']' returned non-zero exit status 1
zhaoxu@ubuntu:/mnt/hgfs/shear/ceshi/connectedhomeip$ source ./scripts/bootstrap.sh 

  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[/][P20414 01:54:44.885 deployer.go:353 W] Deploy aborted, cleaning up /mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/pigweed/.cipd/pkgs/0/TEQA2BwAc08bDs0UtkrDd43ZOdKdgH8TvQijwfjdxIsC
[P20414 01:54:44.898 client.go:1718 E] Failed to install gn/gn/linux-amd64:TEQA2BwAc08bDs0UtkrDd43ZOdKdgH8TvQijwfjdxIsC - symlink TEQA2BwAc08bDs0UtkrDd43ZOdKdgH8TvQijwfjdxIsC /mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/pigweed/.cipd/pkgs/0/0dvpUxZPJggP: operation not supported
[P20414 01:54:45.021 deployer.go:353 W] Deploy aborted, cleaning up /mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/pigweed/.cipd/pkgs/1/wq2hFn8Q-1WOBOuxw2U33IWshwcfJgH-a7Zt2Bq-viYC
[P20414 01:54:45.032 client.go:1718 E] Failed to install infra/ninja/linux-amd64:wq2hFn8Q-1WOBOuxw2U33IWshwcfJgH-a7Zt2Bq-viYC - symlink wq2hFn8Q-1WOBOuxw2U33IWshwcfJgH-a7Zt2Bq-viYC /mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/pigweed/.cipd/pkgs/1/qVxeNfuBAtvJ: operation not supported
[... cut ...]
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/pigweed.ensure', '-root', '/mnt/hgfs/shear/ceshi/connectedhomeip/.environment/cipd/pigweed', '-log-level', 'warning', '-cache-dir', '/home/zhaoxu/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1
zhaoxu@ubuntu:/mnt/hgfs/shear/ceshi/connectedhomeip$ 

HELP

iot-lorawan, 2021-12-02:
Hi,
    我这边souce的时候也出现了一样的client.go:1666 E] [unzip  8/17] Failed to install infra/3pp/tools/go/linux-amd64: symlink 类似问题，后面你这边有解决吗

iot-lorawan, 2021-12-02:
log:
[P6696 16:46:48.909 deployer.go:346 W] [unzip  8/17] Deploy aborted, cleaning up /mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed/.cipd/pkgs/16/-xKRu10txxPJsPJD1jyJi-kISvT6ZvAnXLKqmKPt-ZsC
[P6696 16:46:51.260 client.go:1666 E] [unzip  8/17] Failed to install infra/3pp/tools/go/linux-amd64: symlink -xKRu10txxPJsPJD1jyJi-kISvT6ZvAnXLKqmKPt-ZsC /mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed/.cipd/pkgs/16/dLy3K94wsj7C: operation not supported
Error: failed to update the deployment.
/mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed.ensure -root /mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed -log-level debug -json-output /mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed-output.json -cache-dir /root/.cipd-cache-dir -max-threads 0
None
Traceback (most recent call last):
  File "/mnt/hgfs/matter/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 679, in <module>
    sys.exit(main())
  File "/mnt/hgfs/matter/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in main
    return EnvSetup(**vars(parse())).setup()
  File "/mnt/hgfs/matter/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 414, in setup
    result = step(spin)
  File "/mnt/hgfs/matter/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 497, in cipd
    if not cipd_update.update(cipd=cipd_client,
  File "/mnt/hgfs/matter/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 250, in update
    subprocess.check_call(cmd,
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed.ensure', '-root', '/mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed', '-log-level', 'debug', '-json-output', '/mnt/hgfs/matter/chip/connectedhomeip/.environment/cipd/pigweed-output.json', '-cache-dir', '/root/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1.

closed 2022-01-26 by woody-apple