project-chip/connectedhomeip #8407: python controller connect timeout

JasonHu1, 2021-07-15:
#### Problem
Steps I use：
step1:
git clone --recurse-submodules https://github.com/project-chip/connectedhomeip.git
step2:
cd connectedhomeip 
git submodule update --init
step3:
source scripts/activate.sh
step4:
git checkout test_event_3.1
ps:commit a8ef4197acd4e879141077820abcf5d9bc76535b (HEAD -> test_event_3.1, tag: test_event_3_2021_06_10, origin/test_event_3.1)
step5:
scripts/build_python.sh -m platform
step6:
source out/python_env/bin/activate
step7:
chip-device-ctrl
step8:
 ble-scan
step9:
connect -ble 3840 73141520 1231

An error message appears when executing this command
The log is as follows：

chip-device-ctrl > ble-scan
2021-07-15 09:56:40,751 ChipBLEMgr   INFO     scanning started
2021-07-15 09:56:40,904 ChipBLEMgr   INFO     Name            = EFR32_LIGHT
2021-07-15 09:56:40,904 ChipBLEMgr   INFO     ID              = d558231f-fb05-389c-bae2-6b797b364d28
2021-07-15 09:56:40,905 ChipBLEMgr   INFO     RSSI            = -25
2021-07-15 09:56:40,907 ChipBLEMgr   INFO     Address         = 7D:50:4C:30:3E:48
2021-07-15 09:56:40,910 ChipBLEMgr   INFO     Pairing State   = 0
2021-07-15 09:56:40,911 ChipBLEMgr   INFO     Discriminator   = 3840
2021-07-15 09:56:40,911 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-07-15 09:56:40,911 ChipBLEMgr   INFO     Product Id      = 21324
2021-07-15 09:56:40,913 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
2021-07-15 09:56:40,915 ChipBLEMgr   INFO     Adv Data        = 00000f5a234c53
2021-07-15 09:56:40,915 ChipBLEMgr   INFO
2021-07-15 09:56:50,796 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > connect -ble 3840 73141520 1231
Device is assigned with nodeid = 1231
[1626314213.950917][51703] CHIP:BLE: BLE removing known devices.
[1626314213.953479][51703] CHIP:BLE: BLE initiating scan.
[1626314214.602806][51703] CHIP:BLE: New device scanned: 7D:50:4C:30:3E:48
[1626314214.602909][51703] CHIP:BLE: Device discriminator match. Attempting to connect.
[1626314214.605998][51703] CHIP:BLE: Scan complete notification without an active scan.
[1626314215.192769][51703] CHIP:DL: **FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Software caused connection abort**
Failed to establish secure session to device: 4050
[1626314243.882106][51707] **CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 4050 (0x00000FD2): Timeout
Failed to establish secure session to device: 4050**
chip-device-ctrl >

 #### Proposed Solution
sub-device board is silabs 4161A

leasual, 2021-07-19:
i have the same issue. have you solved it? @JasonHu1

JasonHu1, 2021-11-01:
> i have the same issue. have you solved it? @JasonHu1
it not appear that i use the c++ chiptool

closed 2021-11-01 by JasonHu1