project-chip/connectedhomeip #8416: python chip-device-ctrl unable to establish BLE connection to nrf52840DK running lock-app example

cahty82008, 2021-07-15:
**Problem**
python chip-device-ctrl unable to establish BLE connection to nrf52840DK running lighting-app example

**commit**  6cf2f1d

**test environment**: 
python chip-device-ctrl running on raspberry Pi 4B(ubuntu 20.10) + nrf52840DK running lock-app example

**log**
```
rpi@rpi-desktop:~/work_src/connectedhomeip$ source out/python_env/bin/activate

(python_env) rpi@rpi-desktop:~/work_src/connectedhomeip$ sudo out/python_env/bin/chip-device-ctrl --bluetooth-adapter=hci0
[1626352994.404386][115396:115396] CHIP:IN: local node id is 0x000000000001B669
[1626352994.409139][115396:115396] CHIP:DL: Avahi client registered
[1626352994.411070][115396:115396] CHIP:ZCL: Using ZAP configuration...
[1626352994.411195][115396:115396] CHIP:ZCL: deactivate report event
[1626352994.411241][115396:115396] CHIP:CTL: Getting operational keys
[1626352994.411608][115396:115396] CHIP:CTL: Getting root certificate for the controller from the issuer
[1626352994.412604][115396:115396] CHIP:CTL: Generating operational certificate for the controller
[1626352994.413426][115396:115396] CHIP:CTL: Getting intermediate CA certificate from the issuer
[1626352994.413637][115396:115396] CHIP:CTL: GetIntermediateCACertificate returned 0
[1626352994.413851][115396:115396] CHIP:CTL: Generating credentials
[1626352994.414136][115396:115396] CHIP:CTL: Loaded credentials successfully
[1626352994.429047][115396:115405] CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > ble-scan 
2021-07-15 14:43:38,849 ChipBLEMgr   INFO     scanning started
2021-07-15 14:43:42,638 ChipBLEMgr   INFO     Name            = MatterLock
2021-07-15 14:43:42,638 ChipBLEMgr   INFO     ID              = 3c694ac0-ac45-3c36-ae70-2e98ce5b1921
2021-07-15 14:43:42,640 ChipBLEMgr   INFO     RSSI            = -60
2021-07-15 14:43:42,641 ChipBLEMgr   INFO     Address         = FE:AC:D7:A7:78:4F
2021-07-15 14:43:42,646 ChipBLEMgr   INFO     Pairing State   = 0
2021-07-15 14:43:42,646 ChipBLEMgr   INFO     Discriminator   = 3840
2021-07-15 14:43:42,646 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-07-15 14:43:42,647 ChipBLEMgr   INFO     Product Id      = 20043
2021-07-15 14:43:42,650 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
2021-07-15 14:43:42,651 ChipBLEMgr   INFO     Adv Data        = 00000f5a234b4e
2021-07-15 14:43:42,652 ChipBLEMgr   INFO     
2021-07-15 14:43:49,510 ChipBLEMgr   INFO     scanning stopped

chip-device-ctrl > connect -ble 3840 20202021 1234
Device is assigned with nodeid = 1234
[1626353041.904260][115396:115400] CHIP:BLE: BLE removing known devices.
[1626353041.921564][115396:115400] CHIP:BLE: BLE initiating scan.
[1626353042.367335][115396:115400] CHIP:BLE: New device scanned: FE:AC:D7:A7:78:4F
[1626353042.367422][115396:115400] CHIP:BLE: Device discriminator match. Attempting to connect.
[1626353042.371021][115396:115400] CHIP:BLE: Scan complete notification without an active scan.
[1626353044.945062][115396:115400] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1626353044.945306][115396:115405] CHIP:IN: Failed to establish BLE connection: Ble Error 40c (0x0000040C): Remote device closed BLE connection
[1626353044.945630][115396:115400] CHIP:DL: FAIL: BluezSubscribeCharacteristic : GDBus.Error:org.bluez.Error.Failed: Operation failed with ATT error: 0x0e
Failed to establish secure session to device: 50
[1626353071.876436][115396:115405] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 32 (0x00000032): Timeout
Failed to establish secure session to device: 50
```

leasual, 2021-07-19:
i have the same issue. have you solved it? @cahty82008

cahty82008, 2021-07-19:
no, not yet  @leasual

doublemis1, 2021-07-22:
Hi, I have verified this commit on my raspberry PI 4, and everything works.
Please verify if all the packages are installed based on that script: https://github.com/project-chip/connectedhomeip/blob/master/scripts/setup/raspberry_pi/setup_ubuntu_20_04_lts.sh ("You must REBOOT after pi-bluetooth has been installed").
If this issue has still occurred please restart the Bluetooth service on raspberry pi by: `sudo systemctl restart bluetooth.service`

closed 2022-09-22 by stale[bot]