project-chip/connectedhomeip #8433: AddNOC needs to do validation of the received certificate

bzbarsky-apple, 2021-07-15:
#### Problem
Per spec, "11.22.7.8. AddNOC Command" is supposed to do a number of validation steps (some of these listed in "11.22.7.7. AddNOC and UpdateNOC Commands Overview" and some under the AddNOC command itself.

Importantly:

1. `CHIP_Crypto_VerifyChain` needs to happen.
2. The DN encoding rules need to be enforced.  This verifies, for example, that the NOC subject has a FabricId and NodeId.

(other things too, but those were the obvious things that jumped out at me as bare minimum verification).

Right now we just store the NOC and ICAC without doing any sort of verification on them.

 #### Proposed Solution
Implement the spec-required verification.  That will allow us to rely on being able to extract an operational id from an installed NOC, because we have already checked the DN encoding rules.

@pan-apple @shana-apple

bzbarsky-apple, 2022-05-04:
At this point, AddNOC calls into `FabricInfo::SetFabricInfo` which calls `FabricInfo::VerifyCredentials`, which:

1. Validates that the NOC chains up to the root properly.
2. Verifies the NOC has a node id and fabric id.
3. Verifies ICAC fabric id (if present) matches NOC fabric id.
4. Verifies root fabric id (if present) matches NOC fabric id.

We seem to be missing the following checks from "11.18.7.7. AddNOC and UpdateNOC Commands Overview" in the spec

> The public key of the NOC SHALL match the last generated operational public key on this session, and therefore the public key present in the last CSRResponse provided to the Administrator or Commissioner that sent the AddNOC or UpdateNOC command.

and from "11.18.7.8. AddNOC Command" the spec has:

> If the IpkValue field is absent or empty, then the device SHALL process the error by responding with a StatusCode of MissingIpk as described in Section 11.18.7.7.2, “Handling Errors”.

but our impl checks `ipkValue.size() == Crypto::CHIP_CRYPTO_SYMMETRIC_KEY_LENGTH_BYTES` (which the spec does not seem to require?) and does not return the right error.

@emargolis @tcarmelveilleux anything else we are missing?  Do we need a spec update for the IPK size check?

bzbarsky-apple, 2022-05-17:
> We seem to be missing the following checks from "11.18.7.7. AddNOC and UpdateNOC Commands Overview" in the spec

The best place to put these is probably in `FabricInfo::SetFabricInfo`.

bzbarsky-apple, 2022-05-19:
After #18530 what's left here is the IPK issue from https://github.com/project-chip/connectedhomeip/issues/8433#issuecomment-1117597121, I think.

woody-apple, 2022-06-21:
Spec Issue Review: Assigning to @bzbarsky-apple to confirm the current state of things

closed 2022-06-21 by bzbarsky-apple