project-chip/connectedhomeip #8434: We should not require commissioners to add a root cert

bzbarsky-apple, 2021-07-15:
#### Problem
Right now our commissioning flow assumes the commissioner will call AddTrustedRootCertificate, and in particular we store the root cert in our "fabric" representation.  Per spec, a commissioner it's not clear that a commissioner is required to do this, and in any case trusted roots are not stored per-fabric in the spec.

 #### Proposed Solution
Move storage of trusted root certs out of `AdminPairingInfo`.

@shana-apple @pan-apple

bzbarsky-apple, 2022-01-26:
@msandstedt Do your recent changes resolve this?

bzbarsky-apple, 2022-06-18:
This is not resolved, and this is in fact a spec-compliance issue...

andy31415, 2022-07-12:
Review: spec is unclear if keystore is to be shared or not. A spec issue exists:

https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5420

woody-apple, 2022-07-13:
Cert Blocker Review: Closing this as we believe the other issue will go into 1.0 spec.

closed 2022-07-13 by woody-apple